espressif/esp-idf #2134: Socket accept/close memory leak

xarrange, 2018-06-30:
## Environment

- Development Kit:      none
- Core (if using chip or module):|ESP-WROOM32
- IDF version:  esp-idf-3.1-beta1
- Development Env:		other
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

I wrote very small test program that should listen for connections, accept them and close.
Somehow free heap value decreases on every accept/close. And there is no any other stuff working in background

### Expected Behavior
free heap remains on same value

### Actual Behavior
 free heap value decreases

### Code to reproduce this issue
```
void listen_thread(void *arg) {
	printf("Free heap: %u\n", xPortGetFreeHeapSize());
	
	uint16_t port = 2048;
	
	struct sockaddr_in addr;
	addr.sin_family = AF_INET;
	addr.sin_port = htons(port);
	addr.sin_addr.s_addr = htonl(INADDR_ANY);
	
	int server = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
	int yes = 1;
	setsockopt(server, SOL_SOCKET, SO_REUSEADDR, &yes, sizeof(int));
	bind(server, (struct sockaddr*)&addr, sizeof(addr));
		
	listen(server, 1);
	int client = 0;
	
	while (1) {
		printf("Free heap: %u\n", xPortGetFreeHeapSize());
		
		client = accept(server, NULL,NULL);
		vTaskDelay(1000 / portTICK_PERIOD_MS);
		shutdown(client, SHUT_RDWR);
		close(client);
	}	
}

esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
	uint32_t current_ip = 0;
	
	switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
	case SYSTEM_EVENT_STA_GOT_IP:
		xTaskCreate(listen_thread, "lth", 4096, NULL, 7, NULL);			
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:	
		esp_wifi_connect();
        break;
    }
    return ESP_OK;
}

void app_main(void)
{	
	nvs_flash_init();	
	esp_event_loop_init(wifi_event_handler, NULL);
	wifi_conn_init();
}
```

## Debug Logs
```
Free heap: 168204
Free heap: 167564
Free heap: 167280
Free heap: 166992
Free heap: 166704
Free heap: 166480
Free heap: 166128
Free heap: 165836
Free heap: 165548
Free heap: 165668
Free heap: 165668
Free heap: 165668
Free heap: 165668
Free heap: 165464
Free heap: 165464
Free heap: 165260
Free heap: 165260
Free heap: 165260
Free heap: 165052
Free heap: 164856
Free heap: 164716
Free heap: 164444
Free heap: 164236
Free heap: 164036

```

karawin, 2018-07-03:
Some internal buffers are freed with a long tempo. Try to add a pause of many minuts and see the heap size.

FayeY, 2018-07-06:
Hi @xarrange , is there any update? Did the above suggestion help you resolve the issue?

TimXia, 2018-07-08:
Also, for every socket id, some internal memories like locks are allocated and will not be deallocated until system restarts.

xarrange, 2018-07-08:
> Some internal buffers are freed with a long tempo. Try to add a pause of many minuts and see the heap size.

Indeed, ~5 minutes later heap starts to remain on same level (significantly lower then it was) with several peaks from time to time.
If after that I stop listening for connection, how long to wait to get that piece of memory to be freed?

> Also, for every socket id, some internal memories like locks are allocated and will not be deallocated until system restarts.

If so it's a huge bug, but I don't believe it's true. Otherwise all high level protocol functions (HTTP(s), AWS, etc..) are pointless, because after couple connects you just don't have any memory.

projectgus, 2018-07-09:
Hi @xarrange,

The TCP standard requires sockets to go into a TIME_WAIT state after being closed from one end.  ESP-IDF has to do this in order to behave correctly in a TCP/IP network.

There are upper limits on the memory usage of TIME_WAIT sockets - if memory is low or the maximum socket limit is reached then old TIME_WAIT sockets will be cleaned up preemptively. They're also timed out automatically after a period of time.

If you're particularly concerned about this then you can reduce the maximum open socket count in menuconfig:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html?highlight=configuration#envvar-CONFIG_LWIP_MAX_SOCKETS

(This will cause TIME_WAIT sockets to be timed out early and recycled, once the limit is reached.)

> If so it's a huge bug, but I don't believe it's true. Otherwise all high level protocol functions (HTTP(s), AWS, etc..) are pointless, because after couple connects you just don't have any memory.

@TimXia is referring to each unique socket id (file descriptor) supported by the system, not each socket. So the first time a brand new socket (never before used file descriptor) is opened some one-time memory (locks, etc) is allocated. However if this file descriptor is later reused for another different socket, there's no new allocation. So there is a practical upper limit of `CONFIG_LWIP_MAX_SOCKETS` on the number of times this "one time allocation" will occur, after this any new socket will reuse a previously seen file descriptor.

(Note that these one time allocations are totally separate from sockets in TIME_WAIT, but having sockets in TIME_WAIT means that file descriptors often aren't immediately reused after the socket is closed, unless there are no other available file descriptors due to other open sockets.)

xarrange, 2018-07-09:
@projectgus, thanks for deep explanation! No I have full picture

closed 2018-07-09 by xarrange