espressif/esp-idf #2137: bootloader factory reset issue

chegewara, 2018-07-02:
When bootloader `GPIO triggers reset factory` is enabled all seems works except one thing. When `Comma-separated names of partitions to clear on factory reset` is empty NVS is not erased but i cant read any value from it. I have to mention that my factory app is build from arduino-ide and flashed with esp32 download tool, but im guessing it should not be problem. 
Tests procedure:
- i have old version bootloader (without factory reset enabled) and i can read and write nvs wifi credentials,
- with new bootloader version (with factory reset enabled) wifi credentials cant be read; there is no error, so i guess there is empty string read or no string read at all
- both bootloader version are compiled with master branch (make bootloader) and flashed with esp32 download tool, only difference is `GPIO triggers reset factory` option

KonstantinKondrashov, 2018-07-02:
Hi @chegewara. Could you please give sdkconfig and a starting log? 

By default the pin `GPIO triggers reset factory` have to hold down about 5 seconds. Do you see this delay or the message `I (5085) boot: Detect a condition of the factory reset`?

chegewara, 2018-07-02:
Everything works like expected. Bootloader is starting from factory on pin hold (after some time, like expected). Like i said, only issue is that my wifi credentials are not loaded from nvs, even if they are stored. I know they are stored because when i flash back bootloader without GPIO reset to factory enabled i can read it from nvs. Give me few sec and i will provide logs.

chegewara, 2018-07-02:
Here you have logs with factory reset enabled, what is strange first time after reboot from factory wifi credentials has been read from nvs but every next reboot is without nvs data (thats why you logs from have 2 times reboot):
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8608
ho 0 tail 12 room 4
load:0x40078000,len:8672
load:0x40080000,len:10040
entry 0x400804c4
D (75) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (37) boot: ESP-IDF v3.2-dev-87-gb1fa1ba6 2nd stage bootloader
I (37) boot: compile time 07:37:03
I (37) boot: Enabling RNG early entropy source...
D (43) boot: magic e9
D (45) boot: segments 04
D (48) boot: spi_mode 02
D (51) boot: spi_speed 00
D (53) boot: spi_size 02
I (56) boot: SPI Speed      : 40MHz
I (60) boot: SPI Mode       : DIO
I (64) boot: SPI Flash Size : 4MB
D (68) bootloader_flash: mmu set paddr=00000000 count=1
D (73) boot: mapped partition table 0x8000 at 0x3f408000
D (79) flash_parts: partition table verified, 6 entries
I (84) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
D (95) boot: load partition table entry 0x3f408000
D (100) boot: type=1 subtype=2
I (103) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (111) boot: load partition table entry 0x3f408020
D (116) boot: type=1 subtype=0
I (119) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (126) boot: load partition table entry 0x3f408040
D (131) boot: type=1 subtype=1
I (134) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (142) boot: load partition table entry 0x3f408060
D (147) boot: type=0 subtype=0
I (150) boot:  3 factory          factory app      00 00 00010000 000d4800
D (157) boot: load partition table entry 0x3f408080
D (162) boot: type=0 subtype=10
I (166) boot:  4 ota              OTA app          00 10 000f0000 00310000
I (173) boot: End of partition table
D (177) boot: OTA data offset 0xd000
D (181) bootloader_flash: mmu set paddr=00000000 count=1
D (186) boot: OTA sequence values A 0x00000001 B 0xffffffff
D (192) boot: Only OTA sequence A is valid. Mapping seq 0 -> OTA slot 0
I (5199) boot: Detect a condition of the factory reset
I (5199) boot: Data partitions to erase:
D (5199) bootloader_flash: mmu set paddr=00000000 count=1
D (5203) boot_comm: mapped partition table 0x8000 at 0x3f408000
D (5209) flash_parts: partition table verified, 6 entries
I (5215) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (5222) boot_comm:  0 nvs              data  00009000 00004000 [no]
I (5285) boot_comm:  1 otadata          data  0000d000 00002000 [yes]
I (5285) boot_comm:  2 phy_init         data  0000f000 00001000 [no]
D (5288) boot: OTA data offset 0xd000
D (5291) bootloader_flash: mmu set paddr=00000000 count=1
D (5297) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (5303) boot: OTA sequence numbers both empty (all-0xFF)
I (5308) boot: Defaulting to factory image
D (5313) boot: Trying partition index -1 offs 0x10000 size 0xd4800
D (5319) esp_image: reading image header @ 0x10000
[... 164 lines cut ...]
V (619) esp_image: segment 6 map_segment 0 segment_data_offs 0xdfb04 load_addr 0x400c0000
I (627) esp_image: segment 6: paddr=0x000dfb04 vaddr=0x400c0000 size=0x00000 (     0) load
D (636) bootloader_flash: mmu set paddr=000d0000 count=1
V (641) esp_image: image start 0x00010000 end of last section 0x000dfb04
D (648) bootloader_flash: mmu set block paddr=0x000d0000 (was 0xffffffff)
D (655) esp_image: Calculated hash: 4d1aa63d
D (659) bootloader_flash: mmu set paddr=000d0000 count=1
I (668) boot: Loaded app from partition at offset 0x10000
I (671) boot: Disabling RNG early entropy source...
D (676) boot: Mapping segment 0 as IROM
D (680) boot: Mapping segment 2 as DROM
D (684) boot: calling set_cache_and_start_app
D (688) boot: configure drom and irom and start
V (693) boot: d mmu set paddr=00040000 vaddr=400d0000 size=578236 n=9
V (699) boot: rc=0
V (702) boot: rc=0
V (704) boot: i mmu set paddr=00010000 vaddr=3f400000 size=193552 n=3
V (710) boot: rc=0
V (712) boot: rc=0
D (714) boot: start: 0x400828ac
START
     ssid:
password :

```

chegewara, 2018-07-02:
Just in case, here you have logs with boot without factory reset enabled:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8092
load:0x40078000,len:8672
load:0x40080000,len:7812
entry 0x400803d8
D (70) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (35) boot: ESP-IDF v3.2-dev-87-gb1fa1ba6 2nd stage bootloader
I (35) boot: compile time 07:44:39
I (35) boot: Enabling RNG early entropy source...
D (41) boot: magic e9
D (43) boot: segments 04
D (46) boot: spi_mode 02
D (48) boot: spi_speed 00
D (51) boot: spi_size 02
I (53) boot: SPI Speed      : 40MHz
I (58) boot: SPI Mode       : DIO
I (62) boot: SPI Flash Size : 4MB
D (66) bootloader_flash: mmu set paddr=00000000 count=1
D (71) boot: mapped partition table 0x8000 at 0x3f408000
D (76) flash_parts: partition table verified, 6 entries
I (82) boot: Partition Table:
I (85) boot: ## Label            Usage          Type ST Offset   Length
D (92) boot: load partition table entry 0x3f408000
D (97) boot: type=1 subtype=2
I (100) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (108) boot: load partition table entry 0x3f408020
D (113) boot: type=1 subtype=0
I (116) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (124) boot: load partition table entry 0x3f408040
D (129) boot: type=1 subtype=1
I (132) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (139) boot: load partition table entry 0x3f408060
D (144) boot: type=0 subtype=0
I (147) boot:  3 factory          factory app      00 00 00010000 000d4800
D (155) boot: load partition table entry 0x3f408080
D (160) boot: type=0 subtype=10
I (163) boot:  4 ota              OTA app          00 10 000f0000 00310000
I (171) boot: End of partition table
D (175) boot: OTA data offset 0xd000
D (178) bootloader_flash: mmu set paddr=00000000 count=1
D (184) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (189) boot: OTA sequence numbers both empty (all-0xFF)
I (195) boot: Defaulting to factory image
D (200) boot: Trying partition index -1 offs 0x10000 size 0xd4800
D (206) esp_image: reading image header @ 0x10000
D (210) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (217) esp_image: image header: 0xe9 0x07 0x02 0x02 400828ac
V (223) esp_image: loading segment header 0 at offset 0x10018
V (229) esp_image: segment data length 0x2f410 data starts 0x10020
V (235) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
I (243) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2f410 (193552) map
D (252) bootloader_flash: mmu set paddr=00010000 count=3
V (296) esp_image: loading segment header 1 at offset 0x3f430
D (296) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
V (297) esp_image: segment data length 0xbd8 data starts 0x3f438
V (304) esp_image: segment 1 map_segment 0 segment_data_offs 0x3f438 load_addr 0x3ffc0000
I (312) esp_image: segment 1: paddr=0x0003f438 vaddr=0x3ffc0000 size=0x00bd8 (  3032) load
D (321) bootloader_flash: mmu set paddr=00030000 count=2
V (327) esp_image: loading segment header 2 at offset 0x40010
[... 30 lines cut ...]
D (645) bootloader_flash: mmu set block paddr=0x000d0000 (was 0xffffffff)
D (652) esp_image: Calculated hash: 4d1aa63d
D (657) bootloader_flash: mmu set paddr=000d0000 count=1
I (665) boot: Loaded app from partition at offset 0x10000
I (668) boot: Disabling RNG early entropy source...
D (674) boot: Mapping segment 0 as IROM
D (678) boot: Mapping segment 2 as DROM
D (681) boot: calling set_cache_and_start_app
D (686) boot: configure drom and irom and start
V (690) boot: d mmu set paddr=00040000 vaddr=400d0000 size=578236 n=9
V (697) boot: rc=0
V (699) boot: rc=0
V (701) boot: i mmu set paddr=00010000 vaddr=3f400000 size=193552 n=3
V (708) boot: rc=0
V (710) boot: rc=0
D (712) boot: start: 0x400828ac
START
     ssid: SSID
password : PASS
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 201 - NO_AP_FOUND
STA: Failed!
            IP address: 0.0.0.0
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 201 - NO_AP_FOUND

```

chegewara, 2018-07-02:
sdkconfig:
https://gist.github.com/chegewara/0a66af9949abd067ea6bee3bf40659d3

chegewara, 2018-07-02:
Arduino code that is writing/reading nvs strings:
```
#include <Preferences.h>

Preferences pref;

void nvs_init(){
  pref = Preferences();
  pref.begin("wifi");
}

String get_wifi_ssid(){
  return pref.getString("ssid", "");
}

String get_wifi_pass(){
  return pref.getString("pass", "");
}

void set_wifi_ssid(const String ssid){
  pref.putString("ssid", ssid);
}

void set_wifi_pass(const String pass){
  pref.putString("pass", pass);
}

void set_request_token(const String token){
  pref.putString("token", token);
}

String get_request_token(){
  return pref.getString("token", "");
}
```

chegewara, 2018-07-02:
One more thing i forgot to mention. Issue is not only to read NVS but i cant write to nvs either.

chegewara, 2018-07-02:
I think i found issue. Its gpio16, because with gpio14 seems works fine.

FayeY, 2018-07-06:
Hi @chegewara , sorry that I am not an engineer, so I'm a little confused with your last post. 
Is this issue solved now? Can I close it?

chegewara, 2018-07-06:
@FayeY 
Its hard to say from my point of view. This can be issue or pin 16 cant be used in factory reset because hardware restrictions, so its question to dev team now.

negativekelvin, 2018-07-12:
Do you have any log of nvs_init result?