espressif/esp-idf #2138: esp_http_client redirect issue

chegewara, 2018-07-02:
I want to say that i have tested this url (or evn mp3 file, which hardcoded works) with esp_http_client example and it seems to work (redirect as far as i could see), but for some reason its crashing with my code:
```
I (25487) web_radio: https://maestro.emfcdn.com/stream_for/sa008/tunein  // <-- value retrieved from audio/x-mpegurl (m3u playlist) and set as url for new http request (called from outside of this request)

W (25495) web_radio: Success: add track to queue - 2
W (25500) web_radio: HTTP_EVENT_ON_FINISH stack: 4572

I (25506) web_radio: HTTP_EVENT_ON_FINISH
I (25511) web_radio: HTTP chunk encoding Status = 200, content_length = 51
I (25519) web_radio: HTTP_EVENT_DISCONNECTED

E (25536) web_radio: 148: http_client_get URL: https://maestro.emfcdn.com/stream_for/sa008/tunein, 51  // <-- number of bytes in string to check if there is any non-printable char

E (25546) HTTP_CLIENT: Error parse url https://maestro.emfcdn.com/stream_for/sa008/tunein // <-- its redirecting uri

E (25555) HTTP_CLIENT: Error set default configurations
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ef8cb  PS      : 0x00060330  A0      : 0x801328c3  A1      : 0x3ffde830
A2      : 0x00000000  A3      : 0x3ffdabb0  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00000001  A7      : 0x00000004  A8      : 0x800ef4f8  A9      : 0x3ffde810
A10     : 0x00000000  A11     : 0x3ffb6ea0  A12     : 0x3f40eed0  A13     : 0x0000001f
A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000040  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

Backtrace: 0x400ef8cb:0x3ffde830 0x401328c0:0x3ffde850 0x40132967:0x3ffde8d0
```

Backtrace ends in here:
https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L943
https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L487

I have to add that works when i hardcode this url (or any other that is causing crash) in my code. Here is part of my code that is making initialization:
```
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
....
        case HTTP_EVENT_ON_FINISH:
        {
            if(content_type == AUDIO_MPEG)
            {
             
            }else if(content_type == MPEG_URL)
            {
                web_radio_t *radio_config;
                radio_config = calloc(1, sizeof(web_radio_t));
                memcpy(radio_config, evt->user_data, sizeof(web_radio_t));
                // radio_config->url = calloc(strlen(tmp_url)+1, sizeof(char));
                strcpy(radio_config->url, tmp_url);
                web_radio_add_to_queue(radio_config);
            }
....
}
```
```
void http_perform_as_stream_reader(void *_config)
{
    web_radio_t *web_config = _config;
    char *buffer = malloc(MAX_HTTP_RECV_BUFFER);
    if (buffer == NULL) {
        ESP_LOGE(TAG, "Cannot malloc http receive buffer");
        return;
    }
    
    esp_http_client_config_t config = {
        .url = web_config->url, // <-- char* variable
        .event_handler = _http_event_handler,
        .user_data = web_config,
    };
.....
}
```

closed 2018-07-02 by chegewara

chegewara, 2018-07-02:
Im almost sure its my fault, just i dont know where. Only time it works is when i hardcode this line:
`radio_config->url = "https://some_url";'
Could you help me please @projectgus

projectgus, 2018-07-02:
@tuanpmt can you take a look, please?

WayneKeenan, 2018-11-25:
I overcame this error when I increased the stack size of my task.

Gohith, 2019-05-24:
@WayneKeenan 
How to Increase stack size