espressif/esp-idf #2139: symbol 'configMAX_PRIORITIES could not be resolved'

ahmedawais, 2018-07-02:
i am running the examples on ECLIPSE IDE(neon) on ubuntu but facing the error as symbol 'configMAX_PRIORITIES could not be resolved' can anybody tell what is missing ? Although program is able to build but with error
[image]

negativekelvin, 2018-07-02:
freertos/FreeRTOS.h

projectgus, 2018-07-03:
Hi @ahmedawais,

If the project builds despite these errors, it means that Eclipse isn't able to parse information out of the build to learn about which header paths are included.

Double check all of the items shown here are set as shown (we recently changed/clarified a couple of these, as well):
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/eclipse-setup.html#project-properties

If you change any setting, you may have to do a clean build of the project to refresh Eclipse's cache of header paths.

Angus

chegewara, 2018-07-03:
Remember to add {$PROJECT}/build/something(dont remember full path) to your path, its where sdkconfig.h will be stored when you first time build project.

ahmedawais, 2018-07-04:
Thanks by adding freeRTOS it worked.

closed 2018-07-04 by ahmedawais