espressif/esp-idf #2141: Error handling for Ethernet / PHY initialization

Oromis, 2018-07-03:
## Environment

- Development Kit:      none
- Core (if using chip or module): ESP32-WROOM-32
- IDF version: 8d3d3fd (v3.0 branch)
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB + external 5V

## Problem Description

Our ESP32-based product can connect to the internet via either WiFi or Ethernet. We recently had the problem that the LAN8720 chip broke for a customer and this prevented the ESP32 from starting up at all. We use the following code to initialize the ethernet module:

    esp_err_t ret;

    eth_config_t config = DEFAULT_ETHERNET_PHY_CONFIG;
    config.phy_addr = PHY1;
    config.gpio_config = eth_gpio_config_rmii;
    config.tcpip_input = tcpip_adapter_eth_input;
    config.clock_mode = ETH_CLOCK_GPIO0_IN;
    config.phy_power_enable = phy_device_power_enable_via_gpio;

    ret = esp_eth_init(&config);

This works great when the LAN8720 is connected, but if it isn't, it causes the ESP32 to get stuck in an infinite loop trying to reset the PHY until the task watchdog kills it and restarts the microcontroller. The output in the console:

```
I (3385) emac: emac start !!!

I (3389) emac: emac resetting ....
I (3393) emac: emac resetting ....
I (3397) emac: emac resetting ....
I (3401) emac: emac resetting ....
I (3405) emac: emac resetting ....
I (3409) emac: emac resetting ....
I (3415) emac: emac resetting ....
I (3419) emac: emac resetting ....
[...]
I (8017) emac: emac resetting ....
I (8021) emac: emac resetting ....
I (8025) emaTask watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: emacT
Aborting.
abort() was called at PC 0x400d1145 on core 0
```

So, this chunk of code within the internal "emacT" task is the culprit:

```
void emac_reset(void)
{
    REG_SET_BIT(EMAC_DMABUSMODE_REG, EMAC_SW_RST);

    while (REG_GET_BIT(EMAC_DMABUSMODE_REG, EMAC_SW_RST) == 1) {
        //nothing to do ,if stop here,maybe emac have not clk input.
        ESP_LOGI(TAG, "emac resetting ....");
    }

    ESP_LOGI(TAG, "emac reset done");
}
```

### Expected Behavior

If the PHY is not connected or broken, the ethernet initialization function should return an error return code or a system event shall be thrown, allowing programmatic detection of this error.

### Actual Behavior

The `emacT` task gets stuck in an infinite loop. After the timeout expires, the task watchdog restarts the chip (if it's enabled). I do realize that `emac_reset()` is called in a background task, so returning a value from `esp_eth_init()` is difficult - but throwing a system event would be feasible and sufficient in my mind. Just don't go into an infinite loop!

### Steps to repropduce

1. Disconnect a connected PHY from the ESP32 module (if connected)
2. Flash the ethernet example program onto the chip
3. See the message `emac resetting ....` appear on-screen forever

### Code to reproduce this issue

See above / see ethernet example

## Debug Logs

See above

xiongyumail, 2018-07-05:
@Oromis 
You can try to change the system frequency to 240M.

Oromis, 2018-07-05:
@xiongyumail thank you for the suggestion, but the CPU frequency is already set to 240M. Furthermore, I don't see how changing the frequency can influence this behavior. If the PHY is not connected, then we will always end up in this infinite loop, not matter the frequency.

Oromis, 2018-07-06:
I investigated a bit and copied the ESP-IDF ethernet component into my project. After a bit of trial-and-error, I found a solution that's working nicely and could be integrated into the main ESP-IDF project so that others could benefit from it. 

You can find the code changes in my post in the ESP-IDF forums: https://www.esp32.com/viewtopic.php?f=13&t=6286&p=27329#p27329

closed 2018-09-24 by igrr