espressif/esp-idf #2142: Feature request: Arduino as component

me21, 2018-07-03:
There are instructions on how to install Arduino framework as ESP-IDF component: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

Why not make it installed by default?

negativekelvin, 2018-07-03:
Noooo keep it away from me!!!

weyoui, 2018-07-04:
me21, I think it would be better to develop and manage ESP-Arduino and ESP-IDF separately. Tow projects have each own mission, I believe.  It looks ESP-Arduino pursuits Arduino compatibility using Arduino API but ESP-IDF does not. ESP-IDF is trying to do maximizing usage on FreeRTOS and use its benefits - several tasks are running at the same time.

igrr, 2018-11-16:
Sorry for the late response on this! Including Arduino-esp32 into ESP-IDF as a component would complicate licensing and also add more test items to ESP-IDF release flow, so for now we don't think this is a workable option. We are however working on improvements to installation flow of such external components.

closed 2018-11-16 by igrr