espressif/esp-idf #2143: Can't flash custom partition table without factory app

thecashit, 2018-07-03:
Hello!

Tried to make firmware with ESP-IDF Prerelease v3.1-beta1 and funded that i can't flash firmware in custom partition without "factory" app.
All seems to be ok and flashing started, but when i removed line with factory app definition i have following message:
"esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: /Users/thecashit/Documents/LOOK.in/Firmware/Source/build/bootloader/bootloader.bin"

My app have 2 OTA partitions without factory one and with previous versions of IDF it was successfully flashed.

mahavirj, 2018-07-03:
@thecashit 

Fixed in https://github.com/espressif/esp-idf/commit/6b5a15e3da93cd4ef94bee60d89c1e254c79bc21

thecashit, 2018-07-03:
Ok, thnx!

closed 2018-07-03 by thecashit