espressif/esp-idf #2144: WDT reset; esp_cpu_stall at ../cpu_util.c:24, .check_dport_access_end at dport_panic_highint_hdl.S:181

mws-rmain, 2018-07-03:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [n/a]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version: [f198339] (esp-idf v3.1 beta)
- Development Env:		[Eclipse(/make)]
- Operating System:		[Ubuntu]
- Power Supply:         [external 3.3V]

## Problem Description

Getting random WDT resets, consistently pointing to 'check_dport_access_end' as shown in Debug logs below.

- Running dual cores, 
- BLE (ONLY) active on PRO_CPU,
- FreeRTOS running on both cores
- SPI active as master (driving an LCD via u8g2 lib)
- UART0 used for console (ESP_LOGx diagnostics)
- UART1 communicating with a GPS device.

NOTE: My code includes a 'simulation mode' where the stream of GPS messages is simulated (sent as a message, triggered by a timer so functions can be tested without a GPS device present). I've ONLY seen this failure with a physical GPS connected, so I suspect there is some correlation with UART interrupts. 

When operating with a UART:
- a UART driver is installed
- in a separate task, characters from uart_read_bytes() are buffered until an 'end of line' ('\r'), then null-terminated, and sent to a message queue for processing.

### Code to reproduce this issue
Gist available [here](https://gist.github.com/mws-rmain/4bf77f93968e2c62ca3078c76bb1e19d)

## Debug Logs
```
W (5) boot: PRO CPU has been reset by WDT.
W (5) boot: WDT reset info: PRO CPU PC=0x400830a1
0x400830a1: esp_cpu_stall at /home/mainr/esp/esp-idf/components/soc/esp32/cpu_util.c:24

W (5) boot: WDT reset info: APP CPU PC=0x40081a43
0x40081a43: .check_dport_access_end at /home/mainr/esp/esp-idf/components/esp32/dport_panic_highint_hdl.S:181

I (9) boot: ESP-IDF v3.1-beta1-41-gf198339-dirty 2nd stage bootloader
I (17) boot: compile time 11:37:53
```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2160720/sdkconfig.txt)

alirezza, 2018-09-29:
Hi
The same issue when working with UART.
Any suggestion ?

alex00971, 2018-09-30:
One interesting thing,
This issue exists on v3.1 , [22da5f6]  
But fixed on release/v3.1 [bdb65ed] (26 days later)
I check all commits between these two version, there was no working on UART.
Just one on ['feature/light_sleep_gpio_uart_wakeup' ] .

igrr, 2018-09-30:
This was fixed in release v3.1 branch as part of https://github.com/espressif/esp-idf/commit/889c80ec1ca492647769d9348bd58b3159f900c7 (by updating libphy.a)

mws-rmain, 2018-10-01:
@alex00971 The best I can suggest currently is to run single core (in '/component config/freeRTOS/', enable 'Run FreeRTOS only on first core' (FREE_RTOS_UNICORE=y)).
I'm not certain this is required, but recommend that you ensure any calls to start a task either don't specify a core, or specify 'core 0', which will be the core running FreeRTOS in unicore mode.

I haven't tested my code with dual core build, against any recent ESP_IDF releases, but will when I get some time. It would be great if [ [ bdb65ed ](https://github.com/espressif/esp-idf/commit/bdb65ed066b008fbd2c4a28806b397dce3b2f5c1) ] fixed it, as you suggest.
If not, looking forward to release v3.1.1.

projectgus, 2019-03-01:
This fix was released in v3.1.1, sorry the issue wasn't updated earlier.

closed 2019-03-01 by projectgus