espressif/esp-idf #2145: ULP example breaks on first debug output

barde, 2018-07-03:
- [✔️] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [✔️] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [✔️] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1 die]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
fdee0b64
- Development Env:		[Make]
- Operating System:		[Debian Stable]
- Power Supply:         [USB]

## Problem Description

The ULP demo does not run. It crashes on writing the first string on the serial debug.

I tried it with two other 3rd party ESP32 modules and got the same behaviour.

### Expected Behavior

Running ULP demo.

### Actual Behavior

First line does not get printed on the debug console. ESP32 is not sleeping.

### Steps to repropduce

1. cd ~/esp/esp-idf/examples/system/ulp/
2. make menuconfig for setting port
3. make flash monitor

## Debug Logs

```
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4628
load:0x40078000,len:7304
load:0x40080000,len:7016
0x40080000: _iram_start at /home/debar/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080344
0x40080344: _UserExceptionVector at /home/debar/esp/esp-idf/components/freertos/xtensa_vectors.S:570

Not ULP w
```

igrr, 2018-07-04:
Hi @barde, thanks for reporting this.
The example actually works (you can click on the button connected to GPIO0 a few times and see that every 5 pulses the chip will wake up), just the message printed prior to entering deep sleep is not fully printed. Will fix that asap.

StuckInLoop, 2018-07-07:
yes, I am experiencing the same issue,  the core functions seems to be working, but the console output  e.g. printf("debug message....")  seems to be cut off after the first few words. and it seems to be on all the ULP examples which I have tested.

closed 2018-07-28 by igrr