espressif/esp-idf #2148: Document compiling for PSRAM

OtherCrashOverride, 2018-07-04:
The procedure for compiling an esp-idf project to be executed from PSRAM is not stated.  The intent is to load the binary into PSRAM instead of flash.  The mechanism of getting the binary into PSRAM is not relevant.  Only ensuring the code is compiled to execute from the PSRAM memory address range is required.

igrr, 2018-07-04:
> execute from the PSRAM memory address range is required

Hi @OtherCrashOverride, where did you find information that ESP32 can execute code from PSRAM?

OtherCrashOverride, 2018-07-04:
It was stated as such in official docs.  I will recheck if I can find the source again.  Most likely it was the Technical Reference Manual.

OtherCrashOverride, 2018-07-04:
https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf
page 7

After re-reading it, it does not explicitly state code execution is possible from PSRAM:
```
Both ESP32-WROVER and ESP32-WROVER-I integrate a 4 MB of external SPI flash. The 4-MB SPI flash can
be memory-mapped onto the CPU code space, supporting 8, 16 and 32-bit access. Code execution is sup-
ported.
In addition to the 4 MB SPI flash, ESP32-WROVER and ESP32-WROVER-I also integrate an 8 MB PSRAM for
more memory space.
```

igrr, 2018-07-04:
Yes, it's a bit vague but I don't see how this can be interpreted as a fact that code execution from PSRAM is supported. I'll ask my documentation colleagues to make it more explicit that it isn't. Thanks for pointing out this part.

OtherCrashOverride, 2018-07-04:
If code execution is not possible from PSRAM, please explicitly state it in this issue for future reference and close the thread.  Thank you for your attention to the matter.

OtherCrashOverride, 2018-07-04:
After further consideration, it would still be beneficial to document changing the link address.  The scenario is writing code to a flash location and memory mapping that to a known address for execution.

negativekelvin, 2018-07-04:
The reason you don't have to change the link address is because of memory mapping.  You just select a partition to boot from.

OtherCrashOverride, 2018-07-04:
Changing the boot partition is not the goal of this issue.  Compiling for an execution address other than the default is.

projectgus, 2020-07-14:
> The scenario is writing code to a flash location and memory mapping that to a known address for execution.

Sorry noone got back to you about this follow-up question. We don't support this in ESP-IDF, you'd need to modify the ESP-IDF linker script and place an "entry point" section in a well-defined point to make this work.

ESP-IDF (and ESP32 bootloader) encodes an entry point address into the [.bin file header](https://github.com/espressif/esptool/wiki/Firmware-Image-Format#file-header).

If the application is to have some kind of "applet" that runs alongside another ESP-IDF app (so there are two sets of flash cache mappings, etc) - unfortunately we don't have any plans to support this in ESP-IDF.

closed 2020-07-14 by projectgus