espressif/esp-idf #2149: backtrace in make monitor

chegewara, 2018-07-04:
Hi @projectgus 
Backtrace in make monitor sucks completely.Is lines number random value?
```
0x4013e0cd: _tp_event_handler at Z:/msys32/home/imper/alexa/components/web_radio/web_radio.c:284

0x401c416f: http_dispatch_event at Z:/msys32/home/imper/esp/esp-idf/components/esp_http_client/esp_http_client.c:981

0x4013dcd5: http_get_task at Z:/msys32/home/imper/alexa/components/web_radio/web_radio.c:284
```
Now with gdb:
```
(gdb) list *0x4013e0cd
0x4013e0cd is in _tp_event_handler (Z:/msys32/home/imper/alexa/components/web_radio/web_radio.c:123).
118
119                     }
120                 }
121                 if(xEventGroupGetBits(group) & PLAYER_PLAYING_STATE)
122                 {
123                     audio_stream_consumer(evt->data, evt->data_len, evt->user_data);
124                     radio_config->bytes_consumed += evt->data_len;
125                 }
126                 break;
127             }
(gdb) list *0x4013dcd5
0x4013dcd5 is in http_get_task (Z:/msys32/home/imper/alexa/components/web_radio/web_radio.c:207).
202             config = config_queue[current_playing];
203     // todo add PlaybackStarted event
204             ESP_LOGW(TAG, "%d: http_client_get state: %s\n", __LINE__, player_activity_state);
205             // blocks until end of stream
206             xEventGroupSetBits(group, CONTENT_FOREGROUND_BIT);
207             http_perform_as_stream_reader(alexa_session);
208             // xEventGroupClearBits(group, PLAYER_PLAYING_STATE);
209             xEventGroupSetBits(group, PLAYER_FINISHED_STATE);
210
211             // if(content_type != MPEG_URL && content_type != MIME_UNKNOWN)
(gdb) list *0x401c416f
0x401c416f is in http_dispatch_event (Z:/msys32/home/imper/esp/esp-idf/components/esp_http_client/esp_http_client.c:158).
153         if (client->event_handler) {
154             event->event_id = event_id;
155             event->user_data = client->user_data;
156             event->data = data;
157             event->data_len = len;
158             return client->event_handler(event);
159         }
160         return ESP_OK;
161     }
162
```

projectgus, 2018-07-05:
Yes, `idf_monitor.py` calls `xtensa-esp32-elf-addr2line` and there is a bug somewhere in xtensa addr2line which gets line numbers wrong in most object files.

I tried changing `idf_monitor.py` to use gdb for line number lookups a while back, but got stuck finding a good solution for something else - I think it was to do with inline functions.

We may be able to fix the bug in addr2line, instead.

chegewara, 2018-07-05:
@projectgus With all respect to you i have, this post on forum costs me few hours tracing because of this bug. I didnt use monitor to backtrace errors until ive read this:
https://esp32.com/viewtopic.php?f=13&t=6156&start=10#p26637

I dont blame you, its my fault i didnt realized earlier its something wrong until it kept telling me that error is from commented line.

projectgus, 2018-07-05:
Sorry about that. I actually thought I'd added a note in the Monitor docs that the line numbers are sometimes wrong, but I see now that there is no such note.

Discussing this internally just now, there's some upstream debugging of it [in this issue](https://github.com/jcmvbkbc/binutils-gdb-xtensa/issues/5#issuecomment-355585352). If it doesn't look like it will be fixed soon then I'll document the failing (or try gdb again).

chegewara, 2018-07-05:
Like i said, its not your fault. I didnt even open that link to monitor documentation, so even if you would mention about line numbers issue i wouldnt know that. I just saw you recommend to use it and im using it.
New lesson every day.

chegewara, 2018-10-28:
@projectgus 
I dont know if something has been done with this, but it seems to be fixed (at least i think so after few app crashes i had).

projectgus, 2018-10-29:
@chegewara Did you move to the (optional/preview) gcc 8 toolchain? This bug is fixed in the gcc 8 toolchain's binutils.

chegewara, 2018-10-29:
@projectgus I dont think so, i mean i am working on master commit but i didnt update toolchain for quite a long time. This is my gcc version:
`gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5)`

chegewara, 2018-11-04:
@chegewara The last my report that addressing works is accurate only when i work with plain C code, when i am using c++ then addressing still is not accurate.

igrr, 2019-04-12:
Preview release of the toolchain which includes updated binutils version with a fix for this issue is now available, see https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p41667.

closed 2019-06-10 by igrr