espressif/esp-idf #2151: Error in set up of the environment for Wrover Kit in Eclipse.

AmbimatElectronics, 2018-07-05:
## Environment

- Development Kit:     

 [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
Ans:-  ESP32-Wrover-Kit  

- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]  
Ans:-  may be V3 (Black color board)

- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
Ans:-  ESP32-Wrover

- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268

- Development Env:		[Eclipse]
Ans:- Eclipse Mars/Neon/Juno

- Operating System:		[Windows]
Ans:- Windows 64-bit

- Power Supply:         [USB]
Ans:- USB 2.0 cable
## Problem Description
Ans:- 
1) Error while launching command: xtensa-esp32-elf-gdb --version (Eclipse --Juno)
2) local host:3333: the system tried to join a drive to a directory on a joined drive.(Eclipse -- Mars,neon,oxygen).

//Detailed problem description goes here.
Ans:- please find attachment
[image]
[image]

igrr, 2018-07-05:
Please check that OpenOCD is running and port 3333 is not blocked by the firewall.

AmbimatElectronics, 2018-07-05:
Hello,

The port is not blocked.

Not sure how to check if the OpenOCD is running or not.

Neel.

igrr, 2018-07-09:
Hi @AmbimatElectronics, to check whether OpenOCD is running, i think you can try using Task Manager. Also you can open the command line window where you started OpenOCD, and check that it has not crashed or exited; the output should look like:

```
Open On-Chip Debugger 0.10.0-dev (2018-06-05-14:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
```

Running OpenOCD is described here: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/index.html#configuring-esp32-target. This is linked from step 12 in the guide about using Eclipse for debugging (https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse).

AmbimatElectronics, 2018-07-09:
Thanks Ivan.

We were setting everything up in Windows but it looks like the suggested OS is Linux.

Is there a specific Ubuntu version that you know your development environment works best with?

What about the version of Eclipse?

Do let me know.

Neel.

closed 2018-07-24 by AmbimatElectronics