espressif/esp-idf #2152: Fail to connect wpa2 peap

andrefilipin, 2018-07-05:
## Environment

- Development Kit:      ESP32-bit
- IDF version f198339e
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:        USB

## Problem Description

Fail to connect wpa2 peap

### Expected Behavior
Connect succeful
### Actual Behavior
Fail to connect
### Steps to repropduce

1. download esp-idf
2. copy wps example
3. make menuconfig and set username and password
4. make flash monitor

### Code to reproduce this issue

```cpp

#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_EAP_METHOD CONFIG_EAP_METHOD

#define EXAMPLE_EAP_ID CONFIG_EAP_ID
#define EXAMPLE_EAP_USERNAME CONFIG_EAP_USERNAME
#define EXAMPLE_EAP_PASSWORD CONFIG_EAP_PASSWORD

/* Constants that aren't configurable in menuconfig */
/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

/* Constants that aren't configurable in menuconfig */
#define EAP_PEAP 1
#define EAP_TTLS 2

static const char *TAG = "example";

/* CA cert, taken from wpa2_ca.pem
   Client cert, taken from wpa2_client.crt
   Client key, taken from wpa2_client.key

   The PEM, CRT and KEY file were provided by the person or organization
   who configured the AP with wpa2 enterprise.

   To embed it in the app binary, the PEM, CRT and KEY file is named
   in the component.mk COMPONENT_EMBED_TXTFILES variable.
*/
extern uint8_t ca_pem_start[] asm("_binary_wpa2_ca_pem_start");
extern uint8_t ca_pem_end[]   asm("_binary_wpa2_ca_pem_end");
extern uint8_t client_crt_start[] asm("_binary_wpa2_client_crt_start");
extern uint8_t client_crt_end[]   asm("_binary_wpa2_client_crt_end");
extern uint8_t client_key_start[] asm("_binary_wpa2_client_key_start");
extern uint8_t client_key_end[]   asm("_binary_wpa2_client_key_end");

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
[... 751 lines cut ...]
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

negativekelvin, 2018-07-05:
> Received alert 2:20

fatal, bad_record_mac

andrefilipin, 2018-07-05:
@negativekelvin thanks for your comment, 
how can you "translate" "Received alert 2:20" to "fatal, bad_record_mac"?

I'll ask the network owner with this information,

thanks

XinDeng11, 2018-07-17:
Hi @andrefilipin, I saw you mark down the certificate check, it will jump the certificate check to build TLS tunnel. However, it only effect when the server don't check ca cerificate also, have you disable the ca certificate in server? The configure is in the hostapd.conf

andrefilipin, 2018-07-17:
Hello @XinDeng11, 
I'dont have access to server, but I know it's not a freeradius.
When I connect's with my mac it ask to accept the cert, I expect the same behavior on esp, there is a way to do it?

Obs: I'm waiting the network admin to give me the cert files,

Thanks for your help

XinDeng11, 2018-07-18:
It's hard to do now, because we have some urgent thing to do. However, build a server is not hard, even you can do it in your mac book. Just follow the step in the document
[set_up_radius_server_with_hostapd.zip](https://github.com/espressif/esp-idf/files/2204181/set_up_radius_server_with_hostapd.zip)
And we have put the certificate in the floder wifi/wpa2_enterprise/main, you can try EAP method PEAP/TTLS/TLS

l8l8l, 2018-07-19:
seems you're trying to connect to a Cisco WLC using self signed certificate and without Radius behind. 
I have the exact same error.
to export the self signed certificate from WLC is not easy.
https://supportforums.cisco.com/t5/security-and-network-management/export-ssc-self-signed-certificate-from-wlc/td-p/2924297

Expecting Espressif could make a test with a Cisco WLC and find the root cause because it's very popular in many industries. (Cisco WLC+self signed certificate+no Radius)

Thanks in advance.

andrefilipin, 2018-07-19:
@l8l8l this is exactly my enviroment, I asked for the cert files to network manager but without success, 

I found a bug reported at cisco, https://quickview.cloudapps.cisco.com/quickview/bug/CSCuz66826

Do you see relation with our problem?

thanks for your help,

XinDeng11, 2018-08-03:
Hi@andrefilipin @l8l8l, after our discussion. We plan to buy the router do test. Could you offer a link of this router for us? Very thanks

negativekelvin, 2018-08-04:
That's the AP but the controller is air-ct5508-k9.

Have you tried exporting the certificate from your osx keychain?

andrefilipin, 2018-08-04:
@negativekelvin thanks for your reply,

I tried to export the certificate, but i can only export to pem extension, 

how can i export the .crt/key?

thanks

PaulFreund, 2018-08-28:
We are also in a Cisco environment with Radius server in the backend. Maybe @XinDeng11 
 can explain if this should work without client certificate? My Android is able to connect without installing any certificates but I don't know if Android automatically generates a valid self signed certificate. I documented everything at https://www.esp32.com/viewtopic.php?f=2&t=3108&start=10#p28331

PaulFreund, 2018-08-28:
@XinDeng11 As discussed in #2222 I could provide monitor mode logs once I order a compatible WiFi dongle, but please let us clarify one thing first so we are on the same page:

    1. I can't and don't need to change anything on the Wifi Router/Radius server
    2. My Android phone can connect fine with PEAP and do not validate ca certificate
    3. Should ESP32 also be able to connect without setting any certificates? (even if unsafe)

XinDeng11, 2018-08-30:
@PaulFreund 1. radius server need disable the ca certificate check if you want to connect with 32 with ca certificate
2. I am not sure why smart phone can connect
3. ESP32 can connect without cerificate after you change the radius server, this is only avildable in PEAP method

PaulFreund, 2018-08-30:
@XinDeng11 
1. I have asked IT, Radius server does not require client certificates (I don't know if they can be missing or can be self signed)
2. I have ordered a monitor mode capable WiFi interface, I will create a log of my smartphone authenticating
3. If I try to connect without setting certificate via esp_wifi_sta_wpa2_ent_set_cert_key I get:
    I (1596) wpa: EAP-TLS: Private key not configured
    E (1596) wpa: Method private structure allocated failure
I'm using WPA2_CONFIG_INIT_DEFAULT() and have identity, username and password set

XinDeng11, 2018-08-30:
Don't call esp_wifi_sta_wpa2_ent_set_ca_cert() either

PaulFreund, 2018-08-30:
This function is also not present, this is the current iteration of my code:

    void DeviceManager::connectAPEnterprise(std::string ssid, std::string identity, std::string password) {
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

        if(ESP_OK != esp_wifi_init(&cfg)) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_init failed"); }
        
        if(ESP_OK != esp_wifi_set_storage(WIFI_STORAGE_RAM)) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_set_storage failed"); }

        if(ESP_OK != esp_wifi_set_mode(WIFI_MODE_STA)) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_set_mode failed"); }

        if(ESP_OK != esp_wifi_start()) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_start failed"); }

        wifi_config_t sta_config;
        memset(&sta_config,0,sizeof(sta_config));
        strcpy((char*)(&sta_config.sta.ssid), ssid.c_str());

        if(ESP_OK != esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config)) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_set_config failed"); }

        esp_wifi_sta_wpa2_ent_set_identity((const unsigned char*)identity.c_str(), identity.length());
        esp_wifi_sta_wpa2_ent_set_username((const unsigned char*)identity.c_str(), identity.length());
        esp_wifi_sta_wpa2_ent_set_password((const unsigned char*)password.c_str(), password.length());

        esp_wpa2_config_t wpa_config = WPA2_CONFIG_INIT_DEFAULT();
        esp_wifi_sta_wpa2_ent_enable(&wpa_config);

        ESP_LOGI(LOGTAG, "[STA Enterprise] Wifi start");

        if(ESP_OK != esp_wifi_connect()) { ESP_LOGE(LOGTAG, "[STA] esp_wifi_connect failed"); }
    }

Which yields at runtime (Log level Verbose for wifi and wpa):

    I (599) wifi: wifi driver task: 3ffcddc4, prio:23, stack:3584, core=0
    I (599) wifi: wifi firmware version: 633012a
    I (599) wifi: config NVS flash: enabled
    I (609) wifi: config nano formating: disabled
    I (629) wifi: Init dynamic tx buffer num: 32
    I (629) wifi: Init data frame dynamic rx buffer num: 32
    I (629) wifi: Init management frame dynamic rx buffer num: 32
    I (629) wifi: Init static rx buffer size: 1600
    I (629) wifi: Init static rx buffer num: 10
    I (639) wifi: Init dynamic rx buffer num: 32
    I (699) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
    I (699) wifi: mode : sta (XX:XX:XX:XX:XX:XX)
    I (699) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

    I (819) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
    I (1489) wifi: state: init -> auth (b0)
    I (1509) wifi: state: auth -> assoc (0)
    I (1519) wifi: state: assoc -> run (10)
    I (1519) wpa: wpa2_task prio:2, stack:6656

    I (1589) wpa: EAP-TLS: Private key not configured
    E (1589) wpa: Method private structure allocated failure

    I (1639) wpa: >>>>>wpa2 FAILED

XinDeng11, 2018-08-30:
Maybe you can try to invoke esp_wifi_start() after wpa2_ent_enable

PaulFreund, 2018-08-30:
Already tried because I was not sure how initializes what but it does not make a difference. Only thing I can tell is that sometimes after rebooting instead of the wpa2 FAILED it hangs at:

    I (599) wifi: wifi driver task: 3ffcddc4, prio:23, stack:3584, core=0
    I (599) wifi: wifi firmware version: 633012a
    I (599) wifi: config NVS flash: enabled
    I (609) wifi: config nano formating: disabled
    I (629) wifi: Init dynamic tx buffer num: 32
    I (629) wifi: Init data frame dynamic rx buffer num: 32
    I (629) wifi: Init management frame dynamic rx buffer num: 32
    I (629) wifi: Init static rx buffer size: 1600
    I (639) wifi: Init static rx buffer num: 10
    I (639) wifi: Init dynamic rx buffer num: 32
    I (699) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
    I (699) wifi: mode : sta (XX:XX:XX:XX:XX:XX)
    I (719) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

    I (839) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
    I (1509) wifi: state: init -> auth (b0)
    I (1509) wifi: state: auth -> assoc (0)
    I (2509) wifi: state: assoc -> init (400)
    I (2519) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1

XinDeng11, 2018-08-30:
try again, I think it is another issue which we have fixed in the latest code

PaulFreund, 2018-08-30:
Tested with newest master esp-idf cc8ad721f98ffbc7953ece70616c07422b58e06b with updated submodules, wifi firmware now d9df943:

    I (609) wifi: wifi driver task: 3ffcdd34, prio:23, stack:3584, core=0
    I (609) wifi: wifi firmware version: d9df943
    I (609) wifi: config NVS flash: enabled
    I (609) wifi: config nano formating: disabled
    I (629) wifi: Init dynamic tx buffer num: 32
    I (629) wifi: Init data frame dynamic rx buffer num: 32
    I (629) wifi: Init management frame dynamic rx buffer num: 32
    I (639) wifi: Init static rx buffer size: 1600
    I (639) wifi: Init static rx buffer num: 10
    I (649) wifi: Init dynamic rx buffer num: 32
    I (709) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
    I (709) wifi: mode : sta (XX:XX:XX:XX:XX:XX)
    I (709) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

    I (829) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
    I (1499) wifi: state: init -> auth (b0)
    I (1509) wifi: state: auth -> assoc (0)
    I (1529) wifi: state: assoc -> run (10)
    I (1529) wpa: wpa2_task prio:2, stack:6656

    I (1599) wpa: EAP-TLS: Private key not configured
    E (1599) wpa: Method private structure allocated failure

    I (1669) wpa: >>>>>wpa2 FAILED

XinDeng11, 2018-08-30:
It looks still need certificate, I have no idea now, because it still remind you should set private key, maybe we can find something from the air packets

negativekelvin, 2018-08-30:
@XinDeng11 if client certificate/key is not configured then eap-tls should not be registered as an available peer method?

@PaulFreund maybe try building without -DEAP_TLS ? https://github.com/espressif/esp-idf/blob/aaf12390eb14b95589acd98db5c268a2e56bb67e/components/wpa_supplicant/component.mk#L4

XinDeng11, 2018-08-30:
you can try, but it may crash as I remember

PaulFreund, 2018-08-30:
@negativekelvin This is without -DEAP_TLS:

    I (295) wifi: wifi driver task: 3ffcdd34, prio:23, stack:3584, core=0
    I (295) wifi: wifi firmware version: d9df943
    I (295) wifi: config NVS flash: enabled
    I (295) wifi: config nano formating: disabled
    I (315) wifi: Init dynamic tx buffer num: 32
    I (325) wifi: Init data frame dynamic rx buffer num: 32
    I (325) wifi: Init management frame dynamic rx buffer num: 32
    I (325) wifi: Init static rx buffer size: 1600
    I (325) wifi: Init static rx buffer num: 10
    I (335) wifi: Init dynamic rx buffer num: 32
    I (395) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
    I (395) wifi: mode : sta (XX:XX:XX:XX:XX:XX)
    I (395) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

    I (515) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
    I (1375) wifi: state: init -> auth (b0)
    I (1385) wifi: state: auth -> assoc (0)
    I (1415) wifi: state: assoc -> run (10)
    I (1415) wpa: wpa2_task prio:2, stack:6656

    D (1545) wpa: TLS: using phase1 config options
    D (1545) wpa: SSL: Received packet(len=6) - Flags 0x20
    D (1545) wpa: EAP-PEAP: Start (server ver=0, own ver=1)
    D (1555) wpa: EAP-PEAP: Using PEAP version 0
    D (1555) wpa: TLSv1: Send ClientHello
    D (1555) wpa: SSL: 62 bytes left to be sent out (of total 62 bytes)
    D (1785) wpa: SSL: Received packet(len=1296) - Flags 0xc0
    D (1785) wpa: SSL: TLS Message Length: 3633
    I (1785) wpa: SSL: Need 2347 bytes more input data
    D (1785) wpa: SSL: Building ACK (type=25 id=5 ver=0)

    D (1875) wpa: SSL: Received packet(len=1296) - Flags 0x40
    I (1875) wpa: SSL: Need 1057 bytes more input data
    D (1875) wpa: SSL: Building ACK (type=25 id=6 ver=0)

    D (1935) wpa: SSL: Received packet(len=1063) - Flags 0x00
    D (1935) wpa: TLSv1: Received content type 22 version 3.1 length 3628
    D (1935) wpa: TLSv1: Received ServerHello
    D (1945) wpa: TLSv1: Using TLS v1.0
    D (1945) wpa: TLSv1: Selected cipher suite: 0x0035
    D (1945) wpa: TLSv1: Received Certificate (certificate_list len 3536)
    D (1955) wpa: TLSv1: Certificate 0 (len 1908)
    D (1955) wpa: X509: Version X.509v3
    D (1965) wpa: X509: serialNumber 53251
    D (1965) wpa: X509: issuer CENSORED
    D (1975) wpa: X509: Validity: notBefore: 0 notAfter: 0
    D (1975) wpa: X509: subject CENSORED
    D (1995) wpa: X509: Extension: extnID=2.5.29.14 critical=0
    D (1995) wpa: ASN.1: Extended tag data: 0x04
    D (2005) wpa: X509: Extension: extnID=2.5.29.15 critical=255
    D (2005) wpa: X509: KeyUsage 0x5
    D (2005) wpa: X509: Extension: extnID=2.5.29.35 critical=0
    D (2015) wpa: X509: Extension: extnID=2.5.29.31 critical=0
    D (2015) wpa: X509: Extension: extnID=1.3.6.1.5.5.7.1.1 critical=0
    D (2025) wpa: X509: Extension: extnID=1.3.6.1.4.1.311.21.7 critical=0
    D (2035) wpa: X509: Extension: extnID=2.5.29.37 critical=0
    D (2035) wpa: X509: Extension: extnID=2.5.29.32 critical=0
    D (2045) wpa: X509: Extension: extnID=1.3.6.1.4.1.311.21.10 critical=0

    Stack smashing protect failure!
    [...]

unfortunately crashes but I can actually see negotiation going on. Also leaving EAP_TLS inside and removing the call eap_peer_tls_register leads to a crash.

@XinDeng11 

    I (1599) wpa: EAP-TLS: Private key not configured

This happens in eap_tls_init of eap_tls.c inside wpa_supplicant. I did not have a lot of time to go through wpa_supplicant and I don't know the terminiology but what irritates me is that I can only find one call to eap_method* init functions in eap_peap.c  line 730 which would lead to a wpa_printf a few lines later. Should the next method be tried then?

    E (1599) wpa: Method private structure allocated failure

This looks like an actual error and I can't find the text inside wpa_supplicant which suggests it is actually inside the wifi library, is this correct?

PaulFreund, 2018-08-30:
PS: Here is the stack trace

    abort() was called at PC 0x400d3508 on core 1
    0x400d3508: __stack_chk_fail at /mnt/c/esp/esp-idf/components/esp32/stack_check.c:36

    Backtrace: 0x4008f48c:0x3ffd4ea0 0x4008f661:0x3ffd4ec0 0x400d3508:0x3ffd4ee0 0x400f4a39:0x3ffd4f00 0x400f4af4:0x3ffd4fd0 0x400f18d1:0x3ffd5020 0x400f5cea:0x3ffd5040 0x400f62c5:0x3ffd50a0 0x400ef869:0x3ffd50d0 0x400ed0ee:0x3ffd5120 0x400ed149:0x3ffd5160 0x400e9c33:0x3ffd5180 0x400e9e25:0x3ffd51c0 0x400ec362:0x3ffd51e0 0x4012cc2a:0x3ffd5260 0x4012cebd:0x3ffd5290
    0x4008f48c: invoke_abort at /mnt/c/esp/esp-idf/components/esp32/panic.c:660

    0x4008f661: abort at /mnt/c/esp/esp-idf/components/esp32/panic.c:660

    0x400d3508: __stack_chk_fail at /mnt/c/esp/esp-idf/components/esp32/stack_check.c:36

    0x400f4a39: x509_parse_tbs_certificate at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/x509v3.c:1446

    0x400f4af4: x509_certificate_parse at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/x509v3.c:1555

    0x400f18d1: tls_parse_cert at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_common.c:164

    0x400f5cea: tls_process_certificate at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c:336

    0x400f62c5: tlsv1_client_process_handshake at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c:958

    0x400ef869: tlsv1_client_handshake at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client.c:801

    0x400ed0ee: tls_connection_handshake2 at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tls_internal.c:568

    0x400ed149: tls_connection_handshake at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tls_internal.c:568

    0x400e9c33: eap_tls_process_input at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c:482

    0x400e9e25: eap_peer_tls_process_helper at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c:630

    0x400ec362: eap_peap_process at /mnt/c/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_peap.c:1097

    0x4012cc2a: eap_sm_process_request at ??:?

    0x4012cebd: wpa2Task at ??:?

negativekelvin, 2018-08-30:
Well stack overflow is easy to fix, size should be in menuconfig.  You could try nano printf option too.

XinDeng11, 2018-08-31:
I told you it will crash because the TLS  tunnel can not finish after close EAP-TLS macro, the best way is try to find somthing in air packet now

negativekelvin, 2018-08-31:
@PaulFreund what if you replace these two lines 
https://github.com/espressif/esp-idf/blob/a3c43251b4b37bc3ec105a10f2e1413b14420da3/components/wpa_supplicant/src/wpa2/eap_peer/eap.c#L204-L205

with just

eap_peer_tls_register();

PaulFreund, 2018-08-31:
I have some logs now, unfortunately I feel they are incomplete, I tried to fix the channel but it seems I still miss a lot. I use a Alfa Networks AWUS036NHA with Kali Linux and Wireshark. Monitor mode via airmon-ng fixed to the channel of the access point. 

If you need the full logs we have to talk about a NDA and do this in private but I can share some screenshots of the communication with my phone and the one with ESP32

In every log I filtered for the MAC address of the STA in any of the address fields.

1. Those logs are with most recent master and the exact code I posted above (result is I (1599) wpa: EAP-TLS: Private key not configured). The log always ends after this
[image]

[image]

[image]

2. Those logs are with recent master and EAP_TLS disabled, crashing
[image]

[image]

3. Here are logs from my Android phone (connection working)
[image]

[image]

[image]

@XinDeng11 please tell me any additional information you might need and which packages we need the details (and maybe what details

@negativekelvin I don't think the reason is a too small stack but rather a buffer overflow. Also I can not change the stack size of the wifi task in menuconfig.

negativekelvin, 2018-09-01:
facepalm

https://github.com/espressif/esp-idf/blob/a557e8cc14880717f242918a754cbc37d85e036a/components/wpa_supplicant/src/wpa2/tls/x509v3.c#L546-L547

https://github.com/espressif/esp-idf/blob/a557e8cc14880717f242918a754cbc37d85e036a/components/wpa_supplicant/src/wpa2/tls/x509v3.c#L563-L564

negativekelvin, 2018-09-01:
https://github.com/espressif/esp-idf/pull/2354

XinDeng11, 2018-09-03:
@PaulFreund Could you share me the original wireshark packet to me?

negativekelvin, 2018-09-03:
@XinDeng11 this is fixed, check the PR.  BTW PEAP works fine without the EAP_TLS flag but now everything should work.

PaulFreund, 2018-09-03:
@negativekelvin Thank you so much for solving this! Good catch!

@XinDeng11 Issue really is resolved, I don't need any more assistance apart from merging the changes

andrefilipin, 2018-09-03:
@XinDeng11 @negativekelvin thanks for support, I really appreciate it.
I'll test in my enviroment and back soon
Thanks again

andrefilipin, 2018-09-04:
@XinDeng11 @negativekelvin i'm still having problem to connect
verbose log:
```I (756) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1736) wifi: state: init -> auth (b0)
I (1736) wifi: state: auth -> assoc (0)
I (1736) wifi: state: assoc -> run (10)
I (1746) wpa: wpa2_task prio:2, stack:6656

D (1756) wpa: TLS: using phase1 config options
D (1756) wpa: SSL: Received packet(len=6) - Flags 0x21
D (1756) wpa: EAP-PEAP: Start (server ver=1, own ver=1)
D (1756) wpa: EAP-PEAP: Using PEAP version 1
D (1766) wpa: TLSv1: Send ClientHello
D (1766) wpa: SSL: 62 bytes left to be sent out (of total 62 bytes)
D (1776) wpa: SSL: Received packet(len=1200) - Flags 0xc1
D (1776) wpa: SSL: TLS Message Length: 1248
I (1786) wpa: SSL: Need 58 bytes more input data
D (1786) wpa: SSL: Building ACK (type=25 id=4 ver=1) 

D (1796) wpa: SSL: Received packet(len=64) - Flags 0x01
D (1796) wpa: TLSv1: Received content type 22 version 3.1 length 74
D (1806) wpa: TLSv1: Received ServerHello
D (1806) wpa: TLSv1: Using TLS v1.0
D (1816) wpa: TLSv1: Selected cipher suite: 0x002f
D (1816) wpa: TLSv1: Received content type 22 version 3.1 length 1155
D (1826) wpa: TLSv1: Received Certificate (certificate_list len 1151)
D (1826) wpa: TLSv1: Certificate 0 (len 1145)
D (1836) wpa: X509: Version X.509v3
D (1836) wpa: X509: serialNumber 1199223
D (1846) wpa: X509: issuer O=Cisco Systems, CN=Cisco Manufacturing CA
D (1846) wpa: X509: Validity: notBefore: 0 notAfter: 0
D (1856) wpa: X509: subject C=US, ST=California, L=San Jose, O=Cisco Systems, CN=AIR-CT5508-K9-XXX/emailAddress=support@cisco.com
D (1866) wpa: X509: Extension: extnID=2.5.29.15 critical=0
D (1876) wpa: X509: KeyUsage 0x5
D (1876) wpa: X509: Extension: extnID=2.5.29.14 critical=0
D (1876) wpa: X509: Extension: extnID=2.5.29.35 critical=0
D (1886) wpa: X509: Extension: extnID=2.5.29.31 critical=0
D (1896) wpa: X509: Extension: extnID=1.3.6.1.5.5.7.1.1 critical=0
D (1896) wpa: X509: Extension: extnID=1.3.6.1.4.1.311.20.2 critical=0
D (1906) wpa: X509: Version X.509v3
D (1906) wpa: X509: serialNumber XXX
D (1916) wpa: X509: issuer O=Cisco Systems, CN=Cisco Manufacturing CA
D (1916) wpa: X509: Validity: notBefore: 0 notAfter: 0
D (1926) wpa: X509: subject C=US, ST=California, L=San Jose, O=Cisco Systems, CN=AIR-CT5508-K9-649ef3bf2d00/emailAddress=support@cisco.com
D (1936) wpa: X509: Extension: extnID=2.5.29.15 critical=0
D (1946) wpa: X509: KeyUsage 0x5
D (1946) wpa: X509: Extension: extnID=2.5.29.14 critical=0
D (1946) wpa: X509: Extension: extnID=2.5.29.35 critical=0
D (1956) wpa: X509: Extension: extnID=2.5.29.31 critical=0
D (1966) wpa: X509: Extension: extnID=1.3.6.1.5.5.7.1.1 critical=0
D (1966) wpa: X509: Extension: extnID=1.3.6.1.4.1.311.20.2 critical=0
D (1976) wpa: X509: Validate certificate chain
D (1976) wpa: X509: 0: C=US, ST=California, L=San Jose, O=Cisco Systems, CN=AIR-CT5508-K9-XXX/emailAddress=support@cisco.com
D (1986) wpa: X509: Did not find any of the issuers from the list of trusted certificates
D (1996) wpa: X509: Certificate chain validation disabled - ignore unknown CA issue
D (2006) wpa: X509: Certificate chain valid
D (2016) wpa: TLSv1: Received content type 22 version 3.1 length 4
D (2016) wpa: TLSv1: Received ServerHelloDone
D (2026) wpa: TLSv1: Send ClientKeyExchange
D (2206) wpa: TLSv1: Send ChangeCipherSpec
D (2206) wpa: TLSv1: Record Layer - New write cipher suite 0x002f
D (2206) wpa: TLSv1: Send Finished
D (2206) wpa: SSL: 326 bytes left to be sent out (of total 326 bytes)
D (2416) wpa: SSL: Received packet(len=17) - Flags 0x81
D (2416) wpa: SSL: TLS Message Length: 7
D (2416) wpa: TLSv1: Received content type 21 version 3.1 length 2
D (2426) wpa: TLSv1: Received alert 2:20
D (2426) wpa: SSL: No data to be sent out
D (2436) wpa: SSL: Building ACK (type=25 id=6 ver=1) 

I (2446) wpa: >>>>>wpa2 FAILED

D (2446) wpa: TLSv1: Selected cipher suite: 0x0000
D (2446) wpa: TLSv1: Record Layer - New write cipher suite 0x0000
D (2456) wpa: TLSv1: Record Layer - New read cipher suite 0x0000 

I (4616) example: ~~~~~~~~~~~
I (4616) example: IP:0.0.0.0
I (4616) example: MASK:0.0.0.0
I (4616) example: GW:0.0.0.0
I (4616) example: ~~~~~~~~~~~
I (6616) example: ~~~~~~~~~~~```

I removed -DEAP_TLS tag and applied the commit with @negativekelvin changes

@PaulFreund maybe you can help me with any tip

thanks guys

PaulFreund, 2018-09-04:
@andrefilipin For this solution don't remove -DEAP_TLS! Also don't call set cert or set key

andrefilipin, 2018-09-04:
@PaulFreund 
The error on my log is:
```
D (2426) wpa: TLSv1: Received alert 2:20
D (2426) wpa: SSL: No data to be sent out
```
and yours:
```
I (1589) wpa: EAP-TLS: Private key not configured
E (1589) wpa: Method private structure allocated failure
```
I thinking they are distinct problems and maybe require a different solution

negativekelvin, 2018-09-04:
@andrefilipin sorry, I was not sure the buffer overflow would be causing your issue.  You should definitely check with your admin that the software update that fixes the Cisco bug you linked has been installed.

negativekelvin, 2018-09-06:
@andrefilipin check https://github.com/espressif/esp-idf/issues/2381

FayeY, 2018-10-22:
Hi, this issue should have been solved in the latest master, please have a try, and feel free to reopen if the issue persists. Thanks.

closed 2018-10-22 by FayeY

chegewara, 2018-10-22:
@FayeY github is down

negativekelvin, 2018-11-09:
@ybuyankin Your server is requesting a client certificate

ybuyankin, 2018-11-09:
@negativekelvin Thanks yes I've noticed that but it's generally the same thing - it does not appear to be a problem for any other device to connect to this server except for the ESP32. We generally aim to make it compatible with any 'valid' network setup, do we? And by 'valid' I mean any setup which allows for common Win/MacOS/Android devices to connect. And they do.

ybuyankin, 2018-11-09:
Local admin said that was the guide to set up the Cisco so it should be quite common [https://www.cisco.com/c/en/us/support/docs/wireless/5500-series-wireless-controllers/115988-nps-wlc-config-000.html](url)

negativekelvin, 2018-11-10:
Debug log shows it fails abruptly without server sending any response compared to andrefilipin's log where server will send tls alert before failure.  Unfortunately failure message is generated in closed source libwpa2. Can you check server log or use sniffer?

ybuyankin, 2018-11-12:
@negativekelvin Thanks, yes I'll try to look deeper into this. My guess so far was that it fails upon sending a non-configured certificate. I've tried to uncomment the line which sets the client cert (with provided sample one) but still got almost the same results:
```
D (2636) wpa: X509: Did not find any of the issuers from the list of trusted certificates
D (2636) wpa: X509: Certificate chain validation disabled - ignore unknown CA issue
D (2646) wpa: X509: Certificate chain valid
D (2656) wpa: TLSv1: Received CertificateRequest
D (2656) wpa: TLSv1: Received ServerHelloDone
D (2656) wpa: TLSv1: Send Certificate
D (2666) wpa: TLSv1: Full client certificate chain not configured - validation may fail
D (2676) wpa: TLSv1: Send ClientKeyExchange
D (2856) wpa: TLSv1: Send CertificateVerify
D (4756) wpa: TLSv1: Send ChangeCipherSpec
D (4756) wpa: TLSv1: Record Layer - New write cipher suite 0x002f
D (4756) wpa: TLSv1: Send Finished
D (4766) wpa: SSL: 1458 bytes left to be sent out (of total 1458 bytes)
D (4766) wpa: SSL: sending 1400 bytes, more fragments will follow
D (4806) wpa: SSL: Received packet(len=6) - Flags 0x00
D (4806) wpa: SSL: 58 bytes left to be sent out (of total 1458 bytes)
I (4876) wpa: >>>>>wpa2 FAILED

```

ybuyankin, 2018-11-14:
Some observations with Wireshark so far:

1. Other successfully connected to the same AP devices are not getting CertificateRequest from server
2. Other devices are using TLS 1.2 while TLS 1.0 is selected with ESP32 (while support for TLS 1.2 is claimed).

It really bothers me as a whole thing seems to be out of control.

samca208, 2020-11-23:
Hi Tried the espressif WPA2 example https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise.
Also utilized the CA certificate, server certificate and key of my organization wit the result of wpa>>>>>>>wpa2 FAILED
Also tried the original certificates provided with the example. Obviously they had to fail as the certificates are not valid for my network but I got the same result to when I've used my system certificates. Find below the logs:

D (3930) wifi:recv auth: seq=2, status=0
I (3930) wifi:state: auth -> assoc (0)
D (3940) wifi:restart connect 1s timer for assoc
D (3940) wifi:recv assoc: type=0x10
D (3940) wifi:filter: set rx policy=6
I (3950) wifi:state: assoc -> run (10)
I (3950) wpa: wpa2_task prio:2, stack:6656

D (3950) wifi:start 30s connect timer for 4 way handshake
D (3960) wpa: WPA2: wifi->wpa2 api completed sig(0)
D (3960) wpa: WPA2: wpa2 api return, sm->state(1)
D (3970) wpa: IEEE 802.1X RX: version=2 type=0 length=50

D (3980) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=54):
D (3980) wpa: 02 00 00 32 01 01 00 32 01 00 6e 65 74 77 6f 72
D (3990) wpa: 6b 69 64 3d 53 54 4d 46 47 2c 6e 61 73 69 64 3d
D (3990) wpa: 4b 49 52 2d 43 57 49 57 4c 43 30 31 2c 70 6f 72
D (4000) wpa: 74 69 64 3d 31 33
D (4000) wpa: WPA2: wifi->wpa2 api completed sig(1)
D (4010) wpa: WPA2: wpa2 api return, sm->state(1)
D (4010) wpa: IEEE 802.1X RX: version=2 type=0 length=50

D (4020) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=54):
D (4020) wpa: 02 00 00 32 01 02 00 32 01 00 6e 65 74 77 6f 72
D (4030) wpa: 6b 69 64 3d 53 54 4d 46 47 2c 6e 61 73 69 64 3d
D (4030) wpa: 4b 49 52 2d 43 57 49 57 4c 43 30 31 2c 70 6f 72
D (4040) wpa: 74 69 64 3d 31 33
D (4040) wpa: WPA2: wifi->wpa2 api completed sig(1)
D (4050) wpa: WPA2: wpa2 api return, sm->state(1)
D (4050) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (4080) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (4100) wpa: IEEE 802.1X RX: version=2 type=0 length=4

D (4100) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=8):
D (4100) wpa: 02 00 00 04 04 ff 00 04
I (4110) wpa: >>>>>wpa2 FAILED

D (4110) wpa: WPA2: wifi->wpa2 api completed sig(1)
D (4120) wpa: WPA2: wpa2 api return, sm->state(3)
D (4120) wpa: WPA2: queue deleted
D (4120) wpa: WPA2: task deleted
D (4130) wpa: WPA2: wifi->wpa2 api completed sig(2)
D (4130) wpa: WPA2: wpa2 api return, sm->state(3)
D (4140) wpa: wpa2 eap_peer_sm_deinit: free data lock
D (4190) wifi:rsn valid: gcipher=3 ucipher=3 akm=4