espressif/esp-idf #2153: Tx Q not empty

lochoang75, 2018-07-06:
I use wifi and ble coexistence. after 1600ms  for BLE I start wifi and download data and stop wifi. It work fine but after 5-10 minutes it println out error and i can't find where it come from.
[image]

TimXia, 2018-07-08:
@lochoang75 Did it print all the time? It's just an information for debug.

lochoang75, 2018-07-08:
@TimXia it usually work normally, but some time it happen BLE still working but wifi seem not work. Can you show me the file which print this in library ?

TimXia, 2018-07-09:
@lochoang75 Can WiFi recover working somewhile after printing this log? If so, it is not an issue.

lochoang75, 2018-07-09:
Btw, does wifi and BLE advertise can work at the same time ? it's best fit to my project.

TimXia, 2018-08-13:
@snake Please advise.

TimXia, 2018-08-13:
@tianhao Please advise.

tijnkooijmans, 2018-08-19:
Same issue here, Wifi doesn’t recover until chip resets.

TianHao-Espressif, 2018-08-20:
@lochoang75, wifi and BLE advertising can work at the same time. I fixed some bugs of coexist in latest master branch and release/v3.1. It may help.

baycom, 2018-10-29:
Same problem here: My project uses BLE and WIFI at the same time and is compiled with latest esp-idf. After a couple of minutes I get the same message resulting in a complete loss of WIFI connectivity. When BLE is not being used, the problem does not occur.

tijnkooijmans, 2018-10-30:
I also have this issue still with 3.1 release (7cd75f30f275e91a68b69856d70fcae37354b665), but it only happens when I'm doing a BLE scan while the Wifi reconnects. It then never recovers and keeps spamming this error. What is the exact IDF commit that should have fixed this?

TianHao-Espressif, 2018-10-31:
@baycom , could you describe your operations of BLE and WIFI? Besides, do you enable SW_COEXIST_ENABLE in menuconfig?

@tijnkooijmans , 7cd75f seems a little old, the coexist latest update is in October. Please try the latest release/v3.1 branch. Thanks.

tijnkooijmans, 2018-10-31:
Thanks for your reply, I will try with a later version to see if it still happens.

baycom, 2018-11-01:
Tried a lot of things - it seems there is some kind of deadlock with esp_restart() failing to restart the system or hanging calls to netconn_write or the wifi tx q not empty bug . Free heap memory goes down from 120kb to about 70kb and then a lot of errors occur. In my code there does not seem to be any memory leak or mutex lockup. 
Now I end up in setting up watchdogs for my tasks that discover any hangs and call heap_caps_dump_all() which leads to a reboot:
...
`Block 0x3ffbd8b0 data size 0x0000001c bytes next block 0x3ffbd8d0
Block 0x3ffbd8d0 data size 0x00000024 bytes next block 0x3ffbd8f8
Block 0x3ffbd8f8 data size 0x00000014 bytes next block 0x3ffbd910
Block 0x3ffbd910 data size 0x00000014 bytes next block 0x3ffbd928
Block 0x3ffbd928 data size 0x00000018 bytes next block 0x3ffbd944
Block 0x3ffbd944 data size 0x00000090 bytes nextGuru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4000921f  PS      : 0x00060834  A0      : 0x80007d16  A1      : 0x3ffbb470  
A2      : 0xeb80c000  A3      : 0x48000000  A4      : 0x00000000  A5      : 0x3ffcec20  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x3ff40000  A9      : 0x00000020  
A10     : 0x00800000  A11     : 0x3ff4001c  A12     : 0x00060a20  A13     : 0x00000001  
A14     : 0x00060a23  A15     : 0x0000000b  SAR     : 0x00000005  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x4000921f:0x3ffbb470 0x40007d13:0x3ffbb490 0x40007c69:0x3ffbb4b0 0x40008148:0x3ffbb4d0 0x40094e71:0x3ffbb560 0x400d3f9f:0x3ffbb580 0x400d3fb2:0x3ffbb5a0 0x400d468a:0x3ffbb5c0 0x400d2466:0x3ffbb5f0 0x400918d9:0x3ffbb610
0x40094e71: multi_heap_dump at /xxx/esp/esp-idf-master/components/heap/multi_heap.c:713

0x400d3f9f: heap_caps_dump at /xxx/esp/esp-idf-master/components/heap/heap_caps.c:441

0x400d3fb2: heap_caps_dump_all at /xxx/esp/esp-idf-master/components/heap/heap_caps.c:448

0x400d468a: app_main at /xxx/proj/scratch/esp32/gimbal-controller/main/main.cpp:171 (discriminator 1)

0x400d2466: main_task at /xxx/esp/esp-idf-master/components/esp32/cpu_start.c:493

0x400918d9: vPortTaskWrapper at /xxx/esp/esp-idf-master/components/freertos/port.c:403

Core 1 register dump:
PC      : 0x4017d65e  PS      : 0x00060f34  A0      : 0x800d39b1  A1      : 0x3ffbc560  
0x4017d65e: esp_pm_impl_waiti at /xxx/esp/esp-idf-master/components/esp32/pm_esp32.c:474

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80091ae2  A5      : 0x3ffcec20  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d36a6  A9      : 0x3ffbc530  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80091ae2  A13     : 0x3ffcec20  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x4017d65e:0x3ffbc560 0x400d39ae:0x3ffbc580 0x40093171:0x3ffbc5a0 0x400918d9:0x3ffbc5c0
0x4017d65e: esp_pm_impl_waiti at /xxx/esp/esp-idf-master/components/esp32/pm_esp32.c:474

0x400d39ae: esp_vApplicationIdleHook at /xxx/esp/esp-idf-master/components/esp32/freertos_hooks.c:86

0x40093171: prvIdleTask at /xxx/esp/esp-idf-master/components/freertos/tasks.c:3564

0x400918d9: vPortTaskWrapper at /xxx/esp/esp-idf-master/components/freertos/port.c:403

Rebooting...`

This is really quite frustrating and any help would be appreciated.

baycom, 2018-11-01:
Maybe this is important too: the problem seems to occur only in station mode. If the esp32 is set to soft ap mode everything works as expected.

TianHao-Espressif, 2018-11-07:
@TimXia , please take a look

baycom, 2018-11-07:
If you like you can checkout my code: https://github.com/baycom/gimbal-controller

TimXia, 2019-02-01:
@baycom @tijnkooijmans The issue has been solved. Please update IDF to check it.

Alvin1Zhang, 2019-10-31:
@lochoang75 @tijnkooijmans @baycom Thanks for reporting and kind updates, would you please help try updating IDF and help share if your issue has been resolved? Thanks.

baycom, 2019-10-31:
Thank you - the problem is fixed now for my use case.

tijnkooijmans, 2019-11-01:
Also fixed here!

Alvin1Zhang, 2019-11-05:
@baycom @tijnkooijmans Thanks for the feedback, feel free to reopen if the issue still exists. Thanks.

closed 2019-11-05 by Alvin1Zhang