espressif/esp-idf #2154: PWM - frequency

Reiner1210, 2018-07-06:
<code>
ledc_timer_config_t timer;
memset(&timer, 0, sizeof(timer));
timer.freq_hz = 100;
timer.duty_resolution = 10;
timer.speed_mode = LEDC_HIGH_SPEED_MODE;
timer.timer_num = LEDC_TIMER_0;
</code>

I expected that this leads to a 100 Hz frequency, But the result is a 50 Hz frequency. Why the factor 2 ???

costaud, 2018-07-09:
I've tested your code, but the frequency seems correct according to the logic analyser.

FayeY, 2018-07-12:
@Reiner1210 , is there any update? Could you provide a test project so we can reproduce your problem?

Reiner1210, 2018-07-12:
I will retry it - tomorrow

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

Reiner1210, 2018-07-13:
I use the following
	timer.freq_hz = 1000;
	timer.duty_resolution = 10;
	timer.speed_mode = LEDC_HIGH_SPEED_MODE;
	timer.timer_num = LEDC_TIMER_0;
	uint32_t duty = (uint32_t)((float)(1<<timer.duty_resolution) * 0.25f + .5f);

This leads to the following:
[image]

xiongyumail, 2018-08-01:
Hi @Reiner1210 ,
Have you changed the frequency when config, Could you provide more code ?

Alvin1Zhang, 2018-09-25:
@Reiner1210 Could you help provide some updates? Thanks.

closed 2018-10-30 by Alvin1Zhang