espressif/esp-idf #2155: [Question] Brownout flag, Is there register address to read brownout flag after reboot?

farizadp, 2018-07-06:
Hi,

Is there register bit which is set (non-volatile) by the chip if brownout occurs? And can be read after next reboot?

Regards,
Dion

negativekelvin, 2018-07-06:
https://github.com/espressif/esp-idf/blob/c4edb2a4398aef32822b1f917157e402cc246e64/components/esp32/include/rom/rtc.h#L95

farizadp, 2018-07-09:
@negativekelvin : CMIIW. So I can call rtc_get_reset_reason() from application?

Is this flag overwritten by esp_restart_noos() ? Because, the standard rtc_brownout_isr_handler() calls esp_restart_noos().

farizadp, 2018-07-16:
Anyone can answer my question @FayeY or @negativekelvin  ? What is brownout workaround? Is there newer chip revision already?

negativekelvin, 2018-07-16:
The brownout "workaround" as I called it IS the rtc_brownout_isr_handler calling esp_restart_noos instead of returning

farizadp, 2018-07-17:
@negativekelvin :
Do you know whether it is safe to do nvs_write_* inside this isr handler?
Do you also know if there's newer chip revision for PICO-D4?

negativekelvin, 2018-07-17:
No, and writing to flash in general is probably not a good idea since flash may not operate properly during brownout.
Rev 1 is newest afaik

farizadp, 2018-09-06:
Is there update regarding this issue?
Is it now possible to get brownout reset cause when brownout occurs?

closed 2018-11-16 by igrr