espressif/esp-idf #2156: Issue with SD Webserver (ERROR reading file types)

KrrishnaShah, 2018-07-06:
My code runs ok with file name extention .txt but not with .html
	TRACE("f_open: %s\n", ff_trace((res = f_open(fil, "public/index.html", FA_WRITE))));
	if (res != FR_OK)
	{
		 if (res == FR_NO_PATH)
		 {
			TRACE("%s", ff_trace(f_mkdir("public")));
		 }
		 TRACE("f_open: %s\n", ff_trace((res = f_open(fil, "public/index.html", FA_CREATE_NEW))));
	}
	TRACE("f_write: %s\n", ff_trace(f_write(fil, test,  sizeof(test), &len)));
	TRACE("f_close: %s\n", ff_trace(f_close(fil)));

	char buffer[128] = {0};
	TRACE("f_open: %s\n", ff_trace(f_open(fil, "public/index.html", FA_READ)));
	TRACE("f_read: %s\n", ff_trace(f_read(fil, buffer,  128, &len)));
	TRACE("f_close: %s\n", ff_trace(f_close(fil)));
	TRACE("\nfoo.html: %s\n", buffer);
------------------------------------------------------------------------------------
OUTPUT:
------------------------------------------------------------------------------------
	f_open: FR_INVALID_NAME
	f_open: FR_INVALID_NAME
	f_write: FR_INVALID_OBJECT
	f_close: FR_INVALID_OBJECT
	f_open: FR_INVALID_NAME
	f_read: FR_INVALID_OBJECT
	f_close: FR_INVALID_OBJECT
	foo.html: 

**BUT with .txt file runs gr8:**
      FIL * fil = pvPortMalloc(sizeof(FIL)); uint32_t len = 0;

      FRESULT res = FR_OK;
      TRACE("f_open: %s\n", ff_trace((res = f_open(fil, "public/index.txt", FA_WRITE))));
      if (res != FR_OK)
      {
         if (res == FR_NO_PATH)
         {
            TRACE("%s", ff_trace(f_mkdir("public")));
         }
         TRACE("f_open: %s\n", ff_trace((res = f_open(fil, "public/index.txt", FA_CREATE_NEW))));
      }
      TRACE("f_write: %s\n", ff_trace(f_write(fil, test,  sizeof(test), &len)));
      TRACE("f_close: %s\n", ff_trace(f_close(fil)));

      char buffer[128] = {0};
      TRACE("f_open: %s\n", ff_trace(f_open(fil, "public/index.txt", FA_READ)));
      TRACE("f_read: %s\n", ff_trace(f_read(fil, buffer,  128, &len)));
      TRACE("f_close: %s\n", ff_trace(f_close(fil)));
      TRACE("\nfoo.txt: %s\n", buffer);
-------------------------------------------------------
OUTPUT:
--------------------------------------------------------
	f_open: FR_OK
	f_write: FR_OK
	f_close: FR_OK
	f_open: FR_OK
	f_read: FR_OK
	f_close: FR_OK
	foo.txt: <html><title>TEST</title><head><body>TEST page..</body></head></html>

**Someone Guide me from here:**

negativekelvin, 2018-07-06:
Try .htm or enable long filename support in fatfs

KrrishnaShah, 2018-07-11:
Thankyou @negativekelvin 

Eanbling the LFN solved the issue.

closed 2018-07-11 by KrrishnaShah