espressif/esp-idf #2158: Problem with Software controlled WiFi/Bluetooth coexistence in release 3.1

slivings99, 2018-07-06:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32D0WDQ6 (revision 1)]
- IDF version: 698d3ba
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

We recently updated our project to use esp-idf release 3.1 instead of release 3.0. We are running wifi with ble concurrently (coexisting). We had not had any problems with release 3.0 being connected to a wifi station while also serving as a gatt server. But with the upgrade, we are having extreme difficulty in even connecting via ble with GATT timeout errors or simply 0x85 Gatt error. I have found that if we disable "Software controls WiFi/Bluetooth coexistence" in sdkconfig we return to normal behavior. 

I realize I am not giving much information, but mainly I am posting to see if others are having a problem with software controlled coexistence and if there is anything we are missing to have this work. Is there an inherent advantage?

### Expected Behavior

I would expect that software controlled coexistence would also work.

I am willing to provide more log information if desired.

Trubochkin, 2018-07-10:
I think that you have a problem because esp-idf 3.1 is still in beta.
I also have some problems with this release.

tangqingcai, 2018-07-11:
I also met a problem like this;
My test case is wifi and ble working together at the same time;
I found if i enable "Software controls WiFi/Bluetooth coexistence" and select "Balance";
The ble server will be hardly found,and The Services also be hardly discovered; 
But if i disable this option "Software controls WiFi/Bluetooth coexistence";
It works well;
I guess this maybe some problems;

TianHao-Espressif, 2018-08-20:
@slivings99 , I'm sorry for delayed reply. Please try the latest master branch or release/3.1 branch. I have resolved the coexist bugs, one of them is  about your problem.