espressif/esp-idf #2161: Compiling and linking issue when two components have the same component dir name.

ahmedwahdan, 2018-07-07:
## Environment

- Development Kit:     ESP32-DevKitC

## Problem Description
I'm trying to implement AWS_OTA_PAL and had to include additional components.
I have two components dirs call OTA under two different components directories
Thirdparty/OTA
ESP32/OTA
But in build directory only one of them is compiled and linked.
### Expected Behavior
Both component to be compiled and linked.
### Actual Behavior
Only one component is build and linked.

negativekelvin, 2018-07-07:
https://github.com/espressif/esp-idf/issues/457
https://github.com/espressif/esp-idf/issues/642

So try

Thirdparty/3P_OTA/OTA
ESP32/OTA

ahmedwahdan, 2018-07-07:
I believe the issue is the component dir name OTA

projectgus, 2018-07-09:
Hi @ahmedwahdan,

Thanks for the clear bug report. Unfortunately there is no way to compile multiple components with the same component name into a project. (The component name is the name of the directory containing component.mk).

This is an intended feature of the ESP-IDF build system, because it allows projects to replace ESP-IDF components. There is some more information about this in the build system docs here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#multiple-components-with-the-same-name

Suggest renaming one of the components, or rearranging the directory structure so that the component-level directories have unique names.

closed 2018-07-09 by projectgus