espressif/esp-idf #2165: ble advertising

chegewara, 2018-07-09:
This does not seems to be bug, but just in case im posting it. After disconnecting peer device from esp32 there is `E` message about fail to start ble adv, but it does not seems to be truth. Esp32 is advertising itself anc can be discovered. Here is log:
```
I (33427) HAL_BLE: Client disconnected, restarting advertising
I (33427) BT_APPL: BTA_DmSetBleAdvParamsAll: 40, 80

I (33427) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

I (33437) BT_APPL: BTA_DmSetBleAdvParamsAll: 40, 80

I (33437) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

E (33457) BT_APPL: bta_dm_ble_set_adv_params_all(), fail to start ble adv.
I (33457) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
```

Weijian-Espressif, 2018-07-16:
@chegewara how to reproduce the issue ?

chegewara, 2018-07-16:
@Weijian-Espressif I have no idea what i did, i will try to recollect which project it was and back to this issue.

Weijian-Espressif, 2018-08-06:
hi @	chegewara:
  sorry , I can not reproduce the issues.