espressif/esp-idf #2166: Factory reset infinite loop

OtherCrashOverride, 2018-07-09:
## Environment

- Development Kit:      [ODROID-GO]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    ESP-IDF v3.1-beta1-42-gd5c0610 2nd stage bootloader
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB|Battery]

## Problem Description
When using GPIO_33 pin to enter factory reset, a panic occurs only when the GPIO is continuously held.  Releasing the GPIO allows the factory reset to boot.

### Expected Behavior
Holding a GPIO should not crash the bootloader.

### Actual Behavior
Bootloader crashes.

### Steps to repropduce

1. Flash factory software (https://github.com/OtherCrashOverride/odroid-go-firmware)
2. Hold "B" button.
3. Observe crash in UART output.

### Code to reproduce this issue
https://github.com/OtherCrashOverride/esp-idf/tree/release/v3.1-odroid
https://github.com/OtherCrashOverride/odroid-go-firmware

## Debug Logs

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6680
load:0x40078000,len:7796
load:0x40080000,len:10192
0x40080000: _iram_start at /srv/disk0/odroid-go-public/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x400804a4
I (29) boot: ESP-IDF v3.1-beta1-42-gd5c0610 2nd stage bootloader
I (29) boot: compile time 20:04:20
I (31) boot: Enabling RNG early entropy source...
I (35) qio_mode: Enabling QIO for flash chip GD
I (40) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 16MB
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 firmware         factory app      00 00 00010000 000f0000
I (93) boot:  4 app_0            OTA app          00 10 00100000 00100000
I (101) boot:  5 app_1            OTA app          00 11 00200000 00100000
I (108) boot:  6 app_2            OTA app          00 12 00300000 00100000
I (116) boot:  7 app_3            OTA app          00 13 00400000 00100000
I (123) boot:  8 app_4            OTA app          00 14 00500000 00100000
I (131) boot:  9 app_5            OTA app          00 15 00600000 00100000
I (138) boot: 10 app_6            OTA app          00 16 00700000 00100000
I (146) boot: 11 app_7            OTA app          00 17 00800000 00100000
I (154) boot: 12 app_8            OTA app          00 18 00900000 00100000
I (161) boot: 13 app_9            OTA app          00 19 00a00000 00100000
I (169) boot: 14 storage          Unknown data     01 82 00f00000 00100000
I (176) boot: End of partition table
I (181) boot: Defaulting to factory image
I (5185) boot: Detect a condition of the factory reset
I (5185) boot: Data partitions to erase: nvs
I (5185) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (5300) boot_comm:  0 nvs              data  00009000 00004000 [yes]
I (5357) boot_comm:  1 otadata          data  0000d000 00002000 [yes]
I (5357) boot_comm:  2 phy_init         data  0000f000 00001000 [no]
I (5360) boot_comm: 14 storage          data  00f00000 00100000 [no]
I (5367) boot: Defaulting to factory image
I (5372) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11f2c ( 73516) map
I (5414) esp_image: segment 1: paddr=0x00021f54 vaddr=0x3ffb0000 size=0x024f8 (  9464) load
I (5419) esp_image: segment 2: paddr=0x00024454 vaddr=0x3ffb24f8 size=0x00000 (     0) load
I (5421) esp_image: segment 3: paddr=0x0002445c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /srv/disk0/odroid-go-public/esp-idf/components/freertos/xtensa_vectors.S:1685

I (5431) esp_image: segment 4: paddr=0x00024864 vaddr=0x40080400 size=0x0b7ac ( 47020) load
I (5464) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x2748c (160908) map
0x400d0018: _flash_cache_start at ??:?

I (5538) esp_image: segment 6: paddr=0x000574ac vaddr=0x4008bbac size=0x03314 ( 13076) load
0x4008bbac: rtc_clk_cpu_freq_get at /srv/disk0/odroid-go-public/esp-idf/components/soc/esp32/rtc_clk.c:445

I (5545) esp_image: segment 7: paddr=0x0005a7c8 vaddr=0x400c0000 size=0x00000 (     0) load
I (5545) esp_image: segment 8: paddr=0x0005a7d0 vaddr=0x50000000 size=0x00000 (     0) load
I (5562) boot: Loaded app from partition at offset 0x10000
I (5562) boot: Disabling RNG early entropy source...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d4e81  PS      : 0x00060b30  A0      : 0x800d156f  A1      : 0x3ffe3be0  
0x400d4e81: psram_read_id at /srv/disk0/odroid-go-public/esp-idf/components/esp32/spiram_psram.c:324

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ff133fc  
A6      : 0x3ff00044  A7      : 0x00000080  A8      : 0x80083f90  A9      : 0x00c86018  
A10     : 0x28ce2afc  A11     : 0x00001b00  A12     : 0x00001300  A13     : 0x00000000  
A14     : 0x7fef6fdc  A15     : 0x00000200  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000023c  LBEG    : 0x4008e550  LEND    : 0x4008e55b  LCOUNT  : 0x00000000  
0x4008e550: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4008e55b: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x400d4e81:0x3ffe3be0 0x400d156c:0x3ffe3c10 0x4008124b:0x3ffe3c30 0x40078acf:0x3ffe3c50 0x40078b81:0x3ffe3c80 0x40078b9f:0x3ffe3cc0 0x40078e8d:0x3ffe3ce0 0x400804c9:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x400d4e81: psram_read_id at /srv/disk0/odroid-go-public/esp-idf/components/esp32/spiram_psram.c:324

0x400d156c: esp_spiram_init at /srv/disk0/odroid-go-public/esp-idf/components/esp32/spiram.c:109

0x4008124b: call_start_cpu0 at /srv/disk0/odroid-go-public/esp-idf/components/esp32/cpu_start.c:152

Rebooting...
ets Jun  8 2016 00:22:57

```

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
https://github.com/OtherCrashOverride/odroid-go-firmware/blob/master/sdkconfig
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

OtherCrashOverride, 2018-07-11:
As suggested by the backtrace, testing has confirmed this issue only occurs when PSRAM is enabled in the sdkconfig.

KonstantinKondrashov, 2019-03-18:
@OtherCrashOverride Do you have any updates for this issue? Thanks.

OtherCrashOverride, 2019-03-24:
@KonstantinKondrashov I have no additional information to provide.  The workaround remains to disable PSRAM.

KonstantinKondrashov, 2019-04-24:
Hi @OtherCrashOverride, I have reproduced your issue and fix is in our internal repo. The issue existed only for v3.1 we will update it soon. 

You can use this patch [4858.zip](https://github.com/espressif/esp-idf/files/3110360/4858.zip).

Alvin1Zhang, 2020-06-08:
The fix has been available on Github, feel free to reopen if the issue still happens. Thanks.

closed 2020-06-08 by Alvin1Zhang