espressif/esp-idf #2167: Possibility of multiconnect mode in gatt_client example

KryvosheyaIvan, 2018-07-09:
Hello!
Several months ago I started using gatt_client example for my purposes. Than I needed to use me esp32 to connect to multiple BLE GATT servers (CC2640). When I connected two ble servers I noticed that at the beginning all works fine, I send command to switch led on and off: two ble servers light their leds on and off . It works fine for several minutes.
But after that only one of the server does as command orders, though WRITE_CHAR_EVT says that It transmit the order and no disconnect event!
Can it be that in gatt_client example GATT client can connect only to 1 server?
I mean in another example "gattc_multi_connect" there is an option in makemenuconfig.
Can it be the issue?
Thank you for any help.

Weijian-Espressif, 2018-07-12:
hi @KryvosheyaIvan , Our ESP32 support multiconnection. Can you provide your test code?

KryvosheyaIvan, 2018-07-12:
@Weijian-Espressif, I am sorry. All that I wrote is true but I was wrong when I wrote that there was WRITE_CHAR event. Nevertheless. I have several (two) GATT servers and I send commands to them. It can work properly 1min, 3, 5, 15 (or even longer) but then at one moment one of servers cant get my message and another works fine. 
In my program first I register my devices, then perform scanning, and then one by one I connect founded devices. Also, for evere server I have one separate task. Also tasks cant work at the same time, because I use mutex to share BLE resource.
What is strange and bad is that though I cant write to server, I have not any events before! I , mean there are no disconnect event, close event as It have to be.
This is one of my functions to write to certan char in certain service
`/* function writes value to a specific character in specific service */
void writeCharact_in_Service_v3(uint8_t fLOCK_ID_x, esp_bt_uuid_t* pService, esp_bt_uuid_t* pCharacteristic, const char* CMD, uint8_t length)
{

	get_server = false;
	vTaskDelay(100 / portTICK_PERIOD_MS); //speed
	esp_ble_gattc_search_service(gl_profile_tab[fLOCK_ID_x].gattc_if, gl_profile_tab[fLOCK_ID_x].conn_id, pService);//discover services

	//Wait forever while service is not founded
	vTaskDelay(100 / portTICK_PERIOD_MS);
	xSemaphoreTake(smphr_service_search_cmplt, portMAX_DELAY); //forever

    if (get_server)
    {
        uint16_t count = 2; //number of char`s we want to find
        esp_gatt_status_t status = esp_ble_gattc_get_attr_count( gl_profile_tab[fLOCK_ID_x].gattc_if,
        														 gl_profile_tab[fLOCK_ID_x].conn_id,
                                                                 ESP_GATT_DB_CHARACTERISTIC,
                                                                 gl_profile_tab[fLOCK_ID_x].service_start_handle,
                                                                 gl_profile_tab[fLOCK_ID_x].service_end_handle,
                                                                 INVALID_HANDLE,
                                                                 &count);

        vTaskDelay(100 / portTICK_PERIOD_MS); 

        if (status != ESP_GATT_OK){
            ESP_LOGE(GATTC_TAG, "esp_ble_gattc_get_attr_count error");
        }

        if (count > 0){
        //debug	printf(" \n There is at least 1 characteristic \n");
            char_elem_result = (esp_gattc_char_elem_t *)malloc(sizeof(esp_gattc_char_elem_t) * count);
            if (!char_elem_result){
                ESP_LOGE(GATTC_TAG, "gattc no mem");
            }else{
                status = esp_ble_gattc_get_char_by_uuid( gl_profile_tab[fLOCK_ID_x].gattc_if,
                										 gl_profile_tab[fLOCK_ID_x].conn_id,
                                                         gl_profile_tab[fLOCK_ID_x].service_start_handle,
                                                         gl_profile_tab[fLOCK_ID_x].service_end_handle,
														 *pCharacteristic,
                                                         char_elem_result,
                                                         &count);

                vTaskDelay(100 / portTICK_PERIOD_MS);

                /* Write characteristics value */
                esp_err_t ret_write = -2;

                ret_write = esp_ble_gattc_write_char(  gl_profile_tab[fLOCK_ID_x].gattc_if,
                						   gl_profile_tab[fLOCK_ID_x].conn_id,
										   	char_elem_result->char_handle,
                                            length,								//+
											CMD,								//+
                                            ESP_GATT_WRITE_TYPE_RSP,			//+ ESP_GATT_WRITE_TYPE_NO_RSP
											ESP_GATT_AUTH_REQ_NONE);			//+ ESP_GATT_AUTH_REQ_NONE

                printf("\n return code: %d", ret_write);
                vTaskDelay(200 / portTICK_PERIOD_MS); //speed

                ESP_LOGE(GATTC_TAG, "Forever waiting until written");

                uint8_t not_wr_countount = 0;
                while ( !Write_event[fLOCK_ID_x] )
                {
                	not_wr_count++;
                	ESP_LOGE(GATTC_TAG, "not written yet");
                	vTaskDelay(200 / portTICK_PERIOD_MS);
                	if (not_wr_count > 50)
                	{
                		ESP_LOGE(GATTC_TAG, "Closing connection...");

                		esp_ble_gap_disconnect(bda_connected[fLOCK_ID_x]);
                		break;	//EXIT WHILE LOOP
                	}
                }
                ESP_LOGI(GATTC_TAG, "WRITTEN[%d]",fLOCK_ID_x);

                if (status != ESP_GATT_OK){
                    ESP_LOGE(GATTC_TAG, "esp_ble_gattc_get_char_by_uuid error");
                }
            }
            // free char_elem_result
            free(char_elem_result);
        }else{
            ESP_LOGE(GATTC_TAG, "no char found");
        }
    }//end of if(get_server)
}`

Do you need all code?

Weijian-Espressif, 2018-07-16:
hi @KryvosheyaIvan 
  gatt client will get disconnect event if BLE disconnect. Can you provide your all test code?
you can also test our multiconnection demo.

Weijian-Espressif, 2018-09-07:
hi @KryvosheyaIvan , Has the problem been solved?

KryvosheyaIvan, 2018-09-07:
@Weijian-Espressif , yes, thanks

closed 2018-09-07 by KryvosheyaIvan