espressif/esp-idf #2168: Cmake WPS

FlorisColpaert, 2018-07-09:
Hi,

I try to build the WPS example with cmake. I succeeded in older code but with the latest versions, it fails to build.
problem :
[100%] Linking CXX executable wps_example.elf
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x18): undefined reference to `uuid_gen_mac_addr'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x20): undefined reference to `wps_build_assoc_req_ie'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x24): undefined reference to `wps_build_assoc_resp_ie'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x28): undefined reference to `wps_build_probe_req_ie'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x2c): undefined reference to `wps_build_public_key'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x30): undefined reference to `wps_enrollee_get_msg'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x34): undefined reference to `wps_enrollee_process_msg'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x38): undefined reference to `wps_generate_pin'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x3c): undefined reference to `wps_is_selected_pin_registrar'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x40): undefined reference to `wps_is_selected_pbc_registrar'
esp32/libesp32.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wps_crypto_funcs+0x44): undefined reference to `eap_msg_alloc'

## Environment
- Core (if using chip or module): ESP-WROOM32
- IDF version commit e7faa1d851556bcfa05cf512175fe3e10463484b
- Development Env:		[Cmake]
- Operating System:		[Ubuntu]

mahavirj, 2018-07-10:
@FlorisColpaert Thanks for reporting, fix will be available soon.

FlorisColpaert, 2018-07-11:
@mahavirj  Fixed it in my local branch by adding src folder in CMakeLists 
set(COMPONENT_SRCDIRS src/crypto port src/fast_crypto src/wps)
components/wpa_supplicant/CMakeLists.txt