espressif/esp-idf #2169: Error: No bootable app partitions in the partition table

lmayainsite, 2018-07-09:
## Environment

- Development Kit:      Own board
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Core (if using chip or module): [ESP32-WROOM-32U]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): c2b39f4
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB|external 12V - Regulator 12 to 3.3V]

## Problem Description

Im using a module ESP32-WROOM-32U and my own circuit design. First I flashed the blink example and works good(selecting Patition Table: Single factory app, no OTA). Later after testing in my own project with the below characteristics in the Partition table(selecting .csv document), I got a problem. The problem didn't occur in the first flash, it's ocurred few days after. 
This is the partion table:
**nvs,data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
storage,  data, spiffs,  0x10000, 0x10000
factory,  0,    0,       0x20000, 1M
ota_0,    0,    ota_0,   0x120000,1M
ota_1,    0,    ota_1,   0x220000,1M**

Later, I copied this characteristics in the partion table of the blink example, and I got this log 

**I (30) boot: ESP-IDF v3.0.1-rc-dirty 2nd stage bootloader
I (30) boot: compile time 10:26:07
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 26.7MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 storage          Unknown data     01 82 00010000 00010000
I (89) boot:  4 factory          factory app      00 00 00020000 00100000
I (96) boot:  5 ota_0            OTA app          00 10 00120000 00100000
I (104) boot:  6 ota_1            OTA app          00 11 00220000 00100000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x071e4 ( 29156) map
I (141) esp_image: segment 1: paddr=0x0002720c vaddr=0x3ffb0000 size=0x022b0 (  8880) load
I (145) esp_image: segment 2: paddr=0x000294c4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/programador/esp_32/esp-idf/components/freertos/./xtensa_vectors.S:1685**

**I (150) esp_image: segment 3: paddr=0x000298cc vaddr=0x40080400 size=0x06744 ( 26436) load
I (171) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x165a4 ( 91556) map
0x400d0018: _stext at ??:?**

**I (208) esp_image: segment 5: paddr=0x000465c4 vaddr=0x40086b44 size=0x01e54 (  7764) load
0x40086b44: block_data_size at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:142
 (inlined by) multi_heap_realloc_impl at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:525**

**I (212) esp_image: segment 6: paddr=0x00048420 vaddr=0x400c0000 size=0x00000 (     0) load
E (215) esp_image: Checksum failed. Calculated 0xad read 0x8a
E (221) boot: Factory app partition is not bootable
E (227) esp_image: image at 0x120000 has invalid magic byte
W (233) esp_image: image at 0x120000 has invalid SPI mode 255
W (240) esp_image: image at 0x120000 has invalid SPI size 15
E (246) boot: OTA app partition slot 0 is not bootable
E (252) esp_image: image at 0x220000 has invalid magic byte
W (258) esp_image: image at 0x220000 has invalid SPI mode 255
W (264) esp_image: image at 0x220000 has invalid SPI size 15
E (271) boot: OTA app partition slot 1 is not bootable
E (277) boot: No bootable app partitions in the partition table
user code done**

Later I putted again the default characterisitcs in my partition table, selecting Single factory app, no OTA. I flashed the blink example again and this was the log.

**I (30) boot: ESP-IDF v3.0.1-rc-dirty 2nd stage bootloader
I (30) boot: compile time 10:34:55
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 26.7MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x071e4 ( 29156) map
I (106) esp_image: segment 1: paddr=0x0001720c vaddr=0x3ffb0000 size=0x022b0 (  8880) load
I (110) esp_image: segment 2: paddr=0x000194c4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/programador/esp_32/esp-idf/components/freertos/./xtensa_vectors.S:1685**

**I (115) esp_image: segment 3: paddr=0x000198cc vaddr=0x40080400 size=0x06744 ( 26436) load
I (136) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x165a4 ( 91556) map
0x400d0018: _stext at ??:?**

**I (173) esp_image: segment 5: paddr=0x000365c4 vaddr=0x40086b44 size=0x01e54 (  7764) load
0x40086b44: block_data_size at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:142
 (inlined by) multi_heap_realloc_impl at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:525**

**I (177) esp_image: segment 6: paddr=0x00038420 vaddr=0x400c0000 size=0x00000 (     0) load
E (180) esp_image: Checksum failed. Calculated 0xa6 read 0x8a
E (187) boot: Factory app partition is not bootable
E (192) boot: No bootable app partitions in the partition table
user code done**

After this, I can't run all the examples with out get this kind of logs. 

Is this a hardware problem in my circuit? 
Is this a hardware problem in my ESP32 Module?
Or is this a programming problem? 

Thanks

chegewara, 2018-07-09:
Did you try `make erase_flash`?

lmayainsite, 2018-07-09:
Yes, I did it right now. It works for the blink example, but not works for ethernet example and i2c example

negativekelvin, 2018-07-10:
Are you sure app is always being flashed to 0x20000

lmayainsite, 2018-07-10:
I always use make erase_flash and sometime works. Yes, my app is being flashed to 0x20000. I was checking error and we discover that maybe some gpios connecting to a PIC generate errors. I dont know why, because I just connet 4 gpios to my PIC configure as outputs(GPIO32,GPIO33,GPIO14 and GPIO13). When I shutdown the PIC, It's works without problems. When I turn on my PIC, the errors appears.
Is there something to keep in mind for these GPIOS?
Thanks.

TobeChan, 2018-09-12:
I have a problem same with you my friend !if i open the Ethernet code and OTA code in a same project,Compilation is ok,but the monitor will prinft like:

I (523) esp_image: segment 8: paddr=0x001114c4 vaddr=0x50000000 size=0x00000 (
   0) load
E (530) esp_image: Image length 1053936 doesn't fit in partition length 1048576
E (538) boot: Factory app partition is not bootable
E (543) esp_image: image at 0x110000 has invalid magic byte
W (549) esp_image: image at 0x110000 has invalid SPI mode 204
E (556) boot: OTA app partition slot 0 is not bootable
E (562) esp_image: image at 0x210000 has invalid magic byte
W (568) esp_image: image at 0x210000 has invalid SPI mode 255
W (574) esp_image: image at 0x210000 has invalid SPI size 15
E (581) boot: OTA app partition slot 1 is not bootable
E (586) boot: No bootable app partitions in the partition table
Fatal exception (0): IllegalInstruction
epc1=0x40080786, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x0
0000000

i don't know how to solve it,can you help me ?Thanks

roadsteadsteve, 2020-09-18:
I am getting this error. Has this been resolved yet?

chegewara, 2020-09-19:
@lmayainsite i see few things in your logs now:
- `boot: ESP-IDF v3.0.1-rc-dirty 2nd stage bootloader`, are you using IDF v3.0? its old, EOL and no loger supported
- `boot: SPI Speed : 26.7MHz`, this may be related, i know about only 1 esp32 board that is using flash with 26MHz clock, all other are using 40/80MHz
- can you paste this line from logs, its right after reset(similar to this):
`rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))`

I would bet that its a problem with SPI speed:
```
W (233) esp_image: image at 0x120000 has invalid SPI mode 255
W (240) esp_image: image at 0x120000 has invalid SPI size 15
```

igrr, 2023-03-27:
I'll close this since there was no response for a while. If anyone meets a similar looking issue, please verify that the application has been flashed to the device correctly. You can use parttool.py to read the binary back from flash and compare it to the original binary you were flashing. If this doesn't help resolve the issue, please open a new ticket.

closed 2023-03-27 by igrr