espressif/esp-idf #2171: Using BLE then trying to use WiFi to connect to a cloud service gives MBEDTLS error...

wegunterjr, 2018-07-09:
Development Kit: devkitc
Kit version (for WroverKit/PicoKit/DevKitC): NA
Core (if using chip or module): ESP32
IDF version (any version)
Development Env:	PlatformIO
Operating System:	Windows
Power Supply: USB

I have tried to stop bluetooth, and have even tried to clear memory after stopping, but i still get this error:

`
[image]
`

THoughts on how to fix?  I have to do BLE first, then i connect to wifi - communication will be between device and mobile client over BLE, BUT will be updating data to an MQTT Broker.

Thanks!

negativekelvin, 2018-07-09:
Do you mean MBEDTLS_ERR_SSL_ALLOC_FAILED -0x7F00

wegunterjr, 2018-07-09:
it's not just an AWS issue. I have tried it with other MQTT brokers, and I get the same error when following my flow.

wegunterjr, 2018-07-09:
Checking this one...  https://github.com/espressif/esp-idf/issues/346

negativekelvin, 2018-07-09:
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN and esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)

wegunterjr, 2018-07-10:
so, the standard size is 16384, is it possible to make it larger without error?

negativekelvin, 2018-07-10:
You mean smaller?  Technically not unless you can guarantee server will follow it.

wegunterjr, 2018-07-13:
yes..smaller. I think the real problem is that I only have about 30k of heap when i get to the TLS handshake and it fails because there is not enough memory.

chegewara, 2018-07-13:
From my observations with arduino-esp32 you need 75kB free heap or even more to perform mbedTLS handshake or you will get errors.

wegunterjr, 2018-07-13:
oh wow!... I have that if I don't use BLE... as soon as I use BLE, it won't shake the hand.

wegunterjr, 2018-07-13:
@chegewara what changes did you make?  i am actually using the arduino/platformio ide right now.. so, using those libraries, but some of the information transfers...

chegewara, 2018-07-13:
What i did can sounds weird, but i have `vTaskDelete(NULL)` in my setup(). Its because regular arduino loop task got about 8kB stack, release memory from classic bluetooth with `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)`, i have bunch of tasks that starts and end before my actually loop is running. 
There must be a easier way to do that, but this works for me.
Because its amazon IoT which is using mqtt protocol and its active all the time its much easier than connecting with https protocol. When i have active connection with IoT then i can finally connect to ble device. Its important because you cant free memory from ble by disconnecting device or something.
So, first connect with mbedTLS and at the end with ble.

wegunterjr, 2018-07-13:
I will try that... but also, someone presented this as an option this morning:
[https://github.com/copercini/esp32-iot-examples/blob/master/ESP32_aws_iot/ESP32_aws_iot.ino](url)

chegewara, 2018-07-13:
Yes, i am using library write by @copercini but like i said, it was not easy to merge IoT with ble.

wegunterjr, 2018-07-13:
yikes...just tried to do the TLS handshake and then setup BLE and it reboots... check out my heap:
Heap beginning of setup() AWS IOT using WiFiClientSecure program: 152156
Heap afterWiFi begin: 122556
Checking wifi.......
connecting to AWS...
connected!
Heap afterBLE init: **3076**`

chegewara, 2018-07-14:
I think we should move this discussion on forum or to my repository:
https://github.com/chegewara/esp32-snippets/issues

mahavirj, 2018-10-23:
@wegunterjr @chegewara AWS IoT examples on `master` branch have been optimized for better dynamic memory allocations. Please refer https://github.com/espressif/esp-idf/commit/5585f699d11c76f3c4d51888ebc6e258bad13e48#diff-f81ac7f81ab0dde348df60f02dd3c071. Please help to close this issue once verified.

chegewara, 2018-10-23:
@mahavirj For me there is no issue, ive been working with arduino ide and my app is working. In worst case scenario i can move to esp-idf and/or use wrover board. Only issue i have is using option to allocate mbedTLS in SPIRAM, but this is different issue and i have to test it more.

mahavirj, 2018-10-25:
@chegewara Thanks for confirmation. We had recently added option for selecting memory allocation strategy for mbedtls, please refer https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L3. If you run into any issue, please feel free to raise new issue.

closed 2018-10-25 by mahavirj