espressif/esp-idf #2180: request for new feature in: tools/windows/eclipse_make.py

dsptech, 2018-07-10:
not an issue.
---------------

I would suggest to replace the line 
`
make = subprocess.Popen(["make"] + sys.argv[1:] +  ["BATCH_BUILD=1"], stdout=subprocess.PIPE)
`
with
`
make = subprocess.Popen(["make"] + sys.argv[1:] + sys.argv[2:] + ["BATCH_BUILD=1"], stdout=subprocess.PIPE)
`

It's backward compatible, and the 2nd line of code allow the inserting of additional parameters, like: -j<N> for parallel compiling.

dsptech, 2018-07-10:
I discovered a little issue. 
The proposed new command will not work because make is not able to manage the "parallel" switch before the command.
"make -j5 all" or "make -j5 clean" (as issued from eclipsed, ..... and even written in the readme) produce always the same incorrect result.
"make all -j5" and "make clean -j5" produce the correct results instead.

So, regarding the proposed enhacement, arg1 and arg2 need to be swapped, but I'm not able to tell if it is a polite solution.

dsptech, 2018-07-11:
In some cases the target is accepted anyway and, so, I need now more investigation to have reproducible data.
I close this request because is not strictly make behaviors related. In the case, I will open again in the future.

closed 2018-07-11 by dsptech

projectgus, 2018-07-11:
Hi @dsptech ,

Do I understand correctly that your goal is to support parallel make for Windows?

The item `sys.argv[1:]` should already expand to "all items on the command line after the first one", so the command line:

```
eclipse_make.py -j3 all
```
Should cause the following command to be run
```
make -j3 all BATCH_BUILD=1
```

Meaning if you change the Build Command in Eclipse project properties to
```
c:\path\to\esp-idf\tools\eclipse_make.py -j4
```
... then that should work already.

Is that not working as expected?

dsptech, 2018-07-11:
Hi, 
had already changed the command as you proposed and I can confirm that it work. 

My original request was referred instead to the eclipse built-in "enabled parallel build" options and, when the request was open I was not aware about some issues.

If I enable the eclipse built-in "enabled parallel build" option, by requesting for 4 tasks, then eclipse will issue all commands using the -j4 switch as first argument. 
Examples : "eclipse_make.py -j4 all", "eclipse_make.py -j4 clean", "eclipse_make.py -j4 flash" (the last one if the flash target has been added to the project).

In this case, the first issue that I seen was: "eclipse_make.py -j4 clean" does not perform the project cleaning but compile it again instead. 
But, if I change the mentioned PYTON line using "sys.argv[2:] + sys.argv[1:]" (the inverted order is intentionally) the MAKE will work as expected in ALL issued commands. 
... but again... , see the situation below.

Meanwhile, I performed several test in the mingw shell (and not from eclipse). In this condition, I seen that, sometime, the target is ignored if the switch "-j4"  appear as first argument.

So, any ignored target will revert the request to "make -j4" ( aka "make -j4 all" ) and the requested operation will be not performed.

Actually I have no detailed information or reproducible data about the above situation nor I know how to relate them to the makefile and/or the python command (passing two arguments to the make, instead of one argument, will change the behavior). 
So, I closed my request and I waiting to have these information.

dsptech, 2018-07-16:
Hi @projectgus 
I have un update: It was a my issue only.
Other tests showned some randomic "make" behaviors. So, I fully removed esp-idf and installed it again, and the mingw as well. 
Now everything work as expected even without changes on mentioned pyton line, as you already said.
In my previous installations something was missed or get corrupted.
I'm sorry I bothered you.
Thank you.
Regards.