espressif/esp-idf #2181: wpa2-enterprise does not work.

vitcon-iot, 2018-07-11:
### Hardware:
Board:							ESP32-DevKitC
Core Installation/update date:			[Arduino core for ESP32](https://github.com/espressif/arduino-esp32), 10/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600
Operating System:                                  Windows
Power Supply: USB with external 5V

### Problem Description:
The wpa2 enterprise feature is not working after the recent update.
If set to wpa2-enterprise, a ```Stack smashing protect failure``` error occurs during connection.

#### Expected Behavior
Connected to AP.

#### Actual Behavior
Stack smashing protect failure and rebooting.

#### Code to reproduce this issue:
```cpp
#include "esp_wpa2.h"
#include "esp_log.h"
#include <WiFi.h>

const char* ssid = "EngineeringBay"; // your ssid
#define EAP_ID "user1"
#define EAP_USERNAME "user1"
#define EAP_PASSWORD "pass1"

void setup() {
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.begin(115200);
  delay(10);

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  // WPA2 enterprise magic starts here
  WiFi.disconnect(true);
  WiFi.mode(WIFI_STA); //call before esp_wif_ista_wpa2_ent_enable()
  int err = 0;
  /*
    esp_wifi_sta_wpa2_ent_clear_ca_cert();
    esp_wifi_sta_wpa2_ent_clear_cert_key();
    esp_wifi_sta_wpa2_ent_set_disable_time_check(true);
  */
  err = esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_ID, strlen(EAP_ID));
  err = esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
  err = esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));

  esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
  err = esp_wifi_sta_wpa2_ent_enable(&config);

  // WPA2 enterprise magic ends here

  WiFi.begin(ssid);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  //Do nothing.
}
```

### Debug Logs:
```
I (28) wifi: wifi driver task: 3ffb7240, prio:23, stack:4096, core=0
I (67) wifi: wifi firmware version: 4fd6a2c
I (68) wifi: config NVS flash: enabled
I (71) wifi: config nano formating: disabled
I (85) wifi: Init dynamic tx buffer num: 32
I (85) wifi: Init data frame dynamic rx buffer num: 10
I (85) wifi: Init management frame dynamic rx buffer num: 10
I (89) wifi: Init static rx buffer size: 1600
I (93) wifi: Init static rx buffer num: 4
I (97) wifi: Init dynamic rx buffer num: 10
I (163) wifi: mode : sta (30:ae:a4:08:93:ac)
I (164) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (343) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
.I (1074) wifi: state: init -> auth (b0)
I (1078) wifi: state: auth -> assoc (0)
I (1091) wifi: state: assoc -> run (10)
I (1091) wpa: wpa2_task prio:2, stack:6656

Stack smashing protect failure!

abort() was called at PC 0x400e8e1c on core 1

Backtrace: 0x4008f670:0x3ffbb880 0x4008f873:0x3ffbb8a0 0x400e8e1c:0x3ffbb8c0 0x400dd9fd:0x3ffbb8e0 0x400ddaaa:0x3ffbb9a0 0x400dbb8b:0x3ffbb9e0 0x400da91f:0x3ffbba00 0x400dad76:0x3ffbba40 0x400d9b5d:0x3ffbba60 0x400d9580:0x3ffbbab0 0x400d95ed:0x3ffbbaf0 0x400d7feb:0x3ffbbb10 0x400d81ed:0x3ffbbb50 0x400d7341:0x3ffbbb70 0x400e679e:0x3ffbbbf0 0x400e6a2a:0x3ffbbc20

Rebooting...
```
#### Exception Decoder:
```
Decoding 17 results
0x400e8e1c: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
0x4008f670: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 649
0x4008f873: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 649
0x400e8e1c: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
0x400dd9fd: x509_parse_tbs_certificate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/x509v3.c line 1438
0x400ddaaa: x509_certificate_parse at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/x509v3.c line 1547
0x400dbb8b: tls_parse_cert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_common.c line 321
0x400da91f: tls_process_certificate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c line 336
0x400dad76: tlsv1_client_process_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c line 958
0x400d9b5d: tlsv1_client_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client.c line 801
0x400d9580: tls_connection_handshake2 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tls_internal.c line 568
0x400d95ed: tls_connection_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tls_internal.c line 568
0x400d7feb: eap_tls_process_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c line 482
0x400d81ed: eap_peer_tls_process_helper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c line 630
0x400d7341: eap_peap_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_peap.c line 1097
0x400e679e: eap_sm_process_request at ?? line ?
0x400e6a2a: wpa2Task at ?? line ?
```

The code works correctly on older Arduino cores for ESP32 on other PCs. (Installed on June 8, 2018)
Can I get help with this issue?

negativekelvin, 2018-07-11:
Can you reproduce with esp-idf example? Otherwise you should report as issue at https://github.com/espressif/arduino-esp32 ?

me-no-dev, 2018-07-11:
@negativekelvin this is IDF api above ;) nothing to do with Arduino. And `WiFi.mode(WIFI_STA);` is a well written function that will start the radio and that is it.

negativekelvin, 2018-07-11:
Ok I thought Arduino task context may be issue but wpa2_task is separate

XinDeng11, 2018-07-18:
@vitcon-iot can you offer the reproduce step and eap method you used?

vitcon-iot, 2018-07-18:
@XinDeng11 
1.  Install [Arduino IDE 1.8.5](https://www.arduino.cc/en/Main/Software)
2. Install [Arduino core for esp32](https://github.com/espressif/arduino-esp32) according to the procedure provided by [arduino-esp32](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md).
3. Upload the above code using the Arduino IDE.

That's it. Nothing special.

XinDeng11, 2018-07-18:
@vitcon-iot We just test all the EAP method we support, all of them are OK, can you use our latest IDF to test, rather than arduino

piyord, 2018-07-20:
Same for me with same error on ESP-WROOM-32 and on  ESP-WROVER

Working great with old arduino-esp32 core with date of June. Updated core yesterday and stop working.

piyord, 2018-07-20:
Network is academic Radius wpa2 enterprise

vitcon-iot, 2018-07-24:
@XinDeng11 
Tested using the latest IDF and connected to wpa2-enterprise.
[The SDK version of Arduino core for esp32 seems to be different.](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include)
Is there a way to apply the latest IDF SDK to Arduino core for esp32?

piyord, 2018-07-25:
Ups sorry, I did not see that this was IDF api issues, not arduino

piyord, 2018-07-25:
I see that @vitcon-iot  posted this issue in arduino repository:

https://github.com/espressif/arduino-esp32/issues/1618

and other user advised him to post the issue in IDF's repository.

vitcon-iot, 2018-08-09:
I found that the stack smashing protection mode of the wpa2-enterprise example in esp-idf is set to None.
Setting the stack smashing protection mode to normal in esp-idf also causes the same problem.

```
I (140) wifi: wifi driver task: 3ffc0d00, prio:23, stack:3584, core=0
I (140) wifi: wifi firmware version: 4c398ce
I (140) wifi: config NVS flash: enabled
I (150) wifi: config nano formating: disabled
I (150) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (160) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (190) wifi: Init dynamic tx buffer num: 32
I (190) wifi: Init data frame dynamic rx buffer num: 32
I (190) wifi: Init management frame dynamic rx buffer num: 32
I (190) wifi: Init static rx buffer size: 1600
I (200) wifi: Init static rx buffer num: 10
I (200) wifi: Init dynamic rx buffer num: 32
I (200) example: Setting WiFi configuration SSID EngineeringBay...
I (210) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (290) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (300) wifi: mode : sta (30:ae:a4:08:93:ac)
I (420) wifi: n:8 1, o:1 0, ap:255 255, sta:8 1, prof:1
I (1420) wifi: state: init -> auth (b0)
I (1430) wifi: state: auth -> assoc (0)
I (1430) wifi: state: assoc -> run (10)
I (1430) wpa: wpa2_task prio:2, stack:6656

I (1510) wpa: SSL: Need 1578 bytes more input data
I (1550) wpa: SSL: Need 554 bytes more input data

Stack smashing protect failure!

abort() was called at PC 0x400d2a4c on core 1
0x400d2a4c: __stack_chk_fail at C:/msys32/home/JJH/esp/esp-idf/components/esp32/stack_check.c:36

Backtrace: 0x4008e35c:0x3ffc85b0 0x4008e537:0x3ffc85d0 0x400d2a4c:0x3ffc85f0 0x400ec0c1:0x3ffc8610 0x400ec166:0x3ffc86d0 0x400efe9f:0x3ffc8710 0x400ea687:0x3ffc8730 0x400eaaaa:0x3ffc8770 0x4
00e9989:0x3ffc8790 0x400e941c:0x3ffc87e0 0x400e9481:0x3ffc8820 0x400e89cb:0x3ffc8840 0x400e8b71:0x3ffc8880 0x400e7475:0x3ffc88a0 0x4011f1aa:0x3ffc8920 0x4011f436:0x3ffc8950
0x4008e35c: invoke_abort at C:/msys32/home/JJH/esp/esp-idf/components/esp32/panic.c:649

0x4008e537: abort at C:/msys32/home/JJH/esp/esp-idf/components/esp32/panic.c:649

0x400d2a4c: __stack_chk_fail at C:/msys32/home/JJH/esp/esp-idf/components/esp32/stack_check.c:36

0x400ec0c1: x509_parse_tbs_certificate at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/x509v3.c:1446

0x400ec166: x509_certificate_parse at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/x509v3.c:1555

0x400efe9f: tls_parse_cert at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_common.c:321

0x400ea687: tls_process_certificate at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c:336

0x400eaaaa: tlsv1_client_process_handshake at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c:958

0x400e9989: tlsv1_client_handshake at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tlsv1_client.c:801

0x400e941c: tls_connection_handshake2 at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tls_internal.c:568

0x400e9481: tls_connection_handshake at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/tls/tls_internal.c:568

0x400e89cb: eap_tls_process_input at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c:482

0x400e8b71: eap_peer_tls_process_helper at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c:630

0x400e7475: eap_peap_process at C:/msys32/home/JJH/esp/esp-idf/components/wpa_supplicant/src/wpa2/eap_peer/eap_peap.c:1097

0x4011f1aa: eap_sm_process_request at ??:?

0x4011f436: wpa2Task at ??:?

Rebooting...
```

This time, I replaced libwpa_supplicant.a of Arduino core for esp32 with the file created by setting stack smashing protection mode to None.

It works.

```
Connecting to network: EngineeringBay
...
WiFi connected
IP address set: 
192.168.0.12
```

negativekelvin, 2018-08-09:
Did you try this commit

https://github.com/espressif/esp-idf/commit/343bd32bd8aa9dcb1e7660267b422ca2781b759b

negativekelvin, 2018-09-01:
https://github.com/espressif/esp-idf/pull/2354

kelunik, 2018-10-02:
Does anyone have some hints how to apply the patch to test it within the Arduino IDE?

vitcon-iot, 2019-02-11:
The wpa2 enterprise feature is now working in Arduino core for ESP32 v1.0.1.

closed 2019-02-11 by vitcon-iot