espressif/esp-idf #2183: WIFI STOPS WITH NO EVENT -- wifi: pm stop, total sleep time:  42770435778 us / 42771016274 us

DavidMora, 2018-07-11:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Core (if using chip or module): ESP32-Wrover
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    6ee3cf67bdf7ae098c8a4240d99c3e4c4ece824b
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description

ESP works fine for short periods of time, but it disconnects from wifi. Debug logs show the following:

I (2621190) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (42788131) wifi: state: run -> auth (7c0)
I (42788132) wifi: pm stop, total sleep time: 42770435778 us / 42771016274 us

However i got no event on my handling routine to perform a reconnect or detect it otherwise. 

If it is important, when i run a reconnect routine with a sequence of wifi  stop -> start -> connect  it will reconnect to the AP successfully. 

This is in STA mode. 

### Expected Behavior

I should be getting SYSTEM_EVENT_STA_STOP or SYSTEM_EVENT_STA_DISCONNECTED.

### Actual Behavior

Disconnects with no event.

bilkusg, 2018-07-25:
I'm experiencing something somewhat similar on an ESP32 devkit v1.  Not entirely sure if it's the same underlying issue, but since nobody else has commented, this may be helpful.

To be precise,
- I listen for all wifi events using:
````
WiFi.onEvent(WiFiEvent); // WiFiEvent just prints each event as it occurs
````
- Then I connect:
````
WiFi.begin(ssid,password);
for (int i=0;i<10;i++) {
   delay(1000);
   if (WiFI.status() == WL_CONNECTED) {
        Serial.println("WiFi now connected");
        break; // all good
   }
````
This works ( for me reliably as I have a good signal from a very close by AP)
During the connection I see events 2 0 4 and 7, i.e. STA_START, WI_FI_READY,STA_CONNECTED and STA_GOT_IP which makes some kind of sense.

But, if I then turn off my AP, I expect to see an STA_DISCONNECTED, instead of which I see a sequence of
WI_FI_READY and STA_START events for a while. However, if I check WiFi.status() it now returns WL_DISCONNECTED.

So I can reliably tell if I've lost my connection by polling status, but not by relying on a disconnect event.

It looks to me as though somewhere in the underlying code, the system is trying to reconnect in the background before reporting failure. But it would be nice to be able to turn that off.....

Not sure if this is a bug, or just documented somewhere I haven't found it yet.....

TimXia, 2018-08-13:
@DavidMora @bilkusg We recently solved an issue like this. Please try the latest IDF master branch. It may already solve your issue.

DavidMora, 2018-08-14:
I can confirm the event is fired, thank you. Is this already on a release or just master?

TimXia, 2018-08-15:
@DavidMora It is already on release/v3.1 and master branch.

closed 2018-08-29 by FayeY

francescofcf, 2018-09-07:
I have a similar problem, I hope that we can find a solution 
https://github.com/espressif/esp-idf/issues/2375

have you solved?

lrodenasSistel, 2018-09-11:
Same issue here using latest release 3.1 (it was also hapenning on previous releases).
My project performs an HTTPS request every 30 seconds aprox, and I start wifi before each one and stop it atfer each one.
At some random point in time, it freezes forever in 
`xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);`
And I get on the UART log this:
`I (1071427) wifi: state: assoc -> run (200)`
Sometimes (200), sometimes (400).

I will try putting a lower timeout and handling the error... not sure yet how, maybe repeating the wifi start or the wifi connect? I'll try

ammaree, 2018-09-12:
We are suddenly, last 5 to 7 days, having the same or a very similar problem.

Using latest as of midday Sept 11 UTC IDF we are seeing problems with DHCP connections.
Get the pm start and stop messages although PM is completely disabled in kconfig.

Environment is completely Mikrotik with RB3011 as router also doing CAPSMan and wAP AC access points, all running latest 6.43 (stable) firmware.

On doing a physical reboot the stations sometimes take ~ 1m50s to connect, some take a lot longer.

On the MikroTik Winbox the lease status is:
a) when connection is lost shows "waiting" for a period
b) after reboot it shows "offered" for 30s counting down to 0:00
c) goes back to waiting for ~1min or more
d) sometimes change to connected after ~1m, if not we need to reboot the station again.

I think the problem started with the surprise change to LWIP v2.0.3. nor deployed on all devices in the field.

TimXia, 2018-09-18:
@ammaree Could you please call `esp_wifi_set_ps(WIFI_PS_NONE)` after `esp_wifi_init()` and test it again?

TimXia, 2018-09-18:
@lrodenasSistel Did you receive the SYSTEM_EVENT_STA_DISCONNECTED event?

lrodenasSistel, 2018-09-19:
> @lrodenasSistel Did you receive the SYSTEM_EVENT_STA_DISCONNECTED event?

@TimXia I think we do not receive any disconnected event when it happens, but how could I make sure of this? Now we have reduced the number of connections/disconnections a lot and it barely happens, but it still happens sometimes, so I could have 1 unit connected and read the log when it happens if that helps.

ammaree, 2018-09-25:
Have done and it seems to remove the messages.
Would it not make sense to make the default WIFI_PS_NONE?

Thanks