espressif/esp-idf #2184: WiFi connected, have IP, but IP stack broken

vonnieda, 2018-07-11:
I'm experiencing an issue where WiFi gets into a state where it thinks it's connected, and has an IP, but all communications to and from the ESP fail, including pings.

This seems to happen after some random number of disconnect / reconnect cycles. I will get a SYSTEM_EVENT_STA_CONNECTED followed by SYSTEM_EVENT_STA_GOT_IP but the ESP can't talk or be talked to.

Here is a log from a recent example:

```
wifi_disconnect
I (88367) wifi: state: run -> init (0)
I (88367) wifi: pm stop, total sleep time: 14144338 us / 24365739 us

I (88367) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
S (88377)  x_wifi.c:224  (0x0000) SYSTEM_EVENT_STA_DISCONNECTED ssid "xx", bssid 80:2a:a8:03:0d:5b, reason WIFI_REASON_ASSOC_LEAVE.
S (88397)  x_wifi.c:331  (0x0000) WiFi connecting to "xx".
E (88517)  x_mqtt.c:251  (0xffffffff) MQTTPublish
I (88517) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (89277) wifi: state: init -> auth (b0)
I (89287) wifi: state: auth -> assoc (0)
I (89287) wifi: state: assoc -> run (10)
I (89317) wifi: connected with xx, channel 11
I (89317) wifi: pm start, type: 1

S (89317)  x_wifi.c:181  (0x0000) SYSTEM_EVENT_STA_CONNECTED ssid "xx", bssid 80:2a:a8:03:0d:5b, channel 11, authmode WIFI_AUTH_WPA2_PSK.
I (90037)  x_mqtt.c:194  (0x0000) WiFi not connected, waiting for it.
I (90257) event: sta ip: 192.168.10.141, mask: 255.255.255.0, gw: 192.168.10.1
S (90257)  x_wifi.c:200  (0x0000) SYSTEM_EVENT_STA_GOT_IP IP 192.168.10.141, GW 192.168.10.1, Netmask 255.255.255.0.
I (90367)  x_mqtt.c:112  (0x0000) Connecting to MQTT at xx, use_tls = 1
I (93067)  x_mqtt.c:103  (0x0000) MQTT subscribed to xx
I (93237)  x_mqtt.c:103  (0x0000) MQTT subscribed to xx
I (93397)  x_mqtt.c:103  (0x0000) MQTT subscribed to xx
S (93497)  x_mqtt.c:170  (0x0000) xx connected to MQTT.
W (94337) wifi: alloc eb len=24 type=3 fail, heap:1540424

W (94337) wifi: m f null

W (95057) wifi: alloc eb len=24 type=3 fail, heap:1538656

W (95067) wifi: m f null

----------------- failures started here ---------------

E (115097) HTTP_REQ: Error write header
E (115097) HTTP_REQ: Error send request
I (115107) x_http_ap:104  (0x0108) API ERROR Status Code 0
E (115117) x_http_ap:167  (0x0108) err
E (115187) HTTP_REQ: Error write header
E (115187) HTTP_REQ: Error send request
```

I suspect that the following lines have something to do with it:
```
W (94337) wifi: alloc eb len=24 type=3 fail, heap:1540424
W (94337) wifi: m f null
W (95057) wifi: alloc eb len=24 type=3 fail, heap:1538656
W (95067) wifi: m f null
```

Once it gets in this state, it seems to persist until I force a WiFi disconnect and reconnect. During this time, attempting to ping the ESP from another computer on the network times out.

## Environment

- Development Kit: Custom
- Core (if using chip or module): ESP32-Wrover
- IDF version: 52f9a5ca
- Power Supply:         external 3.3V

Thanks,
Jason

vonnieda, 2018-07-11:
I just reproduced this again after about 15 disconnect / reconnect cycles, and one again I saw the following messages right when it failed:

```
W (307857) wifi: alloc eb len=24 type=3 fail, heap:1541988

W (307857) wifi: m f null
```

talss89, 2018-07-26:
## Environment
Development Kit: WROVER-KIT-V3
Core: ESP32-WROVER (D0WDQ6 rev1)
IDF version: v3.1-beta1 
Power Supply: 5v USB -> 3.3v onboard.

We seem to be experiencing the same issue. Although ours shows when I make a number of requests to the local HTTP server running on the device or making a HTTP request while AWS IoT is connecting, not after WiFi disconnect / reconnect.

I am reviewing my HTTP server code in detail now to try rule out the possibility of this causing the `wifi alloc eb` error. 

@vonnieda - is your `xx_mqtt.c` module using the AWS IoT embedded C SDK library internally? If so, that's something we both have in common - may be an area to investigate.

```
...
I (2290) wifi: wifi driver task: 3ffc0c58, prio:23, stack:4096, core=0
I (2290) wifi: wifi firmware version: b65dd05
I (2300) wifi: config NVS flash: enabled
I (2300) wifi: config nano formating: disabled
I (2310) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2320) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2340) wifi: Init dynamic tx buffer num: 32
I (2340) wifi: Init data frame dynamic rx buffer num: 12
I (2340) wifi: Init management frame dynamic rx buffer num: 12
I (2340) wifi: Init static tx buffer num: 16
I (2350) wifi: Init static rx buffer size: 1600
I (2350) wifi: Init static rx buffer num: 10
I (2350) wifi: Init dynamic rx buffer num: 12
I (2360) [_wifi_pg_js_init] Setting WiFi station SSID xxx...
I (2430) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (2440) wifi: mode : sta (30:ae:a4:4b:09:c4)
I (2440) [event_handler] WiFi Station Start
I (2700) [_pg_vm_thread] JS VM is now ready.
I (4850) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (5610) wifi: state: init -> auth (b0)
I (5610) wifi: state: auth -> assoc (0)
I (5620) wifi: state: assoc -> run (10)
I (5670) wifi: connected with xxx, channel 6
I (5670) wifi: pm start, type: 1
I (5670) [event_handler] WiFi Station Connected
I (8250) event: sta ip: 192.168.0.101, mask: 255.255.255.0, gw: 192.168.0.1
I (8250) [event_handler] got ip:192.168.0.101
Connected
System memory free: 4152K
I (8490) [iot_thread] MQTT Initialised
I (8510) [webapp_thread] Server listening on port 80
I (11150) [webapp_request_route] POST /foobar
W (11340) wifi: alloc eb len=24 type=3 fail, heap:4156504

W (11340) wifi: m f null

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: aws_iot
E (25470) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (25480) aws_iot: Error(-4) connecting to xxx:8883
...
```

vonnieda, 2018-07-26:
HI @talss89, no I'm not using AWS IOT, but I am using a custom port of Apache Paho and they are similar, and both use mbedtls. For what it's worth, I see this error both when connecting to MQTT and when trying to HTTPS connections, and as you note, it does seem to happen as soon as I start to connect to a service, rather than as soon as WiFi connects. 

I have long suspected this had something to do with mbedtls, but it's only a guess.

talss89, 2018-07-26:
Thanks @vonnieda, that's really useful info. And seems plausible. When I get a moment, I'll have a play with some of the mbedtls kconfig parameters. I wonder if disabling hardware AES has any effect...

I'll update this issue if I figure anything out.

vonnieda, 2018-07-26:
Alright, I think I am pretty close to a root cause on this. It appears to be related to memory allocation in the WiFi or TCP stack. I managed to get a new error message that seems to back this up:

```
W (11714) wifi: alloc eb len=24 type=3 fail
W (11724) wifi: m f null
W (11794) wifi: alloc eb len=36 type=3 fail
W (11794) wifi: mem fail
```

When my app starts it immediately tries to connect to a stored WiFi AP. Another task is blocking, waiting for WiFi to be connected, and as soon as it is, it tries to connect to MQTT over TLS. And yet another task is trying to send some requests to an HTTPS API if WiFi is connected.

I think what is actually happening is that DMA capable RAM is being exhausted. If I put a long delay in my code before hitting the HTTPS API everything is fine until it tries to connect, then I get the "null" failure. Here is a log with some memory logging included:

```
| 8BIT  | 32BIT | 32BIT - 8BIT | INTERNAL | SPIRAM |  DMA  | Heap Total | Used  | Free  | SOA Used | Free  |  HW   |
|=======|=======|==============|==========|========|=======|============|=======|=======|==========|=======|=======|
|1695740|1733568|         37828|     60748| 1672820|  22920|     4375996|2678708|1697288|         1|  32767|  32606|
I (2471) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (3231) wifi: state: init -> auth (b0)
I (3231) wifi: state: auth -> assoc (0)
I (3241) wifi: state: assoc -> run (10)
I (3261) wifi: connected with xx, channel 11
S (3271)  xx_wifi.c:181  (0x0000) SYSTEM_EVENT_STA_CONNECTED ssid "xx", bssid xx, channel 11, authmode WIFI_AUTH_WPA2_PSK.
I (5231) event: sta ip: 192.168.10.141, mask: 255.255.255.0, gw: 192.168.10.1
S (5231)  xx_wifi.c:200  (0x0000) SYSTEM_EVENT_STA_GOT_IP IP 192.168.10.141, GW 192.168.10.1, Netmask 255.255.255.0.
I (5361)  xx_mqtt.c:115  (0x0000) Connecting to MQTT at xx, use_tls = 1
I (6241) wifi: pm start, type:1
|1647372|1685200|         38020|     45892| 1639116|   8064|     4375996|2678708|1697288|         1|  32767|  32606|
I (9541)  xx_mqtt.c:108  (0x0000) MQTT subscribed to xx
I (9631)  xx_mqtt.c:108  (0x0000) MQTT subscribed to xx
I (9711)  xx_mqtt.c:108  (0x0000) MQTT subscribed to xx
S (9731)  xx_mqtt.c:167  (0x0000) xx xx xx connected to MQTT.
|1650656|1688484|         37828|     47784| 1640700|   9956|     4375996|2678708|1697288|         1|  32767|  32606|
|1650924|1688752|         37828|     48056| 1640696|  10228|     4375996|2678708|1697288|         1|  32767|  32606|
|1650924|1688752|         37828|     48056| 1640696|  10228|     4375996|2678708|1697288|         1|  32767|  32606|

-- RUNNING HTTPS API REQUEST

W (25721) wifi: alloc eb len=24 type=3 fail
W (25721) wifi: m f null
|1596988|1634816|         37828|     37828| 1596988|      0|     4375996|2678708|1697288|         1|  32767|  32606|
|1592124|1629952|         37828|     37828| 1592124|      0|     4375996|2678708|1697288|         1|  32767|  32606|
|1591720|1629548|         37828|     37828| 1591720|      0|     4375996|2678708|1697288|         1|  32767|  32606|
```

I'm investigating now if I can find a way to free up more DMA capable RAM to see if that helps. 

One other note: It seems like when these TLS connections fail they don't give back the RAM. Could be a memory leak, or could just be that this is an unrecoverable error and I should be rebooting anyway.

And finally, I'll note that I'm now on the 3.0.2 release instead of the detached HEAD I was using before.

Thanks,
Jason

vonnieda, 2018-07-26:
Okay, I've confirmed that the above is true. I was able to replace the mbedtls allocator with my own by uncommenting `#define MBEDTLS_PLATFORM_MEMORY` in `$IDF_PATH/components/mbedtls/port/include/mbedtls/esp_config.h` and then calling `mbedtls_platform_set_calloc_free(my_calloc, my_free);` and it all works perfectly now.

my_calloc and my_free are just functions that use SPI RAM instead of internal RAM.

So,now the question is how do I override the built in MBEDTLS_PLATFORM_MEMORY without having to hack at the IDF installation. I suppose I can make a local copy of the mbedtls component but it would be nice if this is define were configurable from menuconfig.

And FWIW, I just ran another test where instead of adding the custom allocator I decreased CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL in menuconfig to 256 and that seems to do the trick as well.

So, I think the main issue here is that if you are low on internal memory (below about 60k), mbedtls might eat it all, leaving none for the WiFi or TCP stack.

Jason

projectgus, 2018-07-26:
Hi @vonnieda ,

I replied to your post on the forums about this issue, here:
https://esp32.com/viewtopic.php?f=2&t=6550&p=28126#p28126

liuzfesp, 2018-07-31:
HI @vonnieda, could you check your menuconfig: components -> ESP32-specific -> SPI RAM config -> Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate internal memory.  If it's not enabled, enabled it and then to check whether it can solve your issue.

vonnieda, 2018-07-31:
Hi @liuzfesp - yes, I have this option enabled. It does not solve the issue. The issue, as noted in the forum, is simply that there is no enough internal memory for multiple TLS connections to be open at the same time. I have about 89k in FreeRTOS stacks, along with a number of fixed DMA buffers, and WiFi and BLE, and this leaves me with too little internal memory.

I think we can probably close this issue, since at it's core it's just an out of memory error. Perhaps it would be nice if the wifi "null" error message with a bit more descriptive.

negativekelvin, 2018-07-31:
Well it should probably at least generate a system event so the app can do something about it.

talss89, 2018-07-31:
We also had "Try to allocate memories of WiFi and LWIP in SPIRAM firstly" set. 

I like the idea of generating a system event on wifi alloc fail.

Thanks all.

liuzfesp, 2018-08-07:
We already have a feature which is used to optimize WiFi log, not only the log printed when not enough internal memory, but also the other WiFi log. Currently this feature has a related low priority compare to others. Will update to you once the log feature is completed.

As for the system event, I'm not sure whether it can be successfully post to the event task when there is not enough memory.

liuzfesp, 2018-08-07:
Moreover, WiFi stack only print log when failed to allocate the management buffer. For data buffer, it's normal if the memory is not enough in some scenario, for example, test throughput with iperf etc.

snahmad, 2018-09-05:
I am getting similar error
wifi: alloc eb len=192 type=2 fail

I am using WIFI with external RAM.

Internal Heap free: 41120
Heap free: 4195984:4233744

Plenty of external ram and some internal RAM.

Does in default settings WIFI only uses internal ram.

liuzfesp, 2018-09-25:
Yes, @snahmad, the default setting is interval RAM. 
Could you let me know the IDF commit IDF you are using?

snahmad, 2018-09-26:
I am using latest ESP-IDF SDK 3.2

snahmad, 2019-03-08:
Is this issue resolve.
Should I try "Try to allocate memories of WiFi and LWIP in SPIRAM firstly" set. in menuconfig.

snahmad, 2019-03-08:
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y make it work.

liuzfesp, 2019-03-22:
Hi @snahmad if SPIRAM is enabled, CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST should be enabled, otherwise you will encounter memory issue.

closed 2019-03-22 by Alvin1Zhang

Ritesh236, 2019-07-04:
Hi @liuzfesp, 

Is it compulsory to enable CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST when SPI PSRAM is enabked and wifi is also used in parallel? 

If that is compulsory then is there any meaning to give configurable option into menuconfig? 

Please clear out it so that we can use PSRAM along with WiFi together.

burkulesomesh43, 2019-07-05:
Hi @liuzfesp and @Ritesh236,

I have enabled this option but still getting issue. I am using esp-idf 3.2 stable version. is there some another settings?

Ritesh236, 2019-07-06:
Hello ESP32 SDK Developers, 

Would you please help to bruklesh regarding issue which he is facing while using wifi with PARAM?

talss89, 2019-07-06:
Hi @burkulesomesh43 & @Ritesh236,

My guess is that this isn't a bug in ESP-IDF, and is probably related to your application code.

I am not a contributor, but this might help:

- `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST` is not compulsory, but it is sensible if you have PSRAM installed. LWIP will then use `heap_caps_alloc(sz, MALLOC_CAP_SPIRAM)` where it can, reducing internal memory usage.
- The default ESP-IDF malloc() implementation uses `heap_caps_alloc(sz, MALLOC_CAP_8BIT)`. This often results in internal memory being allocated.
- It is possible that you are exhausting internal memory with your application's malloc() usage. I prefer to use `MALLOC_CAP_SPIRAM` whenever I can in application code. This is usually enough to solve the WiFi management buffer alloc failures.
- You may want to look at `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to set a threshold if allocating a size of < threshold then prefer internal memory, else prefer external / PSRAM.
- Also, check out `CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL` to reserve an internal memory area for *only* DMA / `MALLOC_CAP_INTERNAL` allocations.

Read these pages, and it should become clear:
- https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html
- https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/external-ram.html

Although the above docs are for 'latest', a lot of the fundamentals apply to ESP-IDF v3.

liuzfesp did a good write-up on `alloc eb fail`: https://github.com/espressif/esp-idf/issues/1021#issuecomment-469502062

Finally, it may be better to post on the ESP32 forums, rather than this issue, as AFAIK it's not a bug in IDF.

I hope you manage to fix your problem. Good luck.

Ritesh236, 2019-07-06:
Thanks @talss89 for your valuable feedback. 

So,  I have already checked few configurations earlier but still we will check it again and get back to you if any query regarding that. 

And also,  We will create issue over ESP32 community with details for same to get any other help regarding same. 

Thanks again for your valuable feedback

Regards, 
Ritesh Prajapati

burkulesomesh43, 2019-07-06:
@talss89 Thank you for your response.
I will check it from my  side and let you know.

Regards,
Somesh Burkule

burkulesomesh43, 2019-07-16:
> Hi @burkulesomesh43 & @Ritesh236,
> 
> My guess is that this isn't a bug in ESP-IDF, and is probably related to your application code.
> 
> I am not a contributor, but this might help:
> 
> * `CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST` is not compulsory, but it is sensible if you have PSRAM installed. LWIP will then use `heap_caps_malloc(sz, MALLOC_CAP_SPIRAM)` where it can, reducing internal memory usage.
> * The default ESP-IDF malloc() implementation uses `heap_caps_malloc(sz, MALLOC_CAP_8BIT)`. This often results in internal memory being allocated.
> * It is possible that you are exhausting internal memory with your application's malloc() usage. I prefer to use `MALLOC_CAP_SPIRAM` whenever I can in application code. This is usually enough to solve the WiFi management buffer alloc failures.
> * You may want to look at `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to set a threshold if allocating a size of < threshold then prefer internal memory, else prefer external / PSRAM.
> * Also, check out `CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL` to reserve an internal memory area for _only_ DMA / `MALLOC_CAP_INTERNAL` allocations.
> 
> Read these pages, and it should become clear:
> 
> * https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html
> * https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/external-ram.html
> 
> Although the above docs are for 'latest', a lot of the fundamentals apply to ESP-IDF v3.
> 
> liuzfesp did a good write-up on `alloc eb fail`: [#1021 (comment)](https://github.com/espressif/esp-idf/issues/1021#issuecomment-469502062)
> 
> Finally, it may be better to post on the ESP32 forums, rather than this issue, as AFAIK it's not a bug in IDF.
> 
> I hope you manage to fix your problem. Good luck.

Hi talss89,
I am using asprintf in my code to allocate memory.
How can we allocate it for MALLOC_CAP_SPIRAM?

milamber-ls, 2019-08-04:
I have tried all of the above but I still have the issue. can we please re open it?

negativekelvin, 2019-08-04:
https://github.com/espressif/esp-idf/issues/3592#issuecomment-517926444

gadget-man, 2020-07-21:
> Okay, I've confirmed that the above is true. I was able to replace the mbedtls allocator with my own by uncommenting `#define MBEDTLS_PLATFORM_MEMORY` in [...]

Jason, I'm having a similar issue with my ESP32-WROVER and want to try what you suggest above, but am not quite sure how to achieve it. I see from another post that MBEDTLS_PLATFORM_MEMORY is now added as a config option, so I believe I should set CONFIG_MBEDTLS_PLATFORM_MEMORY=y (I'm using mongoose-os), but I'm not clear how or where to implement  `mbedtls_platform_set_calloc_free(my_calloc, my_free);`.  Can you provide a hint?

vonnieda, 2020-07-21:
Hi @gadget-man,

You should call `mbedtls_platform_set_calloc_free` as early in your bootup code as you can. I'm not familiar with Mongoose, so I'm not sure if it does stuff before handing off control, but in my case I call it at the very beginning of my app_main() function.

The function pointers you pass just need to mirror the [calloc](http://www.cplusplus.com/reference/cstdlib/calloc/) and [free](http://www.cplusplus.com/reference/cstdlib/free/) functions. Mine are complicated, because I use a custom allocator, but it could be as simple as something like this:

```
static void* my_calloc(size_t num, size_t size) {
	/**
	 * 1. Allocate num * size bytes from SPIRAM.
	 * 2. Check if the allocation succeeded. If it failed, return NULL.
	 * 3. Zero the allocated memory to adhere to the contract of calloc.
	 * 4. Return the pointer to the allocated memory.
	 */
	void* ptr = heap_caps_malloc(num * size, MALLOC_CAP_SPIRAM);
	if (ptr == NULL) {
		return NULL;
	}
	memset(ptr, 0, num * size);
	return ptr;
}

static void my_free(void* ptr) {
	/**
	 * Memory allocated with heap_caps_malloc() can be freed using the
	 * system free, but this is here as a demonstration as to how
	 * you'd override it if needed.
	 */
	free(ptr);
}

void app_main()
{
	mbedtls_platform_set_calloc_free(my_calloc, my_free);
}
```

gadget-man, 2020-07-21:
> Hi @gadget-man,
>
> You should call `mbedtls_platform_set_calloc_free` as early in your bootup code as you can. I'm not familiar with Mongoose, so I'm not sure if it does stuff before handing off [...]

Many thanks for the quick response. When I call `mbdtls_platform_set_calloc_free` in my app_init (mgos), I get an error on compiling: `implicit declaration of function 'mbedtls_platform_set_calloc_free'`. I'll ask the mongoose guys what I'm doing wrong.

vonnieda, 2020-07-21:
You may need to #include "mbedtls/platform.h"

gadget-man, 2020-07-21:
Unfortunately that doesn't work. I've also checked that config.h has been modified to include: 
`#ifdef CONFIG_MBEDTLS_PLATFORM_MEMORY
#define MBEDTLS_PLATFORM_MEMORY
#endif`
Also also set `CONFIG_MBEDTLS_PLATFORM_MEMORY=y` in my build_vars