espressif/esp-idf #2186: ulp pulse counter not working with other pins except gpio0

Mr-Techtron, 2018-07-12:
- [√ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ √] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ √] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

## Environment

- Development Kit:      ESP32-DevKit by DoIt
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 04caff2f
- Development Env:	Eclipse
- Operating System:	Windows
- Power Supply:  USB

## Problem Description

In the example of ulp, pulse counting only works with GPIO0 pin. When I tried with other pins, it does not take any effect. I've only changed ulp_io_number and gpio_num accordingly in the given example and set ulp_debounce_counter and ulp_debounce_max_count to 1.

negativekelvin, 2018-07-12:
https://github.com/espressif/esp-idf/blob/master/examples/system/ulp/main/ulp_example_main.c#L61

negativekelvin, 2018-07-12:
What values did you try?

Mr-Techtron, 2018-07-12:
gpio1,gpio3 & gpio16

negativekelvin, 2018-07-12:
None of those pins are rtcio pins

Mr-Techtron, 2018-07-12:
i mean to say rtc_gpio1, rtc_gpio3, rtc_gpio16

negativekelvin, 2018-07-12:
Rtc_gpio1 is not even available on wroom32

Mr-Techtron, 2018-07-13:
[image]

negativekelvin, 2018-07-13:
Yes I know but gpio37 and gpio38 are not brought out to pads on wroom32.

igrr, 2018-07-13:
Thanks @negativekelvin for submitting the fix in https://github.com/espressif/esp-idf/pull/2190!

In addition to that, example is missing `rtc_gpio_init(gpio_num)` call before the `rtc_gpio_set_direction`. Will update example to fix that and simplify GPIO setup (automatically look up RTC IO number based on GPIO number).

Mr-Techtron, 2018-07-13:
@negativekelvin Oh I see but later i tried RTC_GPIO10 & RTC_GPIO12, it didnt work either; thanks for your help :)

igrr, 2018-07-28:
Issues reported here have been fixed in #2190 and 4b97a1edb25bc5368157ba4dcc884741fb347e58

closed 2018-07-28 by igrr