espressif/esp-idf #2187: reboot issue

chegewara, 2018-07-12:
Im not sure its something with my sdkconfig or lately something has been changed. Im getting this errors and sometimes esp32 just does not start (WindowOverflow4 and UserExceptionVector). :
```
load:0x40080000,len:8908
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x400803b4
0x400803b4: _UserExceptionVector at ??:?

I (350) cpu_start: Pro cpu up.
I (350) cpu_start: Starting app cpu, entry point is 0x40081174
0x40081174: call_start_cpu1 at D:/msys32/home/imper/esp/esp-idf/components/esp32/cpu_start.c:225
```

negativekelvin, 2018-07-12:
> This leads to idf_monitor decoding ROM bootloader "entry 0x40080xxx" message as one of the exception vectors, which looks scary to users.

https://github.com/espressif/esp-idf/commit/5551b3f61628fc815293cb6e93d32bf2dfe6a141

negativekelvin, 2018-07-12:
Ther may be a problem but it is not related to those 2 logs

chegewara, 2018-07-12:
Maybe it is or maybe its not, to be honest i dont feel comfort when i see you answering my posts.

closed 2018-10-01 by chegewara