espressif/esp-idf #2188: Multiple definitions for exception handlers when <regex> included

mill1000, 2018-07-12:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v2
- Core (if using chip or module): ESP-WROOM32
- IDF version: v3.1-beta1 
  Commit: c1fdd450 
- Development Env:		Make
- Operating System:		Windows
- Power Supply:                 USB

## Problem Description

When building a project with C++ standard regex (#include \<regex\>), and C++ exceptions are disabled in sdkconfig, the linker reports multiple definitions of the exception handlers.

Here are the first three
> c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__thr                                       ow_bad_exception()':
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc:58: multiple definition of `std::__throw_bad_exception()'
C:/Users/XXX/Git/HelloWorld/build/cxx\libcxx.a(cxx_exception_stubs.o):C:/Users/XXX/Git/esp-idf/components/cxx                                       /cxx_exception_stubs.cpp:12: first defined here
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__thr                                       ow_bad_alloc()':
functexcept.cc:(.text._ZSt17__throw_bad_allocv+0x0): multiple definition of `std::__throw_bad_alloc()'
C:/Users/XXX/Git/HelloWorld/build/cxx\libcxx.a(cxx_exception_stubs.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): fi                                       rst defined here
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__thr                                       ow_bad_cast()':
functexcept.cc:(.text._ZSt16__throw_bad_castv+0x0): multiple definition of `std::__throw_bad_cast()'
C:/Users/XXX/Git/HelloWorld/build/cxx\libcxx.a(cxx_exception_stubs.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): fi                                       rst defined here

The project links successfully if C++ exceptions are enabled in sdkconfig 

### Expected Behavior

Project builds successfully.

### Actual Behavior

Project fails to link

### Steps to repropduce

1. Disable C++ exceptions in sdkconfig
2. Add `#include <regex>`  to a source file
3. Build

### Code to reproduce this issue

```cpp
#include "esp_log.h"
#include <regex>
#include <string>

extern "C" void app_main()
{
  ESP_LOGI("Main","Running");
  std::string s ("subject");
  std::regex e ("(sub)(.*)");
  if (std::regex_match (s,e))
        ESP_LOGI("Main","Match found");
}

```

mrabin8393, 2019-05-08:
Was the issue solved? Facing the same. Thanks.

mill1000, 2019-05-08:
I have since abandoned the work that required regex, but this issue still occurs on esp-idf v3.1.2

HrMitrev, 2019-05-14:
Please add regex to esp_idf

Ivorforce, 2020-06-28:
+1 This is still happening

antmak, 2021-01-18:
Regex has been enabled in the esp-2019r2 version of the toolchain. If the issue occurs with modern IDF versions (and its corresponding toolchain), please open a new issue.

closed 2021-01-18 by antmak