espressif/esp-idf #2192: undefined reference to `__assert_fail'

yannS2016, 2018-07-13:
Good day all!

am experimenting with esp-32 , my aim is to used it in conjunction with iotivity framework, as of now am just testing the esp-idf sample program , though am not getting anywhere.
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v1]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
   04caff2
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

I am trying to compile the hello_world program under -~/esp/hello_world
but i keep on getting getting this error. 

First it was complaining about stubs-32.h as below
**gnu/stubs-32.h: No such file or directory**
following this thread isorted the issue 
https://stackoverflow.com/questions/7412548/error-gnu-stubs-32-h-no-such-file-or-directory-while-compiling-nachos-source

**Now am stuck with the errors below in the logs below**

GENCONFIG
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_clk.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
/home/eitvcs/esp/hello_world/build/bootloader/bootloader_support/libbootloader_support.a(bootloader_init.o):(.literal.bootloader_init+0x44): undefined reference to `__assert_fail'
/home/eitvcs/esp/hello_world/build/bootloader/bootloader_support/libbootloader_support.a(bootloader_init.o): In function `bootloader_init':
/home/eitvcs/esp/esp-idf/components/bootloader_support/src/bootloader_init.c:79: undefined reference to `__assert_fail'
/home/eitvcs/esp/esp-idf/components/bootloader_support/src/bootloader_init.c:80: undefined reference to `__assert_fail'
/home/eitvcs/esp/esp-idf/components/bootloader_support/src/bootloader_init.c:81: undefined reference to `__assert_fail'
/home/eitvcs/esp/esp-idf/components/bootloader_support/src/bootloader_init.c:82: undefined reference to `__assert_fail'
/home/eitvcs/esp/hello_world/build/bootloader/bootloader_support/libbootloader_support.a(bootloader_sha.o):(.literal.bootloader_sha256_data+0x20): more undefined references to `__assert_fail' follow
collect2: error: ld returned 1 exit status
/home/eitvcs/esp/esp-idf/make/project.mk:406: recipe for target '/home/eitvcs/esp/hello_world/build/bootloader/bootloader.elf' failed
make[1]: *** [/home/eitvcs/esp/hello_world/build/bootloader/bootloader.elf] Error 1
/home/eitvcs/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/eitvcs/esp/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/eitvcs/esp/hello_world/build/bootloader/bootloader.bin] Error 2

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2192780/sdkconfig.txt)

ginkgm, 2018-08-26:
hello @yannS2016 ,
I tried the version of your IDF and pass the compile.
Could you please try run ``git submodule update --recursive`` under your **IDF PATH**? i.e. $IDF_PATH, rather than your project folder (~/esp/hello_world).
Maybe the bootloader uses a function called __assert_failed which only exist in a specific submodule version.

Alvin1Zhang, 2018-10-17:
@yannS2016 Could you help share if any updates for this issue? Thanks.

projectgus, 2019-03-01:
Closing due to a lack of updates.

closed 2019-03-01 by projectgus