espressif/esp-idf #2193: openssl missing SSL_CTX_use_certificate_file, SSL_CTX_use_PrivateKey_file

chmorgan, 2018-07-13:
esp-idf master from last week

The openssl to mbedtls compatibility layer is missing two of these useful functions. Any plans to implement them in the near future?

donghengqaz, 2019-08-07:
The two APIs should base on file system, and you can see that they load the cert or key file from the target "const char *file", so that you must add a file system to your software. At esp-idf, we supply an easy way to do this, following APIs also work well :

1. SSL_CTX_use_certificate_ASN1
2. SSL_CTX_use_PrivateKey_ASN1

You can prefer to the example [openssl_server](https://github.com/espressif/esp-idf/tree/master/examples/protocols/openssl_server) to see how to use it.

Alvin1Zhang, 2019-09-29:
@chmorgan Thanks for reporting, and sorry for the slow turnaround. Would you please help share if  any updates for this issue? Thanks.

chmorgan, 2019-09-29:
HI @Alvin1Zhang. I haven't been back to this in a number of months. Appreciate checking in but I'm not sure what the current status is even or if it has been resolved.

Alvin1Zhang, 2019-09-29:
@chmorgan Thanks for the information. Feel free to reopen if you have more updates. Thanks.

closed 2019-09-29 by Alvin1Zhang