espressif/esp-idf #2195: Broken comparison against kconfiglib.expr_value()

ulfalizer, 2018-07-14:
Hello,

Stumbled upon this when browsing the code a bit:

https://github.com/espressif/esp-idf/blob/e7faa1d851556bcfa05cf512175fe3e10463484b/tools/kconfig_new/confgen.py#L217

`expr_value()` returns a 0/1/2 tristate value, so will need to be e.g. this instead:

```python
if kconfiglib.expr_value(cond_expr):
```

Similarly here too:

https://github.com/espressif/esp-idf/blob/e7faa1d851556bcfa05cf512175fe3e10463484b/tools/kconfig_new/confgen.py#L228

And yeah, that 0/1/2 vs. n/m/y thing is a bit confusing. It simplifies comparisons and streamlines the implementation a bit at least.

Cheers,
Ulf

projectgus, 2019-03-01:
Hi @ulfalizer ,

I don't know why I didn't see this before now. Thanks so much, your observation is of course 100% correct! Fix incoming soon.

(Still haven't forgotten about updating to upstream kconfiglib as well - but also haven't had time to do so :|.)

ulfalizer, 2019-03-01:
> (Still haven't forgotten about updating to upstream kconfiglib as well - but also haven't had time to do so :|.)

No worries!

If you want to quickly try out the `menuconfig`, you could apply the patch below (just to work around some env. variables not being set), install Kconfiglib with `pip3 install kconfiglib`, and run `menuconfig` in the esp-idf root. (Shamless plug. :)

No obligation though. I realize you have lots of stuff to do.

```diff
diff --git a/Kconfig b/Kconfig
index 75815ba1a..8bc9a2ac2 100644
--- a/Kconfig
+++ b/Kconfig
@@ -56,7 +56,7 @@ mainmenu "Espressif IoT Development Framework Configuration"
 
     endmenu  # SDK tool configuration
 
-    source "$COMPONENT_KCONFIGS_PROJBUILD"
+    #source "$COMPONENT_KCONFIGS_PROJBUILD"
 
     menu "Compiler options"
 
@@ -203,5 +203,5 @@ mainmenu "Espressif IoT Development Framework Configuration"
     endmenu # Compiler Options
 
     menu "Component config"
-        source "$COMPONENT_KCONFIGS"
+        #source "$COMPONENT_KCONFIGS"
     endmenu
```

closed 2019-03-13 by igrr