espressif/esp-idf #2196: Memory leak in HTTPS_MBEDTLS example

aliraza1944, 2018-07-15:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    a3c43251
- Development Env:		[ESP-IDF]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

The [example mbedtls_https code](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_mbedtls) creates only one task and continues doing multiple requests after the specified countdown.

I have modified the code to make multiple requests to different web addresses and for that I need to create task multiple times. Each task that is created is deleted too but the resources are not released. The free heap memory is reported after each task is created and after it is deleted. At the end of 3 request a while loop blinks an LED and reports free heap size every 5 seconds. I introduced this part of code because it was suggested that the TCP resources are sometimes freed after sometime.

heap value at start after wifi connection : 236484

Free Heap (after first https request ): 189652

Free Heap (after first https request ): 151216

Free Heap (after first https request ): 112936

Heap reported after every 5 seconds (best value) : 121604

### Expected Behavior
The heap should be freed after the task is deleted. (or after sometime the task is deleted)

### Actual Behavior
Free heap value decreases after consecutive task creation. The old task ais deleted before new one is created.

### Steps to repropduce
Run the provided code. Remember to swap in your wifi name and pass in the initialise_wifi function.

### Code to reproduce this issue

<script src="https://gist.github.com/aliraza1944/3a0efb7a6552556c098e520945619c25.js"></script>

## Solutions tried

1. Limiting no. of LWIP connections to 2. (no effect)
2. Tried heap tracing (most of un-freed memory is from mbedtls internal functions)
3. Static Tasks creation re-using only single stack and buffer for tasks.

negativekelvin, 2018-07-15:
You have to manually release the memory like

https://github.com/espressif/esp-idf/blob/50304d719ebd4a2346e24bc21b1a95c0853ec251/components/esp-tls/esp_tls.c#L136

aliraza1944, 2018-07-15:
@negativekelvin  Thank you very much. It solved the issue.

closed 2018-07-15 by aliraza1944