espressif/esp-idf #2197: OTA gets stuck/frozen in poor Wifi conditions

hchaudhary1, 2018-07-16:
## Environment

- Development Kit:      NONE (custom board)
- Core (if using chip or module):  PICO
- IDF version: 3.0 & 3.1
- Development Env: Make
- Operating System: Windows
- Power Supply: external 5V

## Problem Description

When I have poor RSSI on wifi, OTA can get frozen sometimes.

### Expected Behavior

During OTA, if the socket breaks or has a timeout, restart the processor.

### Actual Behavior
Socket times out, the OTA process gets frozen forever

### Steps to repropduce

1. Get to really poor wifi rssi
2. do an OTA
3. repeat until an OTA freezes freezes (1 in 5 for me)

See attached pic
[image]

mahavirj, 2018-08-09:
@hchaudhary1 
We have refactored OTA example/s, and modified such that it uses `esp_http_client` component from IDF instead of lwIP socket level API. 

Refer: https://github.com/espressif/esp-idf/commit/cdc792d72c3121db67d7360087fd59ab520d7ec7

Could you please try once with updated example in your environment?

jitin17, 2018-08-09:
As @mahavirj has pointed out, the OTA example/s now have support for `esp_http_client` component, which has a default timeout of 5000 ms after which the socket timeout occurs, and optionally, you can configure timeout in the `esp_http_client_config_t` struct.