espressif/esp-idf #2198: mesh: <nvs>error in read layer / assoc

QuentinGuillet, 2018-07-16:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM32
- IDF version : ESP-IDF v3.2-dev-179-g04caff2f
- Development Env:		mingw32
- Operating System:		Windows
- Power Supply:        external 5V

## Problem Description

Hello, 
I tried to execute the internal_communication mesh example provided with the ESP-IDF but I keep getting the 0x1102 error, which is ESP_ERR_NVS_NOT_FOUND. I probably missed something on the configuration step, but I could not find an answer to my problem on the online documentation.

### Code to reproduce this issue

the internal_communication mesh example
https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_communication

## Debug Logs

MONITOR
--- idf_monitor on COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
HHi0LB╗=]ER}EQUot'E3)║}ASeA%==)KgZ 0I
+0x4bbi7j Vi1!+0x&֙0Lb╹:4C!+i0&1ɕn:MN
e 009CI (28) boot: ESP-IDF v3.2-dev-179-g04caff2f 2nd stage bootloader
I (28) boot: compile time 16:58:52
I (28) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 80MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1bd18 (113944) map
I (126) esp_image: segment 1: paddr=0x0002bd40 vaddr=0x3ffb0000 size=0x036dc ( 14044) load
I (131) esp_image: segment 2: paddr=0x0002f424 vaddr=0x3ffb36dc size=0x00000 (     0) load
I (132) esp_image: segment 3: paddr=0x0002f42c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at C:/Users/quint/Documents/QuentinG/esp32_win32_msys2_environment_and_toolchain-20180110/msys32/home/quint/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (142) esp_image: segment 4: paddr=0x0002f834 vaddr=0x40080400 size=0x007dc (  2012) load
I (151) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x7cc20 (511008) map
0x400d0018: _stext at ??:?

I (308) esp_image: segment 6: paddr=0x000acc40 vaddr=0x40080bdc size=0x0f9a8 ( 63912) load
I (331) esp_image: segment 7: paddr=0x000bc5f0 vaddr=0x400c0000 size=0x00000 (     0) load
I (331) esp_image: segment 8: paddr=0x000bc5f8 vaddr=0x50000000 size=0x00000 (     0) load
I (347) boot: Loaded app from partition at offset 0x10000
I (347) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Starting app cpu, entry point is 0x40080fe0
0x40080fe0: call_start_cpu1 at C:/Users/quint/Documents/QuentinG/esp32_win32_msys2_environment_and_toolchain-20180110/msys32/home/quint/esp/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (363) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (376) heap_init: At 3FFBAB60 len 000254A0 (149 KiB): DRAM
I (382) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (389) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (395) heap_init: At 40090584 len 0000FA7C (62 KiB): IRAM
I (401) cpu_start: Pro cpu start user code
I (83) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (131) wifi: wifi driver task: 3ffc2324, prio:23, stack:3584, core=0
I (131) wifi: wifi firmware version: 0b39f81
I (131) wifi: config NVS flash: enabled
I (141) wifi: config nano formating: disabled
I (141) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (151) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (181) wifi: Init dynamic tx buffer num: 32
I (181) wifi: Init data frame dynamic rx buffer num: 32
I (181) wifi: Init management frame dynamic rx buffer num: 32
I (181) wifi: Init static rx buffer size: 1600
I (181) wifi: Init static rx buffer num: 10
I (191) wifi: Init dynamic rx buffer num: 32
I (251) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (251) wifi: mode : sta (80:7d:3a:c7:56:7c) + softAP (80:7d:3a:c7:56:7d)
I (261) mesh: <nvs>error in read layer, err:0x1102
I (261) mesh: <nvs>error in read assoc, err:0x1102
I (1241) wifi: Set ps type: 0

I (1251) mesh_main: <MESH_EVENT_STARTED>ID:77:77:77:77:77:77
I (1251) mesh_main: mesh starts successfully, heap:184864, root not fixed

W (1611) wifi: scan number 0

I (1611) mesh: [SCAN][ch:4]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0]
I (1611) mesh: [FAIL][1]root:0, fail:1, normal:0, <pre-root>backoff:0

W (1971) wifi: scan number 0

I (1971) mesh: [SCAN][ch:4]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0]
I (1971) mesh: [FAIL][2]root:0, fail:2, normal:0, <pre-root>backoff:0

W (2341) wifi: scan number 0

I (2341) mesh: [SCAN][ch:4]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0]
I (2341) mesh: [FAIL][3]root:0, fail:3, normal:0, <pre-root>backoff:0

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2198330/sdkconfig.zip)

dongdong004, 2018-07-18:
```
I (261) mesh: error in read layer, err:0x1102
I (261) mesh: error in read assoc, err:0x1102
```
There is no problem with these log. It means the nvs of layer and assoc haven't been saved. 
By the way, only root node will save this now.

FayeY, 2018-07-26:
Hi @QuentinGuillet , did the above suggestion help you resolve the issue?

QuentinGuillet, 2018-07-30:
Hello, sorry for the delay. It helped me understand this log yes thank you. My issue was related to the channel I was using for the mesh network. Thanks

closed 2018-07-30 by QuentinGuillet