espressif/esp-idf #2201: Memory leak while running udp_multicast_example_main.c

JetIgniter2k, 2018-07-18:
## Environment

- Development Kit:      [ESP-WROOM-32]

## Problem Description

### Description:

I found a slow memory leak while running udp_multicast_example_main.c
After about 700 loops I run out of heap space and the system crashes.  I was losing 324 bytes per loop.
Adding freeaddrinfo(res); between lines 536 and 537 in udp_multicast_example_main.c seems to fix the problem.

zhangyanjiaoesp, 2018-07-20:
@projectgus please take a look.

projectgus, 2018-07-23:
Hi @JetIgniter2k ,

Thanks very much for reporting this. I can confirm you are correct, a call for freeaddrinfo() is needed in the place you report and also in a second place if IPV4 is enable in the example config.

A fix should be able shortly.

Angus

closed 2018-07-28 by igrr