espressif/esp-idf #2205: Problem in Access point mode with Android 7.0 and 7.1

FlorisColpaert, 2018-07-19:
Hey,

When I try to connect to the Acces point with Android 7.0 or 7.1.2, I don`t get an IP. It worked with Iphone, Android 5.3, 6.1 and 8.0.

IDF version commit e7faa1d (https://github.com/espressif/esp-idf/commit/e7faa1d851556bcfa05cf512175fe3e10463484b)

Kr.

Floris

zhangyanjiaoesp, 2018-08-02:
@FlorisColpaert  I have tested as following:
1. Switch to the IDF version: https://github.com/espressif/esp-idf/commit/e7faa1d851556bcfa05cf512175fe3e10463484b

2. Set ESP32 as AP

3. HUAWEI nova (Android 7.0 ) connects to AP

4. OPPO A83 (Android 7.1.1 ) connects to AP

both of them can get an IP, either the AP is encrypted or not. 

Can you provide the exact type of your test equipment?

zhangyanjiaoesp, 2018-08-20:
@FlorisColpaert please use the latest IDF to test, commit 020ade652d9b6b0b87b0bebe176a150cc4407749

closed 2018-08-29 by FayeY