espressif/esp-idf #2207: Moto G2 Android phone unable to obtain IP in APSTA mode from ESP32

aniket-chati, 2018-07-19:
## Environment
- Development Kit:      ESP32-DevKitC
- Core (if using chip or module): ESP-WROOM32
- IDF version: v3.2-dev-87-gb1fa1ba6
- Development Env: Eclipse
- Operating System: Ubuntu 18.04 LTS
- Power Supply: External 5V (2A)

## Problem Description
In my project, I need to run ESP32 in the AP+STA mode. I am facing a very peculiar problem.

In the AP+STA mode, the ESP correctly connects to my access point and also starts as an access point.

I am able to connect to the ESP access point using iphone (iPhone 8 with Version 11.4). I am also able to connect to it using android phone (Lenovo Vibe K4 note with Android 6.0). 

However, when I try to connect my Android phone (Moto G2 with Android version 6.0), it gets stuck in the "Obtaining IP address" phase and finally ends up with an "Authentication error". On the ESP side, it correctly detects that the phone has connected but fails to allocate IP.
 
Interestingly, with the same code, if I change from AP+STA to AP mode, the Moto G2 also connects perfectly fine.

I have tried to restore factory settings on the phone. I have also tried to erase NVS data (and also erase entire flash!) of the ESP32. But the problem persists.

I have produced the following log with an open wifi configuration. In this case, on the android terminal, the message says 

### Code to reproduce this issue

```cpp
	ESP_ERROR_CHECK(esp_event_loop_init (event_handler_wifi_apsta, NULL) );
	wifi_event_group_config_mode = xEventGroupCreate();

	// Initialize the tcp stack
	tcpip_adapter_init();
	// Stop DHCP server
	ESP_ERROR_CHECK(tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP));
	// assign a static IP to the network interface
	tcpip_adapter_ip_info_t info;
	memset(&info, 0, sizeof(info));
	IP4_ADDR(&info.ip, 192, 168, 1, 1);
	IP4_ADDR(&info.gw, 192, 168, 1, 1);//ESP acts as router, so gw addr will be its own addr
	IP4_ADDR(&info.netmask, 255, 255, 255, 0);
	ESP_ERROR_CHECK(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &info));
	// start the DHCP server
	ESP_ERROR_CHECK(tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP));
	ESP_LOGI(TAG, "DHCP server started \n");

	wifi_init_config_t wifiInitializationConfig = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&wifiInitializationConfig));

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    wifi_config_t apsta_config = {
          .ap = {
            .ssid = ESP_WIFI_AP_SSID,
            .password = ESP_WIFI_AP_PASS,
			.channel = 0,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK,
            .ssid_hidden = 0,
            .max_connection = 1,
            .beacon_interval = 100
          }
    };

    if (strlen(ESP_WIFI_AP_PASS) == 0) {
        apsta_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &apsta_config));
    ESP_ERROR_CHECK(esp_wifi_start());
	ESP_LOGI(TAG, "Access Point with SSID: %s established.", ESP_WIFI_AP_SSID);
```
## Debug Logs

```
I (982) wifi: ap channel adjust o:1,1 n:6,2
I (982) wifi: n:6 0, o:1 0, ap:6 2, sta:6 0, prof:1
I (1962) wifi: state: init -> auth (b0)
D (1962) nvs: nvs_set ap.chan 1 6
I (1972) wifi: state: auth -> assoc (0)
I (1992) wifi: state: assoc -> run (10)
I (2022) wifi: connected with HOMEWIFI, channel 6
D (2022) nvs: nvs_set sta.chan 1 6
D (2032) nvs: nvs_set_blob sta.apinfo 700
I (2032) wifi: pm start, type: 1

D (2032) event: SYSTEM_EVENT_STA_CONNECTED, ssid:XXXXX, ssid_len:5, bssid:00:xx:x0:00:00:00, channel:6, authmode:3
V (2042) event: enter default callback
V (2042) tcpip_adapter: check: local, if=0 fn=0x400ebadc
0x400ebadc: tcpip_adapter_up_api at /opt/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

V (2052) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (2052) tcpip_adapter: check: remote, if=0 fn=0x400ebadc
0x400ebadc: tcpip_adapter_up_api at /opt/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

V (2062) tcpip_adapter: check: local, if=0 fn=0x400ec360
0x400ec360: tcpip_adapter_dhcpc_start_api at /opt/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

D (2062) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (2072) tcpip_adapter: if0 start ip lost tmr: enter
D (2082) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffcaac8 interval=120 ip=0
D (2092) tcpip_adapter: dhcp client start successfully
V (2092) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (2102) tcpip_adapter: check: remote, if=0 fn=0x400ec360
0x400ec360: tcpip_adapter_dhcpc_start_api at /opt/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1082

V (2102) event: exit default callback
D (4542) tcpip_adapter: if0 dhcpc cb
D (4542) tcpip_adapter: if0 ip changed=1
D (4542) event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.1.2, mask:255.255.255.0, gw:192.168.1.1
V (4542) event: enter default callback
I (4552) event: sta ip: 192.168.1.2, mask: 255.255.255.0, gw: 192.168.1.1
V (4552) event: exit default callback
I (4562) config: Got IP: 192.168.1.2
I (24052) wifi: n:6 0, o:6 0, ap:6 2, sta:6 0, prof:6

>> This is where the iphone joins successfully
I (24052) wifi: station: 00:00:aa:00:0a:aa join, AID=1, bg, 20
D (24052) event: SYSTEM_EVENT_AP_STACONNECTED, mac:00:00:aa:00:0a:aa, aid:1
I (24062) config: System Event: Station 00:00:aa:00:0a:aa joined, AID=1
I (25402) tcpip_adapter: softAP assign IP to station,IP is: 192.168.1.2
D (25402) event: SYSTEM_EVENT_AP_STAIPASSIGNED
I (34662) wifi: station: 00:00:aa:00:0a:aa leave, AID = 1
I (34662) wifi: n:6 0, o:6 0, ap:6 2, sta:6 0, prof:6
D (34662) event: SYSTEM_EVENT_AP_STADISCONNECTED, mac:34:08:bc:48:0a:dc, aid:1
I (34672) config: Station:00:00:aa:00:0a:aa left, AID=1
I (48422) wifi: n:6 0, o:6 0, ap:6 2, sta:6 0, prof:6

>> This is where the Moto G2 fails to get IP and disconnects after some attempt 
I (48422) wifi: station: 00:bb:bb:b0:00:b0 join, AID=1, bgn, 20
D (48422) event: SYSTEM_EVENT_AP_STACONNECTED, mac:00:bb:bb:b0:00:b0, aid:1
I (48432) config: System Event: Station 00:bb:bb:b0:00:b0 joined, AID=1
I (79392) wifi: station: 00:bb:bb:b0:00:b0 leave, AID = 1
I (79392) wifi: n:6 0, o:6 0, ap:6 2, sta:6 0, prof:6
D (79392) event: SYSTEM_EVENT_AP_STADISCONNECTED, mac:00:bb:bb:b0:00:b0, aid:1
I (79392) config: Station:00:bb:bb:b0:00:b0 left, AID=1

```
The android log (on the Moto G2 where it fails) is shown below.
```
07-19 19:25:51.715 4219-4219/? D/PollingManager: connectivity_action: [type: WIFI[], state: DISCONNECTED/DISCONNECTED, reason: (unspecified), extra: <unknown ssid>, roaming: false, failover: false, isAvailable: true]
07-19 19:26:20.848 1799-2331/? D/WifiP2pService: InactiveState{ when=-1ms what=143375 target=com.android.internal.util.StateMachine$SmHandler }
    P2pEnabledState{ when=-1ms what=143375 target=com.android.internal.util.StateMachine$SmHandler }
07-19 19:26:20.858 1799-2338/? E/WifiConfigStore: SSID temp disabled for  "ESP32WIFI"NONE had autoJoinStatus=0 self added false ephemeral false
     message=DHCP FAILURE
07-19 19:26:20.860 1799-2338/? E/WifiConfigStore: blacklisted "ESP32WIFI"NONE to 0 due to IP config failures, count=1 disableReason=0 30:ae:a4:84:c4:61 ipfail=7
07-19 19:26:21.257 1799-2338/? E/WifiConfigStore: saveWifiConfigBSSID Setting BSSID for "ESP32WIFI"NONE to any
```

zhangyanjiaoesp, 2018-08-08:
@aniket-chati I have reproduced your problem, and I will reply you as soon as I find the root cause.

aniket-chati, 2018-08-09:
Thankyou, looking forward to your response on this!

zhangyanjiaoesp, 2018-08-20:
@aniket-chati please use the latest IDF for test, commit 020ade652d9b6b0b87b0bebe176a150cc4407749

closed 2018-08-29 by FayeY