espressif/esp-idf #2214: Provide *.SVD for use with editors

beriberikix, 2018-07-22:
http://www.keil.com/pack/doc/CMSIS_Dev/SVD/html/index.html

Arm defines a System View Description file that documents the memory mapped registers of peripherals. Many tools can consume this `*.svd` file and significantly ease debugging. While an Arm thing, I believe it would be possible for Espressif to create an *.svd for the ESP32 and include it in the esp-idf. The entire CMSIS spec is licensed under [Apache 2](http://www.keil.com/pack/doc/CMSIS_Dev/General/html/index.html#License).

chevdor, 2019-05-19:
This may also allow a boost to support Rust and that would be totally awesome.

aykevl, 2019-05-19:
Not just Rust, this is very important to all non-C languages. For example, having SVD files would be very useful for the [TinyGo](https://tinygo.org/) project (as soon as the LLVM Xtensa backend lands upstream) because it avoids having to convert all these register descriptors manually.

As another example, this is [what someone from Ada/SPARK said](https://youtu.be/y8DsNo7yfgE?t=1935) about SVD support (in relation to RISC-V):
> The SVD format was really a game changer for us in terms of support for ARM microcontrollers. [...] I think the RISC-V community should take inspiration from this [...].

In other words, adding support for non-C languages becomes significantly easier with a machine-readable file (preferably SVD) of all the register definitions. SVD is lacking in some areas (for example, it doesn't specify memory layout) but at least having the registers gets you there most of the way.

jeandudey, 2019-12-04:
Are there any updates on this?

imheresamir, 2021-02-12:
Hoping I could get an SVD for the C3 chip

jessebraham, 2021-05-01:
I'm working with a number of other people on bringing preliminary Rust support to the ESP32-C3. Unfortunately at this point we're having to generate an SVD by parsing the header files from this project, which is obviously not ideal and very prone to error. We would all very much appreciate it if an SVD file for this chip (and hopefully others) could be published, and I'm sure many of us would be willing to work alongside Espressif to make that happen if necessary.

dobairoland, 2021-07-21:
Hi all. Anyone interested in SVD files, https://github.com/espressif/svd has been created. There is a [branch](https://github.com/espressif/svd/tree/esp32c3) for ESP32-C3. You can follow the progress there and report any issues.

beriberikix, 2021-10-11:
Definitely! Awesome work 🙌

Techseeker-404, 2022-05-07:
Which will be the svd file i need to go and work with ESP32 Development board WT32-SC01(has a esp32 wrover-b) . As a newbie in esp32 world any support would be helpful.