espressif/esp-idf #2217: Script error variable generates a compilation error

rrobinet, 2018-07-23:
### Hardware:
Board:							?WeMOS LOLIN32
Core Installation/update date:			?espressif-arduino Latest commit e157ec0 6 days ago
IDE name:							?Arduino IDE
Flash Frequency:					?80Mhz?
Upload Speed:						?921600

### Description:
Hi,
While defining a variable with the name "error", the script fails to compile because of a conflict with another "error" variable of the library.
Is it a way to correct that issue, or should I considered it as a current library constraint
Thanks in advance,
Robert
The simple example below highlights the issue.
### Sketch:
```
byte error;
void setup() {
}
void loop() {
}
```
### Debug Messages:
```
.../Arduino/Sketch/hardware/espressif/esp32/tools/sdk/lib/libbtdm_app.a(arch_main.o):(.bss+0x4): multiple definition of `error'
/var/folders/bq/1_7b3x851wj3d2wxhqqntcnr0000gp/T/arduino_build_333423/sketch/sketch_jul22a.ino.cpp.o:(.bss.error+0x0): first defined here
.../Arduino/Sketch/Hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: Warning: size of symbol `error' changed from 1 in /var/folders/bq/1_7b3x851wj3d2wxhqqntcnr0000gp/T/arduino_build_333423/sketch/sketch_jul22a.ino.cpp.o to 4 in .../Arduino/Sketch/hardware/espressif/esp32/tools/sdk/lib/libbtdm_app.a(arch_main.o)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board WEMOS LOLIN32.
```

beegee-tokyo, 2018-07-24:
There are some variable names that are defined already globally by ESP-IDF and you cannot use them.

Try **`int operator = 0;`** and you get the same error.
In PlatformIO/MVSC it is quite easy to see, because such variable names are highlighted when you type them.
Just change the name from **`error`** to **`myError`** and it will compile.

rrobinet, 2018-07-25:
Is it or could be documented, because this is new since previous versions
Robert

UncleRus, 2018-07-26:
IMHO bad design.
Those framework variables should not be global or, at least, needs to be prefixed, smth like `idf_error`

projectgus, 2018-07-27:
> Those framework variables should not be global or, at least, needs to be prefixed, smth like idf_error

Yes, these non-specific global symbol names are a bug and will be renamed.

projectgus, 2018-07-27:
@beegee-tokyo Can you please post the full error message if you declare a global variable `operator`?

(If you're writing C++ then please note that `operator` is a keyword in C++, so the language won't let you declare a variable of that name. That's not something we have control over in IDF.)

beegee-tokyo, 2018-07-27:
@projectgus 
Here is the error if I define **`int operator;`**:
```
In file included from src\ble.cpp:1:0:
src\main.h:24:13: error: expected type-specifier before ';' token
int operator;
^
```
I personally have no problem with these keywords. I guess beside of operator there are more "names" that cannot be used. Also in MSVC/PlatformIO the keywords are colored. 
[image]

projectgus, 2018-07-27:
Thanks for clarifying, @beegee-tokyo . That's a different type of compiler error, nothing to do with ESP-IDF (anyone programming in C/C++ on any platform has certain names intrinsic to the language that they can't use as variables).

The issue originally reported here is a little different (you're allowed to have a global variable called "error" in C/C++, but ESP-IDF accidentally already declares one in an internal library).

rrobinet, 2018-07-27:
I personally thing that symbols / variables, such as event, error, fault , ... are commonly used in scripts and so library should uses more cryptic ones avoiding compilation issue each time a new  release is published

TianHao-Espressif, 2018-08-20:
@beegee-tokyo , `operator` is a internal keyword of c++, so please do not use `operator` as a variable.

rrobinet, 2018-08-22:
Many thanks
Roberr

closed 2018-08-22 by rrobinet