espressif/esp-idf #2218: Bootloader locks up when OTA is set but no OTA partition is present.

OtherCrashOverride, 2018-07-24:
## Environment

- Development Kit:      [ODROID-GO]
- Kit version (v1]
- Core (if using chip or module): [ESP32-Wrover]
- IDF version f198339ec09e90666150672884535802304d23ec
- Development Env:		[Make]
- Operating System:		[|Ubuntu]
- Power Supply:         [USB|Battery]

## Problem Description

When a project is flashed that has an OTA partition and that OTA is set as active, flashing a new project that does not have an OTA partition causes the bootloader to hang.

### Expected Behavior
If an OTA partition is expected, but not present, the factory partition should be used instead.

### Actual Behavior
Bootloader hangs.  The flash must be erased to restore operation.

### Steps to repropduce

1. Create a project with an OTA partition.
2. Perform an OTA.
3. Flash (make flash) a project that does not have an OTA partition.
4. Observe inability to boot.

### Code to reproduce this issue

## Debug Logs

```
I (29) boot: ESP-IDF v3.1-beta1-42-gd5c0610 2nd stage bootloader
I (29) boot: compile time 20:29:57
I (31) boot: Enabling RNG early entropy source...
I (35) qio_mode: Enabling QIO for flash chip GD
I (40) boot: SPI Speed      : 80MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 16MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  3 firmware         factory app      00 00 00010000 000f0000
I (93) boot: End of partition table

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

OtherCrashOverride, 2018-07-24:
Also note that "GPIO triggers factory reset" is not able to recover from this condition.

negativekelvin, 2018-07-24:
The new project has an otadata partition in its partition table?

KonstantinKondrashov, 2018-07-24:
Hi @OtherCrashOverride !
I reproduced your issue. When you change partition table you need to do a command `make erase_flash`.

Regarding why booting not starting from the factory: here has an error of division on zero (https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_utility.c#L253) because you remove ota_x and remain an ota_data partition (bs->app_count = 0). 

We will fix this.

closed 2018-07-28 by igrr