espressif/esp-idf #2219: Add support for %F and %z in strptime() in time.h and/or argtable3.c

luisonoff, 2018-07-24:
## Environment

Doesn't matter. Current master branch.

## Problem Description

strptime() function returns 0 when using %F or %z formatting. 

## Expected Behavior

Google Calendar API gives event dates with strings like this: "2018-07-24T08:40:00+02:00"
I am using SNTP to get UTC time and I am trying to get how much time is left before a date like this string, without having to search what time zone it is, etc.
My plan was:

1. Get date string from Google Calendar API
2. Convert date string to tm structure using strptime() function with "%FT%T%z" formatting.
3. Convert tm structure to UNIX time using mktime()
4. Compare this UNIX time to actual UNIX time got by SNTP.

Right now it is not possible because strptime() returns 0 when using %z format.
**Any better plan on how to accomplish this task?**

## Actual Behavior

I tried to manually parse the offset to later add/remove it from the converted UNIX time, and it works as long as my environment time zone is UTC, if I change my timezone it doesn't work because mktime() thinks it is local time because no other time zone or offset was parsed. Then I would have to take into account the offset of my timezone as well from UTC, which I could do comparing tm struct from localtime() to gmtime(), but it is getting quite complicated.
Workaround I am doing right now is finding what time zone that calendar is, then momentarily setting TZ to that time zone, use strptime() without %z and mktime(), and later change TZ to my real TZ. But this is messy and not thread safe and makes me know what timezone is that Calendar, get the TZ string, etc.

## Extra details

It seems like strptime() is implemented in argtable3.c, and implementation does not include these options. I searched for the original argtable3.c in https://github.com/argtable/argtable3/blob/master/argtable3.c but it does not include these options as well. 
But these options are standard as can be seen here: https://www.gnu.org/software/libc/manual/html_node/Low_002dLevel-Time-String-Parsing.html

%F should be very easy to add, but not sure about %z (some changes needed in tm declaration in time.h?)...

And also mktime() should work properly in order to detect if tm structure has time zone or offset defined, so it is used instead of local one. But I couldn't find source code for mktime().

I am asking about %z, NOT %Z, which I understand is more complex to add because of the need for TZ tables, etc.
I also believe that %z is meant to be used with offsets like +0200 instead of +02:00, and I believe there is a %:z option as well to take into account for the : symbol, that would be perfect, but using %z is good enough, it is easy to modify string prior to strptime() function.

## Code to reproduce this issue

```cpp
char timeString[] = "2018-07-24T08:40:00+02:00";
struct tm timeParsed;
if(strptime(timeString, "%Y-%m-%dT%H:%M:%S%z", &timeParsed))
{
printf("strptime() worked!\n");
}
else
{
printf("strptime() ERROR!\n");
}
time_t gcalendar = mktime(&timeParsed);
printf("time gcalendar : %ld\n",gcalendar);
```

igrr, 2018-07-25:
Hi @luisonoff,

It looks like %z and %F options are Glibc extensions, i couldn't find them in the standard: http://pubs.opengroup.org/onlinepubs/7908799/xsh/strptime.html.

`strptime` function in ESP-IDF is provided by newlib: https://github.com/espressif/newlib-esp32/blob/master/newlib/libc/time/strptime.c. The version we use right now doesn't support these Glibc extensions. Newer newlib versions provide %F but don't provide %z. We do have a plan to update to more recent newlib version, however this may take a few weeks to happen.

luisonoff, 2018-07-25:
Thank you for your answer Ivan.

Do you mind pointing me to an alternate method to achieve what I am trying to do?
As far as I understand, difftime() needs two UNIX times to compare, and the only function that generates UNIX times is mktime() (apart from time() for actual time). But mktime() always thinks that the tm structure input is in local time.

- Should I always convert first to local time/date before comparing?
- Or is there any way to tell mktime() a different UTC offset or time zone area?

Thank you very much for your time.

igrr, 2018-07-25:
I think converting remote timestamp to local time is the easiest approach, given that C standard doesn't provide UTC counterpart for mktime.

I've put together some code which does this, please have a look: https://gist.github.com/igrr/9d55a9c2e625722acd05ad2020cb6fa5

(no guarantees that it handles all the edge cases though; please check thoroughly if you decide to use it)

igrr, 2022-03-11:
'%F' specifier has been supported in Newlib bundled with IDF, since IDF release v4.0: https://github.com/espressif/newlib-esp32/blob/0e4b0f0c505868df6afe84fdc9277f936e303ad6/newlib/libc/time/strptime.c#L236.

'%z' and '%Z' still appear to be unsupported in upstream Newlib: https://github.com/bminor/newlib/blob/b1743c94e25cd7bb5edbd77f694b1a3e5d90c297/newlib/libc/time/strptime.c#L424-L426

KonstantinKondrashov, 2023-10-11:
Closing it, for now, because ['%F' specifier has been supported in Newlib bundled with IDF, since IDF release v4.0](https://github.com/espressif/newlib-esp32/blob/0e4b0f0c505868df6afe84fdc9277f936e303ad6/newlib/libc/time/strptime.c#L236).
The '%z' and '%Z' specifiers are not going to be supported because it is unsupported in [upstream Newlib](https://github.com/bminor/newlib/blob/b1743c94e25cd7bb5edbd77f694b1a3e5d90c297/newlib/libc/time/strptime.c#L424-L426).
Thanks.

closed 2023-10-11 by KonstantinKondrashov