espressif/esp-idf #2221: Combining the UART and PWM codes

AmbimatElectronics, 2018-07-24:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

[YES] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.

3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (WroverKit): [v2] - Black PCB
- Core (if using chip or module): [ESP32-Wrover]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[Eclipse]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

We are trying to integrate the UART and the PWM codes and with inputs from UART we are trying to change the PWM of different channels. When we add a new task program is not running properly in debug mode.

### Code to reproduce this issue
```cpp

https://gist.github.com/AmbimatElectronics/98dd8d3b0dde7383957b6e54083dd279

```
## Debug Logs

1) On the “Main” tab below, under “Project:”, i have select the “uarst_async_task” project.
2) In next line “C/C++ Application:” i have selected select “uarst_async_task.elf” file.
3) under “Build (if required) before launching” ---> select “Disable auto build”.
4) on “Debugger” tab. In field “GDB Command” --->entered xtensa-esp32-elf-gdb to invoke debugger.
5) on default configuration of “Remote host” i have entered 3333 under the “Port number”.
6) on “Startup” tab. Under “Initialization Commands” uncheck “Reset and Delay (seconds)” and “Halt”“. Then, in entered type 
mon reset halt and 
flushregs (in two separate lines).
7) Under “Load Image and Symbols” uncheck “Load image” option.

i am following esp guidline for debugging  setting which link is below:
https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse

Also see attachments of the screenshots.

This is the error we are facing

[image]

This is where the program gets stuck while debugging: 

[image]

Starting position/sequence of the code.

[image]

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2223138/sdkconfig.zip)
[sdkconfig.old.zip](https://github.com/espressif/esp-idf/files/2223139/sdkconfig.old.zip)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[uart_async_rxtxtasks.elf.zip](https://github.com/espressif/esp-idf/files/2223140/uart_async_rxtxtasks.elf.zip)

- [ ] coredump (This provides stacks of tasks.)

xiongyumail, 2018-08-13:
Hi,@AmbimatElectronics
Because esp32 is a dual core. when using interrupt-related tasks, I suggest you use `xTaskCreatePinnedToCore` to create task.
<https://esp-idf.readthedocs.io/en/latest/api-reference/system/intr_alloc.html?highlight=xTaskCreatePinnedToCore>

Alvin1Zhang, 2018-10-17:
@AmbimatElectronics Could you help share if any updates for this issue? Thanks.

xiongyumail, 2018-10-18:
Has the problem been solved?

AmbimatElectronics, 2018-10-18:
Hello,

We haven’t worked on this challenge for some time.

We will get back to you shortly.

Thanks

Ambimat Electronics
support@ambimat.com

koobest, 2018-10-29:
Hi, @AmbimatElectronics @Alvin1Zhang 
Please allocate enough memory for your task. e.g:
`
 xTaskCreate(rx_task, "uart_rx_task", 1024*2, NULL, configMAX_PRIORITIES,    NULL);
`

AmbimatElectronics, 2018-12-17:
> Hi, @AmbimatElectronics <https://github.com/AmbimatElectronics>,
> Have you solved the problem?

Hello,

No the problem is still open. The solution provided did not work for us.

We have not been able to solve it yet.

Thanks

Ambimat Electronics
support@ambimat.com

wingunder, 2019-01-08:
@AmbimatElectronics, I could reproduce and fix your problem, at least on my side. Please see the following gist for the exact details: https://gist.github.com/wingunder/4727514b4d6a1995d6174603fbe3bfa1

You have the following 2 problems in your example:
1. As @koobest pointed out, you'll have to use at lease 2k bytes for initializing the UART tasks. Please see the following example that I created a while ago (BTW, I assume you even based your example on this.):
https://github.com/espressif/esp-idf/blob/456efd3d57ddead91b0675461a1d2cec3a04c029/examples/peripherals/uart/uart_async_rxtxtasks/main/uart_async_rxtxtasks_main.c#L70-L75
2. You will have to vTaskDelay for at least the time that it takes for your requested ledc fade action, after issuing it. So simply uncomment the the 2 vTaskDelay lines in the pwm_task() function.

With these fixes, it works for me, on commit db8bc3ee, so no patching is needed.

@AmbimatElectronics, could you please try all of this on your side and let us all know?

@koobest, For verification, could you please try to run @AmbimatElectronics's code, as well my fixed gist that I mentioned above?

koobest, 2019-07-15:
@AmbimatElectronics Have  you solved your issue?
@wingunder Thanks, your code works fine.

Alvin1Zhang, 2020-06-08:
@AmbimatElectronics Thanks for reporting and thanks @wingunder for sharing the updates. Will close the ticket now, feel free to reopen with more details. Thanks.

closed 2020-06-08 by Alvin1Zhang