espressif/esp-idf #2222: WPA2-Enterprise not possible with an old Netgear

lollizze, 2018-07-24:
## Environment

- Development Kit: none
- Core (if using chip or module): ESP32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): f198339ec09e90666150672884535802304d23ec
- Development Env:		other
- Operating System:		Windows
- Power Supply:         external 3.3V

## Problem Description

I'm am trying to implement Enterprise, but it seems the module is behaving strange against netgear wg302
Both PEAP and EAPTLS fails when connecting. It looks like esp32 starts to use TKIP instead of AES.
My settings in netgear is WPA2 with Radius AES, if i switch it to WPA and WPA2 with RADIUS AES/TKIP then it connects

### Expected Behavior
Connect success
### Actual Behavior
No connection
### Steps to repropduce
netgear wg302, WPA2 with Radius AES, EAPTLS Radius server

I have attached a wireshark log for both eaptls and peap that shows two ways esp32 responds
One where it fails after Association probably due to the fact it sends TKIP as group cipher and one where the radius exchange succeeds but it then fails at key exchange 

[wireshark_log.zip](https://github.com/espressif/esp-idf/files/2223798/wireshark_log.zip)

XinDeng11, 2018-08-10:
@lollizze Sorry for later reply, yes, it is strange. The group cipher is ccmp in the assoc request, but it change to tkip when key exchange. It should fail when AP check eapol 2, I will check our code flow, thanks

lollizze, 2018-08-10:
@XinDeng11 let me know if i should test anything

XinDeng11, 2018-08-27:
[image]
[image]
[image]
[image]

lollizze, 2018-08-27:
I think our AP must be version 1, because the firmware is 4.2.17

XinDeng11, 2018-08-27:
can smartphone connect to the router?

lollizze, 2018-08-27:
Yes other devices can connect

XinDeng11, 2018-08-27:
Form the sniffer log, I found it fail at 4 way handshake, rather than EAP, I wonder whether it can connect just use wpa2-AES person, could you do that test?

lollizze, 2018-08-27:
Yes you are correct, same behavior with wap2-AES

lollizze, 2018-08-27:
No sorry, i did something wrong. I can connect when it is normal wpa2-aes it is only when it is enterprise

XinDeng11, 2018-08-27:
OK, it looks like a compatible issue about wpa2-aes, have no relation ship with enterprise, I want to compare the difference between the airpacket, if it could, can you send me a airpacket only use wpa2-aes person, thanks

XinDeng11, 2018-08-27:
OK, can you send me one wpa2 aes air packet? I want to compare the difference bewteen them

lollizze, 2018-08-27:
Yes, i will capture that

lollizze, 2018-08-27:
This is interesting, with enterprise it always fails somewhere, with wpa2-aes i can get a successful connection and send data as i got when i tested before. Now when i tried to capture the air traffic i get 5 failures then I am associated but no ip address, probably wrong encryption or something similar
[wireshark_wpa2_aes.zip](https://github.com/espressif/esp-idf/files/2323360/wireshark_wpa2_aes.zip)

XinDeng11, 2018-08-27:
No difference, could you help to catch a packet include beacon, thanks

lollizze, 2018-08-27:
Oh i filtered those, here is same capture only filtered on AP address
[wpa2_aes.zip](https://github.com/espressif/esp-idf/files/2323634/wpa2_aes.zip)

XinDeng11, 2018-08-27:
could you help to catch another packet use wpa2 enterprise, very thanks

XinDeng11, 2018-08-27:
I mean include beacon

XinDeng11, 2018-08-27:
yes, I know, I found the RSN IE is different between that two firwmare, but  I want another packets use the wpa2 enterprise which include beacon to make sure, could you help offer one?

lollizze, 2018-08-27:
Sorry I misunderstood here is the beacons for enterprise
[beacon_ent_wpa2.zip](https://github.com/espressif/esp-idf/files/2323870/beacon_ent_wpa2.zip)

XinDeng11, 2018-08-28:
Hi@lollizze, I found the root cause, look at the packet 4.2.17, The RSN IE in beacon set group cipher as TKIP, however it set group cipher as CCMP in probe response
[image]
[image]
We need parse the RSN from beacon or probe response to set wpa2 parameters, if we receive beacon, the group cipher will be recorded to TKIP, otherwise set to CCMP. From the AP router behavior I think it hope the group cipher is CCMP rather than TKIP. So that why you can success sometime when use wpa2 aes person. However, the WPA Enterprise need finish EAP step first, then set RSN IE, there only beacon can received when do EAP, so that it always record TKIP when do enterprise, that the reason why it always fail.
The firmware 5.2.3 use the same group cipher in beacon and probe response, so that it can success.
[image]
[image]
Althougth there is no clearly requirment the RSN IE shouldn't change in beacon and probe response, but I think it had better to use the same. So I suggestion to update your router's firmware. Thanks to your kindly help.

XinDeng11, 2018-08-28:
it has risk, if we do that, we may loss the wpa2 parameter if the probe response is not received

PaulFreund, 2018-08-28:
@XinDeng11 I'm really sorry to hijack this thread but may I kindly ask you to take a look at https://www.esp32.com/viewtopic.php?f=2&t=3108&start=10#p28331. You seem to have the necessary knowledge to explain this, especially if I should be able to connect without client certificate.

XinDeng11, 2018-08-28:
@PaulFreund If it possible, offer you can offer the air packet like lollize, it will very useful to find the wrong place

PaulFreund, 2018-08-28:
@XinDeng11 sorry I moved my comment to a more appropriate issue because it is not really related to this one: https://github.com/espressif/esp-idf/issues/2152

I'm currently trying to find a wifi card that supports monitor mode but I don't recognize "lollize" I'd try to get a pcap file.

XinDeng11, 2018-08-28:
[add_compatible_to_wpa2.tar.gz](https://github.com/espressif/esp-idf/files/2328000/add_compatible_to_wpa2.tar.gz)
@lollizze I do some change, it might be helpful, hope you can hlep to use the attachment do test, it is based on the lastest IDF, remember to update your IDF to the latest, if it work, we will discussion whether add the change to offical version

lollizze, 2018-08-28:
@XinDeng11 latest on 3.1?

XinDeng11, 2018-08-28:
latest on Github, or you can give me the lib commit ID you used, I will add those change in it

XinDeng11, 2018-08-29:
[add_compatible_to_wpa2.tar.gz](https://github.com/espressif/esp-idf/files/2330514/add_compatible_to_wpa2.tar.gz)
Please use this one

closed 2018-11-02 by FayeY