espressif/esp-idf #2223: apptrace_proc.py is dependent on libelf.so.1

liangli2718, 2018-07-24:
When trying to use application level tracing to achieve Logging to Host via JTAG, I need to use the example python script to decode the recorded log.

The doc I was following is:
https://esp-idf.readthedocs.io/en/latest/api-guides/app_trace.html#logging-to-host

I got error:
 Traceback (most recent call last):
  File ".\logtrace_proc.py", line 7, in <module>
    import pylibelf as elf
  File "C:\msys32\home\intel\esp\esp-idf\tools\esp_app_trace\pylibelf\__init__.py", line 5, in <module>
    lelf=CDLL("libelf.so.1")
  File "C:\Python27\lib\ctypes\__init__.py", line 353, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126] The specified module could not be found

The libelf.so.1 is referenced in 
https://github.com/espressif/esp-idf/blob/master/tools/esp_app_trace/pylibelf/__init__.py

I don't think this is a Windows lib, is there a way to make this python script work on Windows?

Thanks!

liangli2718, 2018-07-26:
Seems the only way is to run this script under linux with libelf in python2.

pavelskirko, 2018-08-03:
Is there a way to run it on windows?

liangli2718, 2018-08-03:
@pavelskirko at least I didn't find a way. Now I am using the script in a linux virtual machine.

pavelskirko, 2018-08-03:
@liangli2718 I was hoping I wouldn't have to do that. Ok, now it is the only way. Thanks.

WayneKeenan, 2019-03-03:
There needs to be a warning in the docs, just wasted a chance of time on macOS to get this far and find it doesn't work.

WayneKeenan, 2019-03-03:
The warning should be for Windows users, at least.

I've just found that `brew install libelf` will install but my Pythons are not picking it up.

drorgl, 2019-10-27:
I'm not sure how reliable it is, but I've made logtrace_proc work with [pylibelf](https://github.com/crackinglandia/pylibelf)

I've only tested it with [app_trace_to_host](https://github.com/espressif/esp-idf/tree/master/examples/system/app_trace_to_host), it seems to be working, it was a 5 minute change.

If the esp-idf team has time, perhaps they can test it and see if it can be part of the official release

igrr, 2022-03-14:
As of commit 8c6a924cdfa6c72e88f2014419f2aeb0c6a0e9a7, apptrace related scripts use https://github.com/eliben/pyelftools, which is a python-only library. I'm sorry that we forgot about this issue and didn't update it in time.

closed 2022-03-14 by igrr