espressif/esp-idf #2230: Sparkfun ESP32 WiFi Low Power MQTT

Sedwin97, 2018-07-26:
Board: Sparkfun ESP32 Thing
IDF version: aaf12390 
Development Env: esp-idf
Operating System: Windows 10
Power Supply: USB

I'm working on research with a PhD student and I need to run an MQTT protocol with Wifi Power Save Mode on the Sparkfun ESP32 Thing board. We are using an unsecured WiFi network and simply connecting to an MQTT broker on a raspberry pi, publishing a message, and disconnecting. Two issues have come up.

1. esp_mqtt_client_stop does not send a disconnect command (checking by using Wireshark)
I went through the API and could not understand why this was a problem. The connect command and ack and a publish packet are sent but not the disconnects.
(I'm also seeing a ton of out-of order, dup ack, and acked unseen segment packets. What might be causing that?)

2. The second problem is that we need to modify a part of the U-APSD operation. I need to keep the board alive for a few ms after it would receive some data.
A similar function is shown in the image. This was used on a Cypress board using WICED and allowed us to modify the return to sleep delay of the wifi power save. I've looked all over the API but wasn't able to find anything that worked similarly. Is this a supported feature?
[image]

Here is the code. I just mixed the power save example and the mqtt example together.
https://gist.github.com/Sedwin97/308ab1f562fbd9984a67023ade0c194e

igrr, 2018-07-30:
@Sedwin97 For the MQTT issue, do i understand right that you are using https://github.com/tuanpmt/espmqtt library? If yes, could you please open the issue there?

Regarding whether return-to-sleep time can be configured: WiFi driver for ESP32 does not provide such API at the moment. @TimXia please advise whether this can be made configurable.

TimXia, 2018-08-08:
@Sedwin97 @igrr Yes, it can be made configurable. But I do not suggest to do this. It will make things more complexible for most of our users do not understand the theory of power save. In fact, we may make the return-to-sleep time dynamic ajustment according to the throughput. It's just an idea now. I will think of it and try to realize it later.

TimXia, 2018-08-13:
@Sedwin97 In fact, ESP32 will keep alive for 50ms after sending or receiving a packet. For most application, 50ms return to sleep delay time is enough. Do mean that you need to modify this time?

Sedwin97, 2018-08-13:
@TimXia Yes. The experiment is to modify that time and try to minimize power consumption while also minimizing latency across IoT boards.

Sedwin97, 2018-09-07:
Hi @TimXia, any updates on when the API might become available

TimXia, 2018-09-09:
@Sedwin97 It's almost done. We are considering whether to provide it on IDF or just provide it to you.

Sedwin97, 2018-09-28:
Hi @TimXia, any updates? Did you provide it on IDF?

TianaESP, 2021-11-16:
@Sedwin97  Sorry for the super late reply. The latest master should have solved the issue. Can you please give it a try? Thanks.

AxelLin, 2021-11-18:
> @Sedwin97 OK. We will provide an API to modify this time.

@TimXia @TianaESP
What is the function name for this API?

Sedwin97, 2021-11-18:
Hi @TianaESP, I don't work on this project anymore so I don't have the ability to try this out anymore

closed 2022-01-05 by jack0c