espressif/esp-idf #2232: Esp32 a service to add multiple character issues.

MessCodes, 2018-07-27:
To develop a BLE application using ESP32, you need to add multiple characters to a service, add a character according to the gatt_server project, add a character failure, and report a 133 error. Log is as follows:
```
I (524) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0
I (524) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40
I (534) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40
I (534) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (534) GATTS_DEMO: **** Add char event: 0xFF01
I (544) GATTS_DEMO: the gatts demo char length = 3
I (554) GATTS_DEMO: prf_char[0] =11
I (554) GATTS_DEMO: prf_char[1] =22
I (554) GATTS_DEMO: prf_char[2] =33

I (564) GATTS_DEMO: ADD_CHAR_EVT, status 133,  attr_handle 0, service_handle 40
I (574) GATTS_DEMO: **** Add char event: 0xEE01
I (574) GATTS_DEMO: ##########################
I (584) GATTS_DEMO: **** Add char-desc event: 0x2B, [0]=0x2A, [1]==0x0
I (584) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40
I (594) GATTS_DEMO: **** Add char-desc event: 0x0, [0]=0x2A, [1]==0x0
I (604) GATTS_DEMO: ADD_DESCR_EVT, status 133, attr_handle 0, service_handle 40
I (614) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1
I (614) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44
I (624) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44
I (634) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44
I (644) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44
```

chegewara, 2018-07-27:
Im not sure, but try to increase this value:
`GATTS_NUM_HANDLE_TEST_A     `
and
`GATTS_NUM_HANDLE_TEST_B     `

MessCodes, 2018-07-27:
@chegewara You are right, Thks!

chegewara, 2018-07-27:
For some reason handle value is increased by 2, so you need to reserve for each service (just conclusion):
`( num of characteristics + num of descriptors) * 2`

dhrishi, 2018-07-27:
Every characteristic has a characteristic declaration and value (both mandatory) each of which is assigned a handle.

So, for e.g., you  have 1 service A with 2 characteristics Y and Z and the characteristic Y has a descriptor D then total handles for that service will be will be:
1 (for service A) + 2 (characteristic Y declaration and characteristic Y value) + 1 (characteristic descriptor D) + 2 (characteristic Z declaration and characteristic Z value) = 6

In general, the number of handles for each service:
1 (for service) + (2 * number of characteristics) + total number of characteristic descriptors

MessCodes, 2018-07-27:
@dhrishi That's it! I have added successfully, Thks!

closed 2018-07-28 by MessCodes